arendst/Tasmota #18468: Protocol kelon168 not work

ermes64, 2023-04-21:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Good day, sorry for my english
I'm using a tuya smart IR blaster tyir-rmc modified with cip 8266 to control an AC Hisense 16000 with tx DG11R2-01 remote, but when I send the IR code from consol with irsend "Protocol":"KELON168","Bits ":168,"Data":"0x830605740000940E0000000000EB00010000280029","Repeat":0}
it is accepted with RESULT = {"IRSend":"Done"} but the IR does not turn on the LED, I tried with an ascilloscope but nothing comes out of the GPIO14, I tried modifying the code with irsend "Protocol":"KELON"," Bits":168,"Data":"0x830605740000940E0000000000EB00010000280029","Repeat":0} and the led turns on but rightly it doesn't control AC, if I try to send data in raw mode it works and I control AC but this is difficult
Could you tell me how to solve the problem
Thank you

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): esp32 and esp8266
- [x ] Tasmota binary firmware version number used: 12.5.0(ir)
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:

  STATUS 0 output here:
17:55:51.992 CMD: status 0
17:55:51.999 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_540F3C","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
17:55:51.004 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz","RestartReason":"Power On","Uptime":"0T06:03:17","StartupUTC":"2023-04-21T10:52:34","Sleep":50,"CfgHolder":4617,"BootCount":35,"BCResetTime":"2023-04-16T12:44:03","SaveCount":88,"SaveAddress":"F9000"}}
17:55:51.007 RSL: STATUS2 = {"StatusFWR":{"Version":"12.5.0(ir)","BuildDateTime":"2023-04-17T08:03:55","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"352/699"}}
17:55:51.011 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["casa-iliad",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A00192800000000","00000180","00006000","00004000","00000000"]}}
17:55:51.018 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":638,"Free":364,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","0F1A0582","04000001","00000003","00000000","00000000","00000000","00000000","04000000","00000080"],"Drivers":"1,2,4,5,7,9,10,12,62","Sensors":"1"}}
17:55:51.023 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-540F3C-3900","IPAddress":"192.168.8.144","Gateway":"192.168.8.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.8.1","DNSServer2":"0.0.0.0","Mac":"C8:C9:A3:54:0F:3C","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
17:55:51.026 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_540F3C","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
17:55:51.032 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-04-21T16:55:51","Local":"2023-04-21T17:55:51","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"05:48","Sunset":"19:49"}}
17:55:51.035 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-04-21T17:55:51"}}
17:55:51.041 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-04-21T17:55:51","Uptime":"0T06:03:17","UptimeSec":21797,"Vcc":3.483,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"casa-iliad","BSSId":"A0:57:E3:FB:C2:CC","Channel":9,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
18:03:07.064 CMD: irsend {"Protocol":"KELON168","Bits":168,"Data":"0x830605740000940E0000000000EB00010000280029","Repeat":0}
18:03:07.071 RSL: RESULT = {"IRSend":"Done"}

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-05-10:
Can you try with latest development build? The IR lib has been updated to latest.

ermes64, 2023-05-16:
Hi
I tried to install the latest version 12.5.0.2(ir) but the problem does not solve

0:26:22.674 MQT: stat/IRAC/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"IRAC","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
20:26:22.680 MQT: stat/IRAC/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-minimal.bin.gz","RestartReason":"Power On","Uptime":"2T09:58:27","StartupUTC":"2023-05-14T09:27:55","Sleep":50,"CfgHolder":4617,"BootCount":62,"BCResetTime":"2023-04-16T12:44:03","SaveCount":136,"SaveAddress":"F9000"}}
20:26:22.685 MQT: stat/IRAC/STATUS2 = {"StatusFWR":{"Version":"12.5.0.2(ir)","BuildDateTime":"2023-05-07T15:26:03","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"337/699"}}
20:26:22.689 MQT: stat/IRAC/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":3,"MqttLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["casa-iliad",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A00192800000000","00000180","00006000","00004000","00000000"]}}
20:26:22.698 MQT: stat/IRAC/STATUS4 = {"StatusMEM":{"ProgramSize":639,"Free":364,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","0F1A0582","04000001","00000003","00000000","00000000","00000000","00000000","04000000","00000080"],"Drivers":"1,2,4,5,7,9,10,12,62,68","Sensors":"1"}}
20:26:22.706 MQT: stat/IRAC/STATUS5 = {"StatusNET":{"Hostname":"IRAC-3900","IPAddress":"192.168.8.144","Gateway":"192.168.8.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.8.1","DNSServer2":"0.0.0.0","Mac":"C8:C9:A3:54:0F:3C","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
20:26:22.711 MQT: stat/IRAC/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.8.128","MqttPort":1883,"MqttClientMask":"IRAC1","MqttClient":"IRAC1","MqttUser":"Admin","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
20:26:22.718 MQT: stat/IRAC/STATUS7 = {"StatusTIM":{"UTC":"2023-05-16T19:26:22","Local":"2023-05-16T20:26:22","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"05:07","Sunset":"20:25"}}
20:26:22.722 MQT: stat/IRAC/STATUS10 = {"StatusSNS":{"Time":"2023-05-16T20:26:22"}}
20:26:22.728 MQT: stat/IRAC/STATUS11 = {"StatusSTS":{"Time":"2023-05-16T20:26:22","Uptime":"2T09:58:27","UptimeSec":208707,"Vcc":3.484,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"casa-iliad","BSSId":"A0:57:E3:FB:C2:CC","Channel":2,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:05"}}}

Jason2866, 2023-05-17:
Mhh, `BuildDateTime":"2023-05-07T15:26:03",` is not a actual one with the changed IR lib.

ermes64, 2023-05-17:
I uploaded the latest version 12,5,0,3 of 17_5_2023 but it does not work, the LEDs do not light up,I tried to change KELON168 with KELON and LEDs light up but does not go AC

19:26:02.062 CMD: irsend {"Protocol":"KELON168","Bits":168,"Data":0x830600A2000080020000000000200002000038003A,"Repeat":0}
19:26:02.064 SRC: WebConsole from 192.168.8.145
19:26:02.068 CMD: Grp 0, Cmd 'IRSEND', Idx 1, Len 97, Pld -99, Data '{"Protocol":"KELON168","Bits":168,"Data":0x830600A2000080020000000000200002000038003A,"Repeat":0}'
19:26:02.090 MQT: stat/IRAC/RESULT = {"IRSend":"Done"}

19:25:17.692 CMD: irsend {"Protocol":"KELON","Bits":168,"Data":0x830600A2000080020000000000200002000038003A,"Repeat":0}
19:25:17.694 SRC: WebConsole from 192.168.8.145
19:25:17.698 CMD: Grp 0, Cmd 'IRSEND', Idx 1, Len 94, Pld -99, Data '{"Protocol":"KELON","Bits":168,"Data":0x830600A2000080020000000000200002000038003A,"Repeat":0}'
19:25:18.198 MQT: stat/IRAC/RESULT = {"IRSend":"Done"}

19:27:17.059 CMD: status 0
19:27:17.061 SRC: WebConsole from 192.168.8.145
19:27:17.064 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
19:27:17.071 MQT: stat/IRAC/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"IRAC","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
19:27:17.107 MQT: stat/IRAC/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-ir.bin.gz","RestartReason":"Power On","Uptime":"0T00:02:30","StartupUTC":"2023-05-17T18:24:47","Sleep":50,"CfgHolder":4617,"BootCount":68,"BCResetTime":"2023-04-16T12:44:03","SaveCount":147,"SaveAddress":"F7000"}}
19:27:17.140 MQT: stat/IRAC/STATUS2 = {"StatusFWR":{"Version":"12.5.0.3(ir)","BuildDateTime":"2023-05-17T11:26:52","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"332/699"}}
19:27:17.157 MQT: stat/IRAC/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":3,"MqttLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["casa-iliad",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A00192800000000","00000180","00006000","00004000","00000000"]}}
19:27:17.189 MQT: stat/IRAC/STATUS4 = {"StatusMEM":{"ProgramSize":648,"Free":356,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","0F1A0582","04000001","00000003","00000000","00000000","00000000","00000000","04000000","00000080"],"Drivers":"1,2,4,5,7,9,10,12,62,68","Sensors":"1"}}
19:27:17.219 MQT: stat/IRAC/STATUS5 = {"StatusNET":{"Hostname":"IRAC-3900","IPAddress":"192.168.8.144","Gateway":"192.168.8.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.8.1","DNSServer2":"0.0.0.0","Mac":"C8:C9:A3:54:0F:3C","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:27:17.245 MQT: stat/IRAC/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.8.128","MqttPort":1883,"MqttClientMask":"IRAC1","MqttClient":"IRAC1","MqttUser":"Admin","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:27:17.265 MQT: stat/IRAC/STATUS7 = {"StatusTIM":{"UTC":"2023-05-17T18:27:17","Local":"2023-05-17T19:27:17","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"05:06","Sunset":"20:26"}}
19:27:17.284 MQT: stat/IRAC/STATUS10 = {"StatusSNS":{"Time":"2023-05-17T19:27:17"}}
19:27:17.293 MQT: stat/IRAC/STATUS11 = {"StatusSTS":{"Time":"2023-05-17T19:27:17","Uptime":"0T00:02:30","UptimeSec":150,"Vcc":3.480,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"casa-iliad","BSSId":"A0:57:E3:FB:C2:CC","Channel":2,"Mode":"11n","RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:05"}}}

closed 2023-06-18 by github-actions[bot]