arendst/Tasmota #18469: ModuleAddress command doesn't work for PZEM-004t V3 energy sensor

fellev, 2023-04-21:
### PROBLEM DESCRIPTION
ModuleAddress command doesn't work for PZEM-004t V3 energy sensor.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [V] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [V] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [V] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [V] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [V] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [V] Device used (e.g., Sonoff Basic): Nodemcy-CH9102-V2.1 connected to PZEM-004T V3
- [V] Tasmota binary firmware version number used: latest development branch 2023.04.03 and latest official release 12.5.0
  - [V] Pre-compiled
  - [V] Self-compiled
- [V] Flashing tools used: esptool.py
- [V] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:39:40.157 CMD: Backlog Template; Module; GPIO 255
19:39:40.394 MQT: stat/power_meter/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
19:39:40.593 MQT: stat/power_meter/RESULT = {"Module":{"0":"Generic"}}
19:39:40.799 MQT: stat/power_meter/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"1472":"PZEM0XX Tx"},"GPIO2":{"0":"None"},"GPIO3":{"1536":"PZEM016 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
19:39:44.498 MQT: tele/power_meter/STATE = {"Time":"2023-04-21T19:39:44","Uptime":"0T00:38:36","UptimeSec":2316,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":3,"Wifi":{"AP":1,"SSId":"MyHomeNetwork","BSSId":"8C:DE:F9:BF:61:88","Channel":6,"Mode":"11n","RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:04"}}
19:39:44.510 MQT: tele/power_meter/SENSOR = {"Time":"2023-04-21T19:39:44","ENERGY":{"TotalStartTime":"2023-04-21T12:57:57","Total":0.023,"Yesterday":0.000,"Today":0.023,"Period":[0,0,0],"Power":[0,0,0],"ApparentPower":[0,0,0],"ReactivePower":[0,0,0],"Factor":[0.00,0.00,0.00],"Frequency":[50,50,50],"Voltage":[231,231,231],"Current":[0.000,0.000,0.000]}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [V] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:40:59.881 MQT: stat/power_meter/STATUS = {"Status":{"Module":0,"DeviceName":"Power Meter","FriendlyName":["pzem004t"],"Topic":"power_meter","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
19:40:59.888 MQT: stat/power_meter/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:39:51","StartupUTC":"2023-04-21T18:01:08","Sleep":50,"CfgHolder":4617,"BootCount":30,"BCResetTime":"2023-04-21T12:57:57","SaveCount":49,"SaveAddress":"F7000"}}
19:40:59.895 MQT: stat/power_meter/STATUS2 = {"StatusFWR":{"Version":"12.1.1.6(tasmota)","BuildDateTime":"2023-04-21T16:31:47","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"364/699"}}
19:40:59.902 MQT: stat/power_meter/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MyHomeNetwork",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
19:40:59.917 MQT: stat/power_meter/STATUS4 = {"StatusMEM":{"ProgramSize":642,"Free":360,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8FDA8787","04368001","000000CF","010013C0","C000D981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,16,18,19,20,21,22,24,27,29,30,35,37,45,62,88","Sensors":"1,2,3,4,5,6"}}
19:40:59.924 MQT: stat/power_meter/STATUS5 = {"StatusNET":{"Hostname":"power-meter-3283","IPAddress":"192.168.31.96","Gateway":"192.168.31.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.31.1","DNSServer2":"0.0.0.0","Mac":"48:55:19:DF:8C:D3","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:40:59.930 MQT: stat/power_meter/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.31.78","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DF8CD3","MqttUser":"DVES_USER","MqttCount":3,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:40:59.939 MQT: stat/power_meter/STATUS7 = {"StatusTIM":{"UTC":"2023-04-21T18:40:59","Local":"2023-04-21T19:40:59","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"05:48","Sunset":"19:49"}}
19:40:59.945 MQT: stat/power_meter/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
19:40:59.958 MQT: stat/power_meter/STATUS10 = {"StatusSNS":{"Time":"2023-04-21T19:40:59","ENERGY":{"TotalStartTime":"2023-04-21T12:57:57","Total":0.023,"Yesterday":0.000,"Today":0.023,"Power":[0,0,0],"ApparentPower":[0,0,0],"ReactivePower":[0,0,0],"Factor":[0.00,0.00,0.00],"Frequency":[50,50,50],"Voltage":[231,231,231],"Current":[0.000,0.000,0.000]}}}
19:40:59.967 MQT: stat/power_meter/STATUS11 = {"StatusSTS":{"Time":"2023-04-21T19:40:59","Uptime":"0T00:39:51","UptimeSec":2391,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":41,"MqttCount":3,"Wifi":{"AP":1,"SSId":"MyHomeNetwork","BSSId":"8C:DE:F9:BF:61:88","Channel":6,"Mode":"11n","RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:04"}}}

- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

13:57:48.032 CMD: ModuleAddress 1
13:57:48.034 SRC: WebConsole from 192.168.31.176
13:57:48.036 CMD: Grp 0, Cmd 'MODULEADDRESS', Idx 1, Len 1, Pld 1, Data '1'
13:57:48.040 RSL: RESULT = {"ModuleAddress":"Done"}
13:57:48.401 MBS: Serial Received: 01 04 14 08 D9 00 00 00 00 00 00 00 00 00 0C 00 00 01 F4 00 00 00 00 B6 83 
13:57:48.403 DMP: 01 04 14 08 D9 00 00 00 00 00 00 00 00 00 0C 00 00 01 F4 00 00 00 00 B6 83
13:57:48.405 NRG: EnergyTotal[0] 0.0120 kWh
13:57:48.407 Set address 1 
13:57:52.442 NRG: Energy reset by invalid data
13:57:53.400 MBS: Serial Send: 01 04 00 00 00 0A 70 0D 
13:57:54.401 MBS: Serial Received: 01 04 14 08 D8 00 00 00 00 
13:57:54.403 DMP: 01 04 14 08 D8 00 00 00 00
13:57:54.404 PAC: PzemAc 1 error 9
13:57:54.406 MBS: Serial Send: 01 04 00 00 00 0A 70 0D 
13:57:55.401 MBS: Serial Received: 01 F4 00 00 00 00 8B 7F 01 04 14 08 D7 00 00 00 00 00 00 00 00 00 0C 00 00 
13:57:55.403 DMP: 01 F4 00 00 00 00 8B 7F 01 04 14 08 D7 00 00 00 00 00 00 00 00 00 0C 00 00
13:57:55.405 PAC: PzemAc 1 error 3
13:57:55.407 MBS: Serial Send: 01 04 00 00 00 0A 70 0D 
13:57:56.401 MBS: Serial Received: 01 F4 00 00 00 00 DC 6A 01 04 14 08 D7 00 00 00 00 00 00 00 00 00 0C 00 00 
13:57:56.404 DMP: 01 F4 00 00 00 00 DC 6A 01 04 14 08 D7 00 00 00 00 00 00 00 00 00 0C 00 00
13:57:56.405 PAC: PzemAc 1 error 3
13:57:56.407 MBS: Serial Send: 01 04 00 00 00 0A 70 0D 
13:57:57.401 MBS: Serial Received: 01 F4 00 00 00 00 DC 6A 01 04 14 08 D7 00 00 00 00 00 00 00 00 00 0C 00 00 
13:57:57.403 DMP: 01 F4 00 00 00 00 DC 6A 01 04 14 08 D7 00 00 00 00 00 00 00 00 00 0C 00 00
13:57:57.405 PAC: PzemAc 1 error 3

### TO REPRODUCE
_Steps to reproduce the behavior:_
Send command 'ModuleAddress 2'

### EXPECTED BEHAVIOUR
PZEM004t device address should be changed to 2 but it doesn't change.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
The problem is here: https://github.com/arendst/Tasmota/blob/master/tasmota/tasmota_xnrg_energy/xnrg_05_pzem_ac.ino#L98
Current code:
PzemAcModbus->Send(0xF8, 0x06, 0x0002, (uint16_t)PzemAc.address);
Possible fix:
uint16_t addr = PzemAc.address;
PzemAcModbus->Send(0xF8, 0x06, 0x0002, 1, (uint16_t *) &addr);
 

**(Please, remember to close the issue when the problem has been addressed)**

smcgann99, 2023-04-23:
I already mentioned this 2 weeks ago [https://github.com/arendst/Tasmota/discussions/18296#discussioncomment-5562087](url)

fellev, 2023-04-23:
Ok, great.
I'm not familiar with the procedure, do you know who can give me permission
to create PR with the fix ?

fellev, 2023-04-23:
Created pull request: https://github.com/arendst/Tasmota/pull/18489

PR #18490 (2023-04-23): Fix for ModuleAddress command
## Description:

**Related issue (if applicable):** fixes #18469
## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [ ] The code change is tested and works with Tasmota core ESP32 V.2.0.8
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).
 [... cut]
closed 2023-04-23 by arendst