arendst/Tasmota #18495: Shelly Duo RGBW can't connect to MQTT broker

Xinayder, 2023-04-24:
### PROBLEM DESCRIPTION
The device cannot connect to the MQTT broker. Fails with return code -2, `MQTT_CONNECT_FAILED`.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Shelly Duo RGBW GU10
- [ ] Tasmota binary firmware version number used: 12.5.0.1
  - [ x ] Pre-compiled
  - [ ] Self-compiled
- [ x ] Flashing tools used: mg2tasmota
- [ x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
{"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"419":"PWM4"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"417":"PWM2"},"GPIO13":{"416":"PWM1"},"GPIO14":{"418":"PWM3"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ x ] Provide the output of this command: `Status 0`:
```lua
20:01:23.738 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_6B0BCF","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
20:00:12.695 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:04:19","StartupUTC":"2023-04-24T18:55:53","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"2023-04-24T19:35:26","SaveCount":53,"SaveAddress":"F9000"}}
20:00:12.729 RSL: STATUS2 = {"StatusFWR":{"Version":"12.5.0.1(tasmota)","BuildDateTime":"2023-04-24T14:20:02","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"375/699"}}
20:00:12.748 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["UPC8853366",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
20:00:12.780 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":633,"Free":368,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15400B","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
20:00:12.814 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-6B0BCF-3023","IPAddress":"192.168.0.157","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"34:94:54:6B:0B:CF","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
20:00:12.838 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"xinayder.me","MqttPort":3881,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6B0BCF","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
20:00:12.860 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-04-24T19:00:12","Local":"2023-04-24T20:00:12","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"05:42","Sunset":"19:54"}}
20:00:12.879 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-04-24T20:00:12"}}
20:00:12.887 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-04-24T20:00:12","Uptime":"0T00:04:19","UptimeSec":259,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":4,"Color":"0B000000","HSBColor":"0,100,4","White":0,"Channel":[4,0,0,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"UPC8853366","BSSId":"34:2C:C4:AA:DD:AD","Channel":6,"Mode":"11n","RSSI":40,"Signal":-80,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ x ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at FA, Count 52
00:00:00.055 QPC: Count 1
00:00:00.058 CFG: CR 375/699, Busy 0
00:00:00.072 ROT: Mode 1
00:00:00.074 SRC: Restart
00:00:00.079 Project tasmota - Tasmota Version 12.5.0.1(tasmota)-2_7_4_9(2023-04-24T14:20:02)
00:00:00.191 WIF: Checking connection...
00:00:00.192 WIF: Attempting connection...
00:00:00.528 WIF: Connecting to AP1 UPC8853366 Channel 6 BSSId 34:2C:C4:AA:DD:AD in mode 11n as tasmota-6B0BCF-3023...
00:00:01.756 WIF: Checking connection...
00:00:01.757 WIF: Connected
00:00:02.010 HTP: Web server active on tasmota-6B0BCF-3023 with IP address 192.168.0.157
00:00:02.407 WIF: Sending Gratuitous ARP
00:00:02.408 NTP: Sync time...
00:00:02.494 WIF: DNS resolved '2.pool.ntp.org' (80.249.164.126) in 84 ms
00:00:02.528 RTC: UTC 2023-04-24T18:55:57, DST 2023-03-26T02:00:00, STD 2023-10-29T03:00:00
19:55:57.000 RTC: Synced by NTP
19:55:57.010 MQT: Attempting connection...
19:55:57.019 WIF: DNS resolved 'xinayder.me' (191.26.70.207) in 7 ms
19:55:57.232 MQT: Connect failed to xinayder.me:3881, rc -2. Retry in 10 sec
19:55:57.241 HTP: Console
19:56:00.455 QPC: Reset
19:56:02.400 APP: Boot Count 14
19:56:02.568 CFG: Saved to flash at F9, Count 53, Bytes 4096
19:56:08.008 MQT: Attempting connection...
19:56:08.010 WIF: DNS resolved 'xinayder.me' (191.26.70.207) in 0 ms
19:56:08.222 MQT: Connect failed to xinayder.me:3881, rc -2. Retry in 20 sec
19:56:15.761 WIF: Checking connection...
19:56:29.010 MQT: Attempting connection...
19:56:29.013 WIF: DNS resolved 'xinayder.me' (191.26.70.207) in 0 ms
19:56:29.224 MQT: Connect failed to xinayder.me:3881, rc -2. Retry in 30 sec
19:56:35.761 WIF: Checking connection...
19:56:54.411 WIF: Sending Gratuitous ARP
19:56:55.754 WIF: Checking connection...
19:57:00.004 MQT: Attempting connection...
19:57:00.136 WIF: DNS resolved 'xinayder.me' (191.26.70.207) in 130 ms
19:57:00.350 MQT: Connect failed to xinayder.me:3881, rc -2. Retry in 40 sec
19:57:15.760 WIF: Checking connection...
19:57:35.762 WIF: Checking connection...
```

### TO REPRODUCE
1. Install Tasmota
2. Setup MQTT

### EXPECTED BEHAVIOUR
Tasmota should be able to connect to the MQTT broker.

### ADDITIONAL CONTEXT
I have recently installed Tasmota on my Shelly Duo RGBW GU10 light bulb using mg2tasmota and using the following template to configure my device: https://github.com/yaourdt/mgos-to-tasmota/pull/31

However, the device won't connect to my MQTT broker, which is up and running and I can easily connect to it via telnet and using mqtt-explorer on my PC and I also have another IoT device at my home, running ESPHome using the same user/password combo and the same hostname and port number.

It doesn't seem to be a DNS issue since it can resolve the hostname to the correct IP address. I tried downgrading to version 12.0.4 but had the same issue.

barbudor, 2023-04-24:
Tasmota cannot connect to your broker. You must have something preventing it like network rules, ACL,...

You are using a non standard port
What else non standard setup are you using on your broker?
Not that Tasmota won't cope with it but you don't give much hints on your global system

Xinayder, 2023-04-24:
I have ACL setup but the user is set to write to all topics. Port 3881 is just the external port which redirects to internal port 1883. I'm using Mosquitto integrated with HASS as my broker.

There are no firewall rules or anything at all, this is pretty much a standard setup if you install Mosquitto on HASS.

I don't think it's an issue with my broker, it might be with Tasmota or the device, since as I mentioned in the OP, everything else can connect to it but this smart light bulb. I might be wrong, but I'm pretty sure if it was an ACL issue it would throw another error instead of -2.

Is there a way I can get logs more detailed than loglevel 4?