arendst/Tasmota #18511: SDM120 Data Jumping about

nobbyneb, 2023-04-26:
Hi, i have a single SDM120 meter connected to a Wemos D1 mini via the widely used TTL to [RS485 convertor](https://www.amazon.co.uk/dp/B07DJ4TGY3?psc=1&ref=ppx_yo2ov_dt_b_product_details) - the data is received by Tasmota and is being parsed via MQTT fine, however the data is erratic and jumps back to '0' for most parameters, every few seconds - the voltage in particular is concerning as this is obviously static (as indicated by the display on the meter itself). I've attached a quick clip.

I am using good quality connection cable, the distance is only around 50cm. I have 5 other SDM120s in a separate (non-Tasmota) setup and the data from them is very stable. I have tried with two different Wemos with the latest Sensors installed and seeing the same behaviour. Is there anything else i should try or is this a known issue?

Many thanks

https://user-images.githubusercontent.com/5697598/234474621-bcca0a4c-0fe7-4111-93cf-85df2921ef8f.MOV

arendst, 2023-05-21:
Closing while no response.

closed 2023-05-21 by arendst

nobbyneb, 2023-05-21:
Apologies i have been away. I am getting the following error, _"NRG: Energy reset by invalid data"_ which coincindes with the appearance of null values. This is completely random, but typically once a minute at least. Screenshot attached. I have replaced the SDM120 Meter and continue to get the same error, so this is isolated to the ESP or RS485 module. I've also attached screenshot of the pin config  I have ordered another modbus reader and will try a different ESP device to see if the issue disappears. Will revert shortly, thank you.
 
[image]

[image]

arendst, 2023-05-22:
> I'm not sure why.

I do. The ESP32 implementation ALWAYS uses hardware serial. The ESP8266 can only use hardware serial on GPIO1/3 or GPIO13/15 resulting in stable comms. ALl other ESP8266 use software serial resulting in occasional misses as you found out.

nobbyneb, 2023-05-22:
That explains it, thank you!

Guido19691, 2025-09-26:
> You need to use a build including support for SDM120, see https://tasmota.github.io/docs/BUILDS/

OK installed the build with sensors, now I have support for the SDMx20. But next strange thing. When I connect the RX/TX of the RS485 converter pcb to GPIO13/15, the D1 mini won't boot up and Tx LED on the RS485 pcb is lit. When I connect to the pins marked Rx/Tx of the D1 mini, now the D1 mini boots up and at least the Tx LED is flashing, but no response from the SDM120. With ESP32 it works (GPIO16/17)....

sfromis, 2025-09-26:
You need to know your pins, and not just connect stuff and pray.

A significant difference between ESP32 and ESP8266 is that with ESP32, you can config gpio pins (including serial) wherever you like due to the multiplexer. Using gpio 16/17 has no special status. With ESP8266, you can only get hardware serial if using the exact right pins, where TX/RX is gpio 1/3 or 15/13 respectively.

With ESP8266, gpio 15 is a strapping pin where boot will fail if the pin is pulled high, and high=inactive is default for UART. Also, gpio 1 is a strapping pin, where boot will fail if pulled low. Thus, you need to be extra careful about what you do with TX.

sfromis, 2025-09-26:
If it works with ESP32, it should also work with ESP8266, as long as you make sure to correctly connect the TX/RX pins, and probably use gpio 1/3 since gpio 15/13 is hit by the strapping issue causing boot failure. Searching the Tasmota [docs](https://tasmota.github.io/docs/) site for "rs485" does give a few hits, but if you already had it working with ESP32, not really much to find there.