arendst/Tasmota #18526: Sonoff B1 after flash can't response

SuiKaSan, 2023-04-27:
### PROBLEM DESCRIPTION
I'm using tasmota v12.5.0.0, core 2_7_4_9 to flash a Sonoff B1 light blub.

I successfully flash the device and completed the WIFI and MQTT setting. Which allow me to get the device's IP and open the ui, but when I change my module to Sonoff B1(26), the ui changed, switch on and off is responding on ui and MQTT but not on device.

Put it this way, everything is working perfectly fine (MQTT, WIFI, UI, etc)  except the device. It won't light up or anything, totally dead.

p.s. I have two Sonoff B1 and it's both working fine before flashing and both dead after flashing, so I don't think it's the device's problem.

Is there something I did lead to the silence of the device?

Thank you very much for review my question!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

04:54:01.380 CMD: Backlog Template; Module; GPIO 255
04:54:01.495 MQT: stat/tasmota_light/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
04:54:01.697 MQT: stat/tasmota_light/RESULT = {"Module":{"26":"Sonoff B1"}}
04:54:01.906 MQT: stat/tasmota_light/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"3008":"MY92x1 DI"},"GPIO13":{"0":"None"},"GPIO14":{"3040":"MY92x1 DCKI"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

04:54:29.647 CMD: Backlog Rule1; Rule2; Rule3
04:54:29.779 MQT: stat/tasmota_light/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
04:54:29.985 MQT: stat/tasmota_light/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
04:54:30.194 MQT: stat/tasmota_light/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

04:55:00.822 CMD: Status 0
04:55:00.830 MQT: stat/tasmota_light/STATUS = {"Status":{"Module":26,"DeviceName":"Tasmota","FriendlyName":["tasmota_light"],"Topic":"tasmota_light","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
04:55:00.836 MQT: stat/tasmota_light/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:56:55","StartupUTC":"2023-04-27T02:58:05","Sleep":50,"CfgHolder":4617,"BootCount":3,"BCResetTime":"2023-04-27T03:56:22","SaveCount":15,"SaveAddress":"F5000"}}
04:55:00.840 MQT: stat/tasmota_light/STATUS2 = {"StatusFWR":{"Version":"12.5.0(tasmota)","BuildDateTime":"2023-04-17T08:03:54","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285N08","CR":"388/699"}}
04:55:00.845 MQT: stat/tasmota_light/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["APTX4869_2.4G",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
04:55:00.856 MQT: stat/tasmota_light/STATUS4 = {"StatusMEM":{"ProgramSize":632,"Free":368,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
04:55:00.864 MQT: stat/tasmota_light/STATUS5 = {"StatusNET":{"Hostname":"tasmota-light-7446","IPAddress":"192.168.31.103","Gateway":"192.168.31.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.31.1","DNSServer2":"0.0.0.0","Mac":"60:01:94:9D:3D:16","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
04:55:00.869 MQT: stat/tasmota_light/STATUS6 = {"StatusMQT":{"MqttHost":"homeassistant","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9D3D16","MqttUser":"test","MqttCount":6,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
04:55:00.875 MQT: stat/tasmota_light/STATUS7 = {"StatusTIM":{"UTC":"2023-04-27T03:55:00","Local":"2023-04-27T04:55:00","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"05:36","Sunset":"19:58"}}
04:55:00.880 MQT: stat/tasmota_light/STATUS10 = {"StatusSNS":{"Time":"2023-04-27T04:55:00"}}
04:55:00.888 MQT: stat/tasmota_light/STATUS11 = {"StatusSTS":{"Time":"2023-04-27T04:55:00","Uptime":"0T00:56:55","UptimeSec":3415,"Heap":18,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":6,"POWER":"ON","Dimmer":10,"Color":"1A1A1A0000","HSBColor":"0,0,10","White":0,"CT":153,"Channel":[10,10,10,0,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"APTX4869_2.4G","BSSId":"D4:DA:21:98:10:E1","Channel":1,"Mode":"11n","RSSI":100,"Signal":-29,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

04:55:23.308 CMD: weblog 4
04:55:23.314 MQT: stat/tasmota_light/RESULT = {"WebLog":4}
04:55:23.870 CFG: Saved to flash at F4, Count 16, Bytes 4096
04:55:31.388 HTP: Console
04:55:33.627 HTP: Main Menu
04:55:36.011 WIF: Checking connection...
04:55:38.943 HTP: Configuration
04:55:40.301 HTP: Main Menu
04:55:56.256 WIF: Checking connection...
04:55:58.370 SRC: WebGui from 192.168.31.180
04:55:58.377 MQT: stat/tasmota_light/RESULT = {"POWER":"OFF"}
04:55:58.380 MQT: stat/tasmota_light/POWER = OFF
04:55:59.114 CFG: Saved to flash at FB, Count 17, Bytes 4096
04:55:59.224 SRC: WebGui from 192.168.31.180
04:55:59.232 MQT: stat/tasmota_light/RESULT = {"POWER":"ON"}
04:55:59.235 MQT: stat/tasmota_light/POWER = ON
04:55:59.589 HTP: Console

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Just flash the device as the instruction said.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
Here is some pic:
[image]
[image]

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2023-04-27:
I'll have a look today.

SuiKaSan, 2023-04-27:
> I'll have a look today.

Thanks! 😄

SuiKaSan, 2023-04-27:
> I'll have a look today.

It's solved!
Someone in Discord named sfromis told me that I have to connect the blub with main power instead of wired setup to light up the bulb. I tried it and works fine! So there's no more problem.
Thank you for your time! 👍

closed 2023-04-27 by SuiKaSan