arendst/Tasmota #18533: VL53L0X show null when some relay is turned up

mefisto22, 2023-04-29:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have one ESP32 with 8 relay and 2 VL53L0X sensor. Every VL53L0X measure without issue until i turning on 1 or more relay. When relay is on, all of VL53L0X measured result is null. I only get back the sensors when turn off all relay and restart ESP. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): ESP32 dev board
- [x ] Tasmota binary firmware version number used: 12.4.0
  - [ ] Pre-compiled
  - [ x] Self-compiled
- [ x] Flashing tools used: tasmota web installer and ota
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
09:06:37.046 MQT: stat/vizoraakna/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
09:06:38.076 MQT: stat/vizoraakna/RESULT = {"GPIO0":{"0":"Nincs"},"GPIO1":{"0":"Nincs"},"GPIO2":{"0":"Nincs"},"GPIO3":{"0":"Nincs"},"GPIO4":{"1312":"DS18x201"},"GPIO5":{"0":"Nincs"},"GPIO6":{"0":"Nincs"},"GPIO7":{"0":"Nincs"},"GPIO8":{"0":"Nincs"},"GPIO9":{"0":"Nincs"},"GPIO10":{"0":"Nincs"},"GPIO11":{"0":"Nincs"},"GPIO12":{"0":"Nincs"},"GPIO13":{"0":"Nincs"},"GPIO14":{"0":"Nincs"},"GPIO15":{"0":"Nincs"},"GPIO16":{"7360":"VL53LXX XSHUT1"},"GPIO17":{"7361":"VL53LXX XSHUT2"},"GPIO18":{"261":"Relé_i6"},"GPIO19":{"262":"Relé_i7"},"GPIO20":{"0":"Nincs"},"GPIO21":{"640":"I2C SDA1"},"GPIO22":{"608":"I2C SCL1"},"GPIO23":{"263":"Relé_i8"},"GPIO24":{"0":"Nincs"},"GPIO25":{"258":"Relé_i3"},"GPIO26":{"256":"Relé_i1"},"GPIO27":{"259":"Relé_i4"},"GPIO32":{"260":"Relé_i5"},"GPIO33":{"257":"Relé_i2"},"GPIO34":{"0":"Nincs"},"GPIO35":{"0":"Nincs"},"GPIO36":{"0":"Nincs"},"GPIO37":{"0":"Nincs"},"GPIO38":{"0":"Nincs"},"GPIO39":{"0":"Nincs"}}

```
- [ x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
09:08:08.761 MQT: stat/vizoraakna/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":145,"Free":366,"Rules":"ON VL53L0X-1#Distance DO Publish stat/vizoraakna/vizszint1 %value% ENDON ON VL53L0X-2#Distance DO Publish stat/vizoraakna/vizszint2 %value% ENDON"}}
09:08:09.074 MQT: stat/vizoraakna/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":64,"Free":447,"Rules":"ON ADS1115#A0 DO Publish stat/vizoraakna/viznyomas %value% ENDON"}}
09:08:09.283 MQT: stat/vizoraakna/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
09:11:57.759 MQT: stat/vizoraakna/STATUS = {"Status":{"Module":1,"DeviceName":"vizoraakna","FriendlyName":["vizoraakna","","","","","","",""],"Topic":"vizoraakna","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
09:11:57.803 MQT: stat/vizoraakna/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:08:27","StartupUTC":"2023-04-29T08:03:30","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"2023-04-28T18:38:43","SaveCount":48}}
09:11:57.845 MQT: stat/vizoraakna/STATUS2 = {"StatusFWR":{"Version":"12.4.0(TasmoCompiler-esp32generic)","BuildDateTime":"2023.04.28 18:53:17","Core":"2_0_6","SDK":"4.4.3.20230129","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"427/699"}}
09:11:57.876 MQT: stat/vizoraakna/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["udvar",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004020","00000000"]}}
09:11:57.908 MQT: stat/vizoraakna/STATUS4 = {"StatusMEM":{"ProgramSize":1240,"Free":1639,"Heap":162,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":"DIO","Features":["0000040E","0FC045CE","00040001","00008043","04000000","C0000881","00000080","00200000","54000008","00000080"],"Drivers":"1,2,4,8,9,10,12,24,26,29,35,38,50,52,62","Sensors":"1,2,5,12,45,127"}}
09:11:57.953 MQT: stat/vizoraakna/STATUS5 = {"StatusNET":{"Hostname":"vizoraakna-8116","IPAddress":"192.168.1.21","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.2","DNSServer2":"192.168.1.1","Mac":"A0:B7:65:F6:3F:B4","IP6Global":"","IP6Local":"fe80::a2b7:65ff:fef6:3fb4","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
09:11:57.995 MQT: stat/vizoraakna/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F63FB4","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
09:11:58.027 MQT: stat/vizoraakna/STATUS7 = {"StatusTIM":{"UTC":"2023-04-29T08:11:58","Local":"2023-04-29T09:11:58","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"05:33","Sunset":"20:01"}}
09:11:58.135 MQT: stat/vizoraakna/STATUS10 = {"StatusSNS":{"Time":"2023-04-29T09:11:58","DS18B20":{"Id":"3C01E076B908","Temperature":11.6},"ADS1115":{"A0":9257,"A1":5890,"A2":5870,"A3":5859},"VL53L0X-1":{"Distance":6.0},"VL53L0X-2":{"Distance":78.9},"TempUnit":"C"}}
09:11:58.167 MQT: stat/vizoraakna/STATUS11 = {"StatusSTS":{"Time":"2023-04-29T09:11:58","Uptime":"0T00:08:28","UptimeSec":508,"Heap":164,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":101,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":32},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","Wifi":{"AP":1,"SSId":"udvar","BSSId":"FC:EC:DA:DA:51:1F","Channel":11,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

09:19:59.965 MQT: tele/vizoraakna/STATE = {"Time":"2023-04-29T09:19:59","Uptime":"0T00:01:28","UptimeSec":88,"Heap":156,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":92,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":35},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","Wifi":{"AP":1,"SSId":"udvar","BSSId":"FC:EC:DA:DA:51:1F","Channel":11,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:03"}}
09:20:00.105 MQT: tele/vizoraakna/SENSOR = {"Time":"2023-04-29T09:20:00","DS18B20":{"Id":"3C01E076B908","Temperature":11.4},"ADS1115":{"A0":9210,"A1":5868,"A2":5891,"A3":5879},"VL53L0X-1":{"Distance":6.3},"VL53L0X-2":{"Distance":76.0},"TempUnit":"C"}
09:20:06.330 SRC: WebGui from 10.0.0.1
09:20:06.340 MQT: stat/vizoraakna/RESULT = {"POWER8":"ON"}
09:20:06.344 MQT: stat/vizoraakna/POWER8 = ON
09:20:10.836 CFG: Saved, Szám 54, Byte-ok 4096
09:20:23.340 BRY: GC from 5494 to 3416 bytes, objects freed 2/35 (in 1 ms) - slots from 40/91 to 37/91
09:20:23.350 MQT: tele/vizoraakna/STATE = {"Time":"2023-04-29T09:20:23","Uptime":"0T00:01:52","UptimeSec":112,"Heap":162,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":1274,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":35},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"ON","Wifi":{"AP":1,"SSId":"udvar","BSSId":"FC:EC:DA:DA:51:1F","Channel":11,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:03"}}
[... cut ...]
_Steps to reproduce the behavior:_
Connect two VL53L0X sensor to ESP32 and relays, and turn on relay.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Relays turn on, and VL53L0X measure like before relays turned on.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the pr
[image]
oblem here._
[image]

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-04-29:
Wiring or power supply problem
Not a Tasmota issue

mefisto22, 2023-04-29:
I measured 5V and 3.3V everywhere, all voltage is OK. I measured cables with multimeter, all cable is OK. Every connection is soldered.

barbudor, 2023-04-29:
Disconnect physically the relays and tell me