arendst/Tasmota #18543: SML + HASS + auto discovery: all values do not carry unit

TheChatty, 2023-04-30:
### PROBLEM DESCRIPTION
I've got a few Tasmota devices autodiscovered by Home Assistant (from integration [Tasmota](https://www.home-assistant.io/integrations/tasmota) and [fritz](https://www.home-assistant.io/integrations/fritz)).

The power plug measurements are **shown in HA with their correct unit** (e.g. kWh):
[image] [image]

My house meters are read by SML and they **don't carry any unit in HA**:
[image] [image]

 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): GoSund SP1 and ESP8266 DevBoard
- [x] Tasmota binary firmware version number used: 12.x
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: esptool.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- configure SML script including correct units
- have device autodiscovered by Home Assistant
- observe all values carry no unit

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
- all values carry correct unit

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
See problem description.

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-04-30:
Why would it be a Tasmota problem?

TheChatty, 2023-04-30:
Because non-SML devices are working properly and SML devices do not. Thus I think I made something wrong with Tasmota.

arendst, 2023-04-30:
No you didn't. The issue is that hatasmota only adds units to predefined names like power, voltage and energy. Your SML names are unknown to hatasmota and therefor you'll need to find a way to tell hass what unit they need.

sfromis, 2023-04-30:
And Tasmota will only automatically include a unit indication in the payloads in cases where you have a configuration option to select which unit you want to use, like C/F or hPa/mmHg.

TheChatty, 2023-04-30:
A **workaround** is to define a template sensor like this:
```
template:
  - sensor:
      - name: "Stromzähler Verbrauch ganzes Haus"
        unique_id: "StromzHaus"
        unit_of_measurement: 'kWh'
        device_class: "energy"
        state_class: "total_increasing"
        state: >-
            {{ float(states('sensor.energiekonsum_strom_e_in')) | round(3) }}
```

But this causes unnecessary data duplication (if complete sensor domain is included in InfluxDB) and does not provide historical values (from real sensor to template sensor).

@arendst As you said the problem is within the sensor.py to only add units to predefined names. There's already an [old hatasmota issue](https://github.com/emontnemery/hatasmota/issues/130) asking for improvement.

@sfromis How can I configure to include my units in the payload?

sfromis, 2023-04-30:
There is no option to add unit specifications to the standard SENSOR payloads. But if you generate your own JSON payloads, you can put whatever there, both from rules, Berry or scripting. Anyway, if hatasmota is to recognize the relation between added unit types (similar to existing units) and specific fields, it would still need modification to recognize the specific fields names. Or a quite different type of payload structure to indicate what field has what unit.

TheChatty, 2023-04-30:
Do you have an example payload containing unit specifications?

sfromis, 2023-04-30:
`{"Time":"2023-04-30T16:48:30.417+02:00","BME280":{"Temperature":28.29,"Humidity":29.08,"DewPoint":8.57,"Pressure":1022.68,"SeaPressure":1024.65},"BH1750":{"Illuminance":29},"PressureUnit":"hPa","TempUnit":"C"}`

Nothing in the JSON to establish that "TempUnit" is for "Temperature", and "PressureUnit" is for "Pressure" and "SeaPressure". Hence just adding further units in the same location is a waste of time, without the relationship being established elsewhere, and still does not make much sense if the unit is invariably "KWh".

As long as the payload format does not separately include a unit specification for each field, the backend (like hatasmota) needs metadata to derive units, no matter if the payload does have units already without these being linked to specific fields.

TheChatty, 2023-04-30:
I'm using SML which provides arbitrary units (not invariably "KWh").

The way to go is like this (as I understood it):
- enhance hatasmota to evaluate a new kind of JSON structure including units for measurements (Tasmota would send custom JSON payload in the meantime)
- enhance Tasmota to include units in JSON payload by default