arendst/Tasmota #18551: Tasmota 8.1.0 says not enough space even when using minimal

ardenking, 2023-05-01:
I've sent the day updating my 30 sonoffs all went fine except 2 that are on 8.1.0 

One is is T1 and the other 4ch sonoff 

when I try manual update to minimal get not enough space message 

All I can see in terminal is this 

16:47:52 UPL: File tasmota-minimal (1).bin ...
16:47:55 MQT: Attempting connection...
16:47:55 MQT: Connected

It would be a PITA to take these down for manual flashing any ideas ?

<html>
<body>

ESP Chip Id | 4785657
-- | --
0x144051
1024kB
4096kB
566kB
436kB
26kB

</body>
</html>

sfromis, 2023-05-01:
What version minimal are you trying to use? It should be from 8.1.0, instead of current. And do remember that you have to follow the gradual upgrade steps.

ardenking, 2023-05-01:
> What version minimal are you trying to use? It should be from 8.1.0, instead of current. And do remember that you have to follow the gradual upgrade steps.

yes its the 8.10 minimal also tried older and newer files as I cant get past this stage the upgrade path is an issue atm

sfromis, 2023-05-01:
Your program size of 566kB means that there is ample room for minimal, if you do it right. Try removing that " (1)" from the file name, blanks may be a troublemaker. The message "not enough space" only means "some error", not specifically about space.

ardenking, 2023-05-01:
thanks for the suggestion but that didn't make a difference the not enough space comes back really quickly too 

I've noticed since trying the update the switches are beaconing the access point even though they are also on my wifi

sfromis, 2023-05-01:
When downloaded the minimal file, on risk factor is not doing it the right way. Especially from Github, that can very easily happen if you do something right-click, and save as. More generally, exact details of the full flow from start to error matters a lot, and you hardly gave any info, outside of a couple of details.