arendst/Tasmota #18589: Touch screen does not work on haspmota with sunton EPS32-8048S050

Be-Virtual, 2023-05-06:
### Discussed in https://github.com/arendst/Tasmota/discussions/18565

<div type='discussions-op-text'>

<sup>Originally posted by **Be-Virtual** May  3, 2023</sup>

### PROBLEM DESCRIPTION
Touch screen does not work on haspmota with Sunton EPS32-8048S050 5 inch screen
Same screen works OK on openhasp
Suppose autoconf file has an error 
See below: To reproduce

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sunton ESP32-8048S050
- [X] Tasmota binary firmware version number used: 12.5.0.1
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: web
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
------------------ TASMOTA / HASPMOTA ----------------------------------
-
Installed Tasmota with unofficial firmware 'Tasmota32-S3 (QIO Flash/OPI PSRAM)

Used the autoconf to set board to Senton 8048S050.

----- Info ------
-
Content of Display.ini (from SunTon-S3-Display.autoconf):
:H,ST7262,800,480,16,RGB,40,41,39,42,2,15,16,4,45,48,47,21,14,8,3,46,9,1,5,6,7,14
:S,2,1,1,0,40,20
:V,0,8,4,8,0,8,4,8,1
:0,00
:1,01
:2,02
:3,03
:TI1,5d,,,-1,38

----- Info ------
-
Content of init.bat (from SunTon-S3-Display.autoconf):
Template {"NAME":"SUNTON","GPIO":[1,1,1,1,1,1,1,1,1,1,6720,704,736,672,1,1,1,1,1,640,608,1,0,0,0,0,0,6210,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":1}
Module 0
DisplayDimmer 70

Tested with the demo from Tasmota site:
UrlFetch https://raw.githubusercontent.com/arendst/Tasmota/development/tasmota/berry/haspmota/haspmota_demo.tapp

Touch screen (capacitive) does not react

If I go to the berry console i can switch pages with p2.show() 

Extra info autoexec.be 
import lv_tasmota_log
import lv_tasmota_info
import lv_wifi_graph

import haspmota
haspmota.start ()

This is the touych config: 
:TI1,5d,,,-1,38

----- To verify screen is OK ------
-
Installed the nightly openhasp to test if my screen was ok
Basic install from nightly, then used this :
// ------------------------------------- Screen 480 X 320 w=(159, 160, 159) -------------------------------------
{"page":0,"id":1,"obj":"btn","action":{"down": "page prev"},"x":0,"y":290,"w":159,"h":32,"bg_color":"#2C3E50","text":"\uE141","text_color":"#FFFFFF","radius":0,"border_side":0,"text_font":24}
{"page":0,"id":2,"obj":"btn","action":{"down": "page back"},"x":160,"y":290,"w":160,"h":32,"bg_color":"#2C3E50","text":"\uE2DC","text_color":"#FFFFFF","radius":0,"border_side":0,"text_font":24}
{"page":0,"id":3,"obj":"btn","action":{"down": "page next"},"x":321,"y":290,"w":159,"h":32,"bg_color":"#2C3E50","text":"\uE142","text_color":"#FFFFFF","radius":0,"border_side":0,"text_font":24}
{"page":1,"comment":" ---- Basic Page 1 ----"}

{"page":1,"id":10,"w":100,"obj":"label","text": "%hostname%","align":0,"text_font":16,"text_color":"white"}
{"page":1,"id":11,"obj":"label","x":100,"y":0,"w":100,"h":35,"text":"%ssid%","align":1,"text_font":16,"text_color":"white"}
{"page":1,"id":12,"obj":"label","x":200,"y":0,"w":120,"h":35,"text":"%ip%","align":2,"text_font":16,"text_color":"white"}

Touch screen (capacitive) reacted directly, switch from touchscreen with navigation keys defined
-

### EXPECTED BEHAVIOUR
_
Correct the display.txt file xwih correct definition of touch screen

### SCREENSHOTS

### ADDITIONAL CONTEXT
_

Seems touch screen is
TAMC_GT911 ts = TAMC_GT911(TOUCH_SDA, TOUCH_SCL, TOUCH_INT, TOUCH_RST, max(TOUCH_MAP_X1, TOUCH_MAP_X2), max(TOUCH_MAP_Y1, TOUCH_MAP_Y2)); 

----- Info ------
-
 arduino definitions for this Sunton 8048S050:
#include <Arduino_GFX_Library.h>
#include "BmpClass.h"
#include "Arduino.h"
#include "Audio.h"
#include "SPI.h"
#include "SD.h"
#include "FS.h"
#include <TAMC_GT911.h>

#define BMP_FILENAME "/logo.bmp"

#define TFT_BL 2

#define SD_SCK 12
#define SD_MISO 13
#define SD_MOSI 11
#define SD_CS 10

#define I2S_DOUT 17
#define I2S_BCLK 19
#define I2S_LRC 18

#define TOUCH_SCL 20
#define TOUCH_SDA 19
#define TOUCH_INT -1
#define TOUCH_RST 38
#define TOUCH_ROTATION ROTATION_NORMAL

#define TOUCH_MAP_X1 0
#define TOUCH_MAP_X2 800
#define TOUCH_MAP_Y1 480
#define TOUCH_MAP_Y2 0

**(Please, remember to close the issue when the problem has been addressed)**
</div>

Jason2866, 2023-05-06:
The Autoconf is for the Display https://www.makerfabs.com/sunton-esp32-s3-4-3-inch-ips-with-touch.html
Yours is a different one. Probably not just the screen size is different.

Jason2866, 2023-05-06:
Have you defined the touch SDA SCL GPIOS in Tasmota Template?
with this settings used in Open Hasp
```
#define TOUCH_SCL 20
#define TOUCH_SDA 19
```

Be-Virtual, 2023-05-06:
Yes, here a copy of the template
[image]

Be-Virtual, 2023-05-06:
Touch model seems to be  OK:
00:00:00.802 I2C: GT911 found at 0x5d
00:00:01.395 LVG: LVGL initialized
00:00:02.161 DSP: Configured display 'ST7262'
I can write to the screen but no touch at all, only error is 00:00:00.120 TFS: File '.drvset003' not found
Not sure it is related

Be-Virtual, 2023-05-06:
> Have you defined the touch SDA SCL GPIOS in Tasmota Template? with this settings used in Open Hasp
> 
> ```
> #define TOUCH_SCL 20
> #define TOUCH_SDA 19
> ```

To be precise, this is not from openhasp but from arduino definitions

Be-Virtual, 2023-05-06:
Do you have any idea where the openhasp config for the screens/touch are stored on github. 
Francis made different files for each of the sunton screens but cannot find where the display/touch is configured

Jason2866, 2023-05-06:
@gemu2015 Tried latest Tasmota version with the sunton 4.3 inch rgb display with the GT911 touch controller.
Get this error at startup
```
00:00:00.000 HDW: ESP32-S3 (PSRAM)
00:00:00.011 UFS: FlashFS mounted with 10284 kB free
00:00:00.017 CFG: Loaded from File, Count 6
00:00:00.018 FRC: Some settings have been reset (2)
00:00:00.020 SPI: Hardware using GPIO12(CLK), GPIO11(MOSI) and GPIO13(MISO)
00:00:00.033 TFS: File 'mcp23x.dat' not found
00:00:00.073 BRY: Berry initialized, RAM used=5354 bytes
00:00:00.086 TFS: File '.drvset003' not found
00:00:00.087 CFG: Energy use defaults as file system not ready or file not found
00:00:00.695 Project sonoff - Tasmota Version 12.5.0.2(lvgl-haspmota)-2_0_9(2023-05-06T14:23:56)
00:00:00.751 I2C: GT911 found at 0x5d
00:00:01.834 TI: GT911 failed
00:00:02.464 RSL: RESULT = {"POWER":"ON"}
00:00:02.465 RSL: POWER = ON
00:00:02.469 LVG: LVGL initialized
00:00:03.235 DSP: Configured display 'ST7262'
00:00:07.079 WIF: Connecting to AP1 Jason_Home_WLAN Channel 9 BSSId 8C:53:C3:85:17:96 in mode 11n as sonoff-E12400-1024...
00:00:09.513 WIF: Connected
00:00:09.760 HTP: Web server active on sonoff-E12400-1024 with IP address 192.168.2.104
14:35:56.346 MQT: Attempting connection...
14:35:56.358 MQT: Connected
14:35:56.360 MQT: tele/sonoff-E12400/LWT = Online (retained)
14:35:56.362 MQT: cmnd/sonoff-E12400/POWER = 
14:35:56.366 MQT: tele/sonoff-E12400/INFO1 = {"Info1":{"Module":"SUNTON","Version":"12.5.0.2(lvgl-haspmota)","FallbackTopic":"cmnd/sonoff-E12400_fb/","GroupTopic":"cmnd/sonoffs/"}}
14:35:56.378 MQT: tele/sonoff-E12400/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"sonoff-E12400-1024","IPAddress":"192.168.2.104","IP6Global":"fd4d:d9a8:7683::f612:faff:fee1:2400","IP6Local":"fe80::f612:faff:fee1:2400"}}
14:35:56.402 MQT: tele/sonoff-E12400/INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":3}}
14:35:56.415 MQT: stat/sonoff-E12400/RESULT = {"POWER":"ON"}
14:35:56.418 MQT: stat/sonoff-E12400/POWER = ON
```

So the touch controller can not work.

gemu2015, 2023-05-06:
i just reflashed with latest dev and got no error, touch working

however self compiled no BERRY no LVGL but  don't think that this is the reason 
 
can not find an obvious error in the code.

`:TI1,5d,*,*,-1,38`

Jason2866, 2023-05-06:
@gemu2015  Weird. Which compile flags do you use in platformio.ini / user_config_override.h ?
I dont manage to compile with rules.

gemu2015, 2023-05-06:
@Jason2866 
among others these are the relevant ones i think 

#undef USE_BERRY
#undef USE_LVGL
#define USE_SCRIPT
#define USE_SPI
#define USE_DISPLAY
#define USE_UNIVERSAL_DISPLAY
#define USE_GT911
#define USE_SDCARD
#define USE_TOUCH_BUTTONS

Jason2866, 2023-05-06:
Nope, still the same compile error.

gemu2015, 2023-05-06:
very strange. i created a new setup exactly with the defines above and no compile error ?
renderer is declared in xdrv13_display.ino und should be known.
probably another strange platform io error

try to insert
extern Renderer *renderer;
in file xdrv55_touch.ino

gemu2015, 2023-05-06:
@Jason2866 
i now compiled with berry and lvgl and touch works as expected

gemu2015, 2023-05-06:
just a remark, there are 2 messages related to GT911
1. check if i2c address responds
I2C: GT911 found at 0x5d
2. 
initialize GT911 gives either
TI: GT911
or
TI: GT911 failed

if seconds message is missing GT911 is not compiled into binary

Jason2866, 2023-05-06:
From the OP the `TI: GT911` message is missing. So the driver seems missing.
I always get `TI: GT911 failed` in my tries when building with LVGL haspmota.
LVGL and haspmota does work. "Only" the touch driver is not working -> `TI: GT911 failed`

Jason2866, 2023-05-06:
@Be-Virtual  In unofficial build the driver for GT911 was missing. Just added now.

Be-Virtual, 2023-05-06:
available from the https://tasmota.github.io/install/ or do I have to download a specific bin ?

Jason2866, 2023-05-06:
Yep, latest build is now available from there.

Be-Virtual, 2023-05-06:
Now I have the error on touch:
00:00:00.000-271/15 CMD: Using USB CDC
00:00:00.001-271/15 HDW: ESP32-S3 (PSRAM)
00:00:00.052-269/14 UFS: FlashFS mounted with 12572 kB free
00:00:00.071 CFG: Loaded from File, Count 16
00:00:00.077 QPC: Count 1
00:00:00.079 SPI: Hardware using GPIO12(CLK), GPIO11(MOSI) and GPIO13(MISO)
00:00:00.084 TFS: File 'mcp23x.dat' not found
00:00:00.236 BRY: Berry initialized, RAM used=5386 bytes
00:00:00.270 TFS: File '.drvset003' not found
00:00:00.270 CFG: Energy use defaults as file system not ready or file not found
00:00:00.882 Project tasmota - Tasmota Version 12.5.0.2(lvgl-haspmota)-2_0_8(2023-05-06T16:36:13)
00:00:00.995 I2C: GT911 found at 0x5d
00:00:02.018 TI: GT911 failed
00:00:02.649 LVG: LVGL initialized
00:00:03.415 DSP: Configured display 'ST7262'
00:00:04.442 BRY: Successfully loaded 'autoexec.be'

Be-Virtual, 2023-05-06:
> just a remark, there are 2 messages related to GT911
> 
>     1. check if i2c address responds
>        I2C: GT911 found at 0x5d
> 
> 
> initialize GT911 gives either TI: GT911 or TI: GT911 failed
> 
> if seconds message is missing GT911 is not compiled into binary

gemu2015, 2023-05-07:
ok, this is my console log, and attached the binary i use, just for testing if the error disappears 

```
00:00:00.001 HDW: ESP32-S3 (PSRAM)
00:00:00.028 UFS: FlashFS mounted with 10288 kB free
00:00:00.042 CFG: Loaded from File, Anzahl 242
00:00:00.047 QPC: Count 1
00:00:00.048 SPI: Hardware using GPIO12(CLK), GPIO11(MOSI) and GPIO13(MISO)
00:00:00.073 I2C: GT911 found at 0x5d
00:00:00.192 TI: GT911
00:00:00.835 DSP: Configured display 'ST7262'
00:00:05.393 Projekt tasmota - sunton Version 12.5.0.2(tasmota)-2_0_6(2023.05.06 17:01:46)
00:00:05.393 UFILESYSTEM OK!
00:00:06.035 WIF: verbinden mit AP1 WLAN-7490 Channel 1 BSSId 50:E6:36:24:8C:EF in Modus 11n als sunton...
00:00:07.935 WIF: verbunden
00:00:08.188 HTP: Web-Server aktiv bei sunton mit IP-Adresse 192.168.188.104
00:00:08.189 RSL: INFO1 = {"Info1":{"Module":"SUNTON","Version":"12.5.0.2(tasmota)","FallbackTopic":"cmnd/sunton_fb/","GroupTopic":"cmnd/tasmotas/"}}
00:00:08.200 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"sunton","IPAddress":"192.168.188.104","IP6Global":"fd12:100f:5027:4b8a:f612:faff:fee1:22d4","IP6Local":"fe80::f612:faff:fee1:22d4"}}
00:00:08.221 RSL: INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":95}}
00:00:08.232 RSL: RESULT = {"POWER":"ON"}
00:00:08.233 RSL: POWER = ON
08:43:51.567 QPC: Reset
```
[tasmota32s3-sunton.bin.zip](https://github.com/arendst/Tasmota/files/11414144/tasmota32s3-sunton.bin.zip)

Be-Virtual, 2023-05-07:
I flashed your bin into my Sunton ESP32-80148S050, reconfigured network  and restarted.

It seems another version of tasmota without berry and the autoconf for lvgl

I copied my init.bat
-
Template {"NAME":"SUNTON","GPIO":[1,1,1,1,1,1,1,1,1,1,6720,704,736,672,1,1,1,1,1,640,608,1,0,0,0,0,0,6210,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":1}
Module 0
DisplayDimmer 70

my display.ini
-
:H,ST7262,800,480,16,RGB,40,41,39,42,2,15,16,4,45,48,47,21,14,8,3,46,9,1,5,6,7,14
:S,2,1,1,0,40,20
:V,0,8,4,8,0,8,4,8,1
:0,00
:1,01
:2,02
:3,03
:TI1,5d,*,*,-1,38
#

and my autoexec.be
-

if !tasmota.memory().contains("psram")
    print("HSP: Error: OpenHASP demo requires PSRAM")
    return
end

import lv_tasmota_log
import lv_tasmota_info
import lv_wifi_graph

import haspmota
haspmota.start()

Result
-
No error on touch screen but also nothing on the screen.
Not even the blue screen with tasmota (showing the screen is working)

Will try with only lvgl code without haspmota to see if screen start up

here my console log
-
00:00:00.001 HDW: ESP32-S3 (PSRAM)
00:00:00.014 UFS: FlashFS mounted with 12588 kB free
00:00:00.022 CFG: Loaded from File, Anzahl 39
00:00:00.028 QPC: Count 1
00:00:00.029 SPI: Hardware using GPIO12(CLK), GPIO11(MOSI) and GPIO13(MISO)
00:00:00.041 I2C: GT911 found at 0x5d
00:00:00.160 TI: GT911
00:00:00.789 DSP: Configured display 'ST7262'
00:00:01.365 Projekt tasmota - sunton Version 12.5.0.2(tasmota)-2_0_6(2023.05.07 09:06:00)
00:00:01.365 UFILESYSTEM OK!
00:00:02.035 WIF: verbinden mit AP1 Wifi208 Channel 1 BSSId 68:D7:9A:48:D2:43 in Modus 11n als sonoff-4776...
00:00:03.892 WIF: verbunden
11:56:30.003 HTP: Web-Server aktiv bei sonoff-4776 mit IP-Adresse 192.168.254.135
11:56:30.143 MQT: Verbindungsversuch...
11:56:31.658 MQT: verbunden
11:56:31.660 MQT: tele/sonoff/LWT = Online (beibehalten)
11:56:31.661 MQT: cmnd/sonoff/POWER = 
11:56:31.664 MQT: tele/sonoff/INFO1 = {"Info1":{"Module":"SUNTON","Version":"12.5.0.2(tasmota)","FallbackTopic":"cmnd/sunton_fb/","GroupTopic":"cmnd/tasmotas/"}}
11:56:31.675 MQT: tele/sonoff/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"sonoff-4776","IPAddress":"192.168.254.135","IP6Global":"","IP6Local":"fe80::f612:faff:fee1:d2a8"}}
11:56:31.697 MQT: tele/sonoff/INFO3 = {"Info3":{"RestartReason":"RTC Watch dog reset digital core and rtc module","BootCount":16}}
11:56:31.709 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
11:56:31.709 MQT: stat/sonoff/POWER = ON

Be-Virtual, 2023-05-07:
Sorry was closed by error (wrong doing) STILL OPEN

gemu2015, 2023-05-07:
ok, this proves that something is wrong with the tasmota provided binary.
currently i have no guess what is wrong there

i may compile a version with berry + lvgl and autoconf for you

but we have to find out what goes wrong in @Jason2866s build

gemu2015, 2023-05-07:
here the bin with berry, lvgl and autoconf

[tasmota32s3-sunton.bin.zip](https://github.com/arendst/Tasmota/files/11414728/tasmota32s3-sunton.bin.zip)

Be-Virtual, 2023-05-07:
I see some diff in our files !

Jason
-
00:00:00.695 Project sonoff - Tasmota Version 12.5.0.2(lvgl-haspmota)-2_0_9(2023-05-06T14:23:56)

Be-Virtual
-
bin from gemu2015 on my EPS32-8048S050
00:00:01.365 Projekt tasmota - sunton Version 12.5.0.2(tasmota)-2_0_6(2023.05.07 09:06:00)
From new install with latest unofficial build
00:00:02.483 Project tasmota - Sunton Version 12.5.0.2(lvgl-haspmota)-2_0_8(2023-05-06T16:36:13)

@Jason2866 where did your get the 2_0_9 version as I have with latest build only 2_0_8

Be-Virtual, 2023-05-07:
Thanks gemu2015, will try it now and come back

Jason2866, 2023-05-07:
@Be-Virtual I do the the framework builds for Tasmota. Tasmota core 2.0.9 is ahead. There are just minor changes like in official espressif core from 2.0.8. to 2.0.9

Jason2866, 2023-05-07:
@gemu2015  Same error with your firmware
```
00:00:00.001 HDW: ESP32-S3 (PSRAM)
00:00:00.022 UFS: FlashFS mounted with 10284 kB free
00:00:00.032 CFG: Loaded from File, Anzahl 20
00:00:00.033 FRC: Some settings have been reset (2)
00:00:00.035 SPI: Hardware using GPIO12(CLK), GPIO11(MOSI) and GPIO13(MISO)
00:00:00.103 BRY: Berry initialized, RAM used=4764 bytes
00:00:00.732 Projekt tasmota - Sonoff Version 12.5.0.2(tasmota)-2_0_6(2023.05.07 12:18:05)
00:00:00.733 UFILESYSTEM OK!
00:00:00.793 I2C: GT911 found at 0x5d
00:00:01.861 TI: GT911 failed
00:00:02.491 RSL: RESULT = {"POWER":"ON"}
00:00:02.491 RSL: POWER = ON
00:00:02.495 LVG: LVGL initialized
00:00:03.260 DSP: Configured display 'ST7262'
```

gemu2015, 2023-05-07:
Ok this latest build was without my undefines ( I exclude nearly everything)
So I guess an incompatibility with one of these additional drivers

gemu2015, 2023-05-07:
i guess i found the reason, please try this

[tasmota32s3-sunton.bin.zip](https://github.com/arendst/Tasmota/files/11414901/tasmota32s3-sunton.bin.zip)

Be-Virtual, 2023-05-07:
I just retried with you first bin (this morning) that gave the touch working.
When I configure the template and activate the module 0. it restarts and the touch fails

Just also tested your last bin

does not start!
Here logs
-
00:00:00.000 HDW: ESP32-S3 (PSRAM)
00:00:00.024 UFS: FlashFS mounted with 12592 kB free
00:00:00.035 CFG: Loaded from File, Anzahl 14
00:00:00.041 QPC: Count 1
00:00:00.042 SPI: Hardware using GPIO12(CLK), GPIO11(MOSI) and GPIO13(MISO)
00:00:00.061 I2C: GT911 found at 0x5d
00:00:01.122 TI: GT911 failed
00:00:01.751 DSP: Configured display 'ST7262'
00:00:02.357 Projekt tasmota - sunton Version 12.5.0.2(tasmota)-2_0_6(2023.05.07 09:06:00)
00:00:02.357 UFILESYSTEM OK!
00:00:02.993 WIF: verbinden mit AP1 Wifi208 Channel 1 BSSId 68:D7:9A:48:D2:43 in Modus 11n als sonoff-4776...
00:00:03.885 WIF: verbunden
00:00:04.136 HTP: Web-Server aktiv bei sonoff-4776 mit IP-Adresse 192.168.254.135
14:09:49.158 MQT: Verbindungsversuch...
14:09:50.161 MQT: Verbindung fehlgeschlagen aufgrund von 192.168.188.49:1883, rc -2. Wiederversuch in 10 s
14:09:53.373 QPC: Reset
14:10:00.942 MQT: Verbindungsversuch...
14:10:01.944 MQT: Verbindung fehlgeschlagen aufgrund von 192.168.188.49:1883, rc -2. Wiederversuch in 20 s
14:10:24.418 MQT: Verbindungsversuch...
14:10:25.421 MQT: Verbindung fehlgeschlagen aufgrund von 192.168.188.49:1883, rc -2. Wiederversuch in 30 s
14:10:59.401 MQT: Verbindungsversuch...
14:11:00.404 MQT: Verbindung fehlgeschlagen aufgrund von 192.168.188.49:1883, rc -2. Wiederversuch in 40 s
14:11:41.156 MQT: Verbindungsversuch...
14:11:42.159 MQT: Verbindung fehlgeschlagen aufgrund von 192.168.188.49:1883, rc -2. Wiederversuch in 50 s
��ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420d1f72
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x128
load:0x403c9700,len:0xb9c
load:0x403cc700,len:0x2878
SHA-256 comparison failed:
Calculated: f25588c3c9da88bbe2740fe2dd8052ee5b34ff2a7d7b8b7003620f8929865764
Expected: a2833441682eed78e52a0f7161a4c6a94e9b0ac69a1390bc1e021f53ed6d5a97
Attempting to boot anyway...
entry 0x403c98bc

Be-Virtual, 2023-05-07:
after full restart only via web installer console:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x128
load:0x403c9700,len:0xb9c
load:0x403cc700,len:0x2878
SHA-256 comparison failed:
Calculated: f25588c3c9da88bbe2740fe2dd8052ee5b34ff2a7d7b8b7003620f8929865764
Expected: a2833441682eed78e52a0f7161a4c6a94e9b0ac69a1390bc1e021f53ed6d5a97
Attempting to boot anyway...
entry 0x403c98bc

gemu2015, 2023-05-07:
i just downloaded my last binary from above and flashed

with this result

```
00:00:00.001 HDW: ESP32-S3 (PSRAM)
00:00:00.042 UFS: FlashFS mounted with 10288 kB free
00:00:00.059 CFG: Loaded from File, Anzahl 262
00:00:00.060 FRC: Some settings have been reset (3)
00:00:00.061 SPI: Hardware using GPIO12(CLK), GPIO11(MOSI) and GPIO13(MISO)
00:00:00.167 BRY: Berry initialized, RAM used=4729 bytes
00:00:00.209 I2C: GT911 found at 0x5d
00:00:00.328 TI: GT911
00:00:00.971 DSP: Configured display 'ST7262'
00:00:00.983 UFS: SDCard mounted (SPI mode) with 15555040 kB free
00:00:00.986 Projekt tasmota - sunton Version 12.5.0.2(tasmota)-2_0_6(2023.05.07 14:00:51)
00:00:00.996 UFILESYSTEM OK!
00:00:00.022 LVG: LVGL initialized
00:00:02.412 WIF: verbinden mit AP1 WLAN-7490 Channel 1 BSSId 50:E6:36:24:8C:EF in Modus 11n als sunton...
00:00:04.290 WIF: verbunden
00:00:04.544 HTP: Web-Server aktiv bei sunton mit IP-Adresse 192.168.188.104
00:00:04.545 RSL: INFO1 = {"Info1":{"Module":"SUNTON","Version":"12.5.0.2(tasmota)","FallbackTopic":"cmnd/sunton_fb/","GroupTopic":"cmnd/tasmotas/"}}
00:00:04.557 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"sunton","IPAddress":"192.168.188.104","IP6Global":"","IP6Local":"fe80::f612:faff:fee1:22d4"}}
00:00:04.569 RSL: INFO3 = {"Info3":{"RestartReason":{"Exception":29,"Reason":"StoreProhibited","EPC":"40377ef5","EXCVADDR":"00000000","CallChain":["40377ef2","4037d55d","40383da1","420a2acb","4207cf3d","42078ee1","4206b259","4208a8f2","42094d74","4201b842"]},"BootCount":104}}
00:00:04.601 RSL: RESULT = {"POWER":"ON"}
00:00:04.602 RSL: POWER = ON
14:19:15.924 QPC: Reset
```

Jason2866, 2023-05-07:
@gemu2015  Your last uploaded firmware does int the GT911 👍 
```
00:00:00.001 HDW: ESP32-S3 (PSRAM)
00:00:00.048 UFS: FlashFS mounted with 10284 kB free
00:00:00.066 CFG: Loaded from File, Anzahl 16
00:00:00.068 FRC: Some settings have been reset (2)
00:00:00.069 SPI: Hardware using GPIO12(CLK), GPIO11(MOSI) and GPIO13(MISO)
00:00:00.192 BRY: Berry initialized, RAM used=4764 bytes
00:00:00.255 I2C: GT911 found at 0x5d
00:00:00.374 TI: GT911
00:00:00.003 DSP: Configured display 'ST7262'
00:00:01.543 Projekt tasmota - Sonoff Version 12.5.0.2(tasmota)-2_0_6(2023.05.07 14:00:51)
00:00:01.544 UFILESYSTEM OK!
00:00:01.549 RSL: RESULT = {"POWER":"ON"}
00:00:01.549 RSL: POWER = ON
00:00:01.553 LVG: LVGL initialized
```

gemu2015, 2023-05-07:
ok, reason was double initialization of i2c, will do a pr

Be-Virtual, 2023-05-07:
I reinstalled everything and now I got
# Before configuring display and template
00:00:00.001 HDW: ESP32-S3 (PSRAM)
00:00:00.029 UFS: FlashFS mounted with 12592 kB free
00:00:00.045 CFG: Loaded from File, Anzahl 11
00:00:00.050 QPC: Count 1
00:00:00.124 BRY: Berry initialized, RAM used=4729 bytes
00:00:00.132 Projekt tasmota - Sonoff Version 12.5.0.2(tasmota)-2_0_6(2023.05.07 14:00:51)
00:00:00.132 UFILESYSTEM OK!
00:00:00.843 WIF: verbinden mit AP1 Wifi208 Channel 1 BSSId 68:D7:9A:48:D2:43 in Modus 11n als sonoff-4776...
00:00:01.196 WIF: verbunden
00:00:01.448 HTP: Web-Server aktiv bei sonoff-4776 mit IP-Adresse 192.168.254.135
14:35:44.881 MQT: Verbindungsversuch...
14:35:44.908 MQT: verbunden
14:35:44.910 MQT: tele/sonoff/LWT = Online (beibehalten)
14:35:44.911 MQT: cmnd/sonoff/POWER = 
14:35:44.914 MQT: tele/sonoff/INFO1 = {"Info1":{"Module":"ESP32S3","Version":"12.5.0.2(tasmota)","FallbackTopic":"cmnd/DVES_E1D2A8_fb/","GroupTopic":"cmnd/tasmotas/"}}
14:35:44.926 MQT: tele/sonoff/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"sonoff-4776","IPAddress":"192.168.254.135","IP6Global":"","IP6Local":"fe80::f612:faff:fee1:d2a8"}}
14:35:44.948 MQT: tele/sonoff/INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":4}}
14:35:48.909 QPC: Reset
# After configuring display and template and executing module 0

00:00:00.001 HDW: ESP32-S3 (PSRAM)
00:00:00.018 UFS: FlashFS mounted with 12592 kB free
00:00:00.029 CFG: Loaded from File, Anzahl 16
00:00:00.034 QPC: Count 1
00:00:00.036 SPI: Hardware using GPIO12(CLK), GPIO11(MOSI) and GPIO13(MISO)
00:00:00.089 BRY: Berry initialized, RAM used=4729 bytes
00:00:00.111 I2C: GT911 found at 0x5d
00:00:01.178 TI: GT911 failed
00:00:01.807 DSP: Configured display 'ST7262'
00:00:02.415 Projekt tasmota - Sonoff Version 12.5.0.2(tasmota)-2_0_6(2023.05.07 14:00:51)
00:00:02.415 UFILESYSTEM OK!
00:00:02.419 RSL: RESULT = {"POWER":"ON"}
00:00:02.420 RSL: POWER = ON
00:00:02.433 LVG: LVGL initialized
00:00:03.884 WIF: verbinden mit AP1 Wifi208 Channel 1 BSSId 68:D7:9A:48:D2:43 in Modus 11n als sonoff-4776...
00:00:05.758 WIF: verbunden
00:00:06.012 HTP: Web-Server aktiv bei sonoff-4776 mit IP-Adresse 192.168.254.135
14:39:30.005 MQT: Verbindungsversuch...
14:39:30.531 MQT: verbunden
14:39:30.533 MQT: tele/sonoff/LWT = Online (beibehalten)
14:39:30.534 MQT: cmnd/sonoff/POWER = 
14:39:30.538 MQT: tele/sonoff/INFO1 = {"Info1":{"Module":"SUNTON","Version":"12.5.0.2(tasmota)","FallbackTopic":"cmnd/DVES_E1D2A8_fb/","GroupTopic":"cmnd/tasmotas/"}}
14:39:30.550 MQT: tele/sonoff/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"sonoff-4776","IPAddress":"192.168.254.135","IP6Global":"","IP6Local":"fe80::f612:faff:fee1:d2a8"}}
14:39:30.573 MQT: tele/sonoff/INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":6}}
14:39:30.575 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
14:39:30.586 MQT: stat/sonoff/POWER = ON
14:39:33.371 QPC: Reset

Jason2866, 2023-05-07:
@gemu2015  There is still "something" wrong. Compiled a version with the change and still get

```
00:00:00.001 HDW: ESP32-S3 (PSRAM)
00:00:00.004 UFS: FlashFS mounted with 10284 kB free
00:00:00.007 CFG: Loaded from File, Count 5
00:00:00.008 FRC: Some settings have been reset (2)
00:00:00.010 SPI: Hardware using GPIO12(CLK), GPIO11(MOSI) and GPIO13(MISO)
00:00:00.023 TFS: File 'mcp23x.dat' not found
00:00:00.048 BRY: Berry initialized, RAM used=5354 bytes
00:00:00.056 TFS: File '.drvset003' not found
00:00:00.057 CFG: Energy use defaults as file system not ready or file not found
00:00:00.664 Project sonoff - Tasmota Version 12.5.0.2(lvgl-haspmota)-2_0_9(2023-05-07T14:52:11)
00:00:00.704 I2C: GT911 found at 0x5d
00:00:01.796 TI: GT911 failed
00:00:02.427 RSL: RESULT = {"POWER":"ON"}
```

gemu2015, 2023-05-07:
strange it solved the issue for me and i get the blue LVGL screen and not touch error

gemu2015, 2023-05-07:
my usual setup did never produce this error, as said i exclude every not needed driver
may be there is a clush with another i2c driver

gemu2015, 2023-05-07:
@Jason2866 
to test if any other i2c driver is involved, undefine i2c pins in Tasmota GUI and define SDA,SCL pins in display.ini.
so only GT911 will be addressed 
`:TI1,5d,20,19,-1,38`

Jason2866, 2023-05-07:
@gemu2015  disabled all other i2c with `i2cdriver` and GT911 does init now.
So just need to find which one of this "I2CDriver":"!7,!8,!9,!10,!58,!59,!77" is it

Jason2866, 2023-05-07:
It is driver No. 8 `USE_SHT | xsns_07 | SHT1X`

`

gemu2015, 2023-05-07:
temporary solution is to simply disable it with gt911 used.

gemu2015, 2023-05-07:
sht1 devices are completely outdated not really i2c compatible, should be removed.

Be-Virtual, 2023-05-07:
had to reinstall haspmota manually but get it working with @gemu2015's firmware.

@Jason2866 ,Happy to test the full std version if needed

PR #18594 (2023-05-07): fix i2c touch init
## Description:

**Related issue (if applicable):** fixes #18589

i2c touch fix

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.8
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-c [... cut]
closed 2023-05-07 by Jason2866