arendst/Tasmota #18590: Tasmota - Berry - LVGL

BassT23, 2023-05-06:
### PROBLEM DESCRIPTION
Please more input for using LVGL with Berry

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32 Dev Kit C v4
- [x] Tasmota binary firmware version number used: 12.4.0
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
22:28:58.657 MQT: stat/Display-Flur/STATUS = {"Status":{"Module":0,"DeviceName":"Display-Flur","FriendlyName":["Display-Flur",""],"Topic":"Display-Flur","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
22:28:58.695 MQT: stat/Display-Flur/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-lvgl.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:38:34","StartupUTC":"2023-05-06T19:50:24","Sleep":50,"CfgHolder":4617,"BootCount":89,"BCResetTime":"2023-05-04T19:03:57","SaveCount":254}}
22:28:58.733 MQT: stat/Display-Flur/STATUS2 = {"StatusFWR":{"Version":"12.4.0(AZ-Touch)","BuildDateTime":"2023.05.06 09:28:48","Core":"2_0_6","SDK":"4.4.3.20230129","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"553/699"}}
22:28:58.751 MQT: stat/Display-Flur/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Trinity",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000001"]}}
22:28:58.787 MQT: stat/Display-Flur/STATUS4 = {"StatusMEM":{"ProgramSize":2113,"Free":766,"Heap":36,"StackLowMark":19,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["00000407","0FA047CE","00141C01","48083003","00200000","C0080881","000018C0","003E8000","54000808","00000080"],"Drivers":"1,2,4,8,9,10,12,13,20,21,24,26,29,35,38,50,52,54,55,59,62,79","Sensors":"1,2,10,11,16,24,27,41,52,57,62,67,70,71,127"}}
22:28:58.837 MQT: stat/Display-Flur/STATUS5 = {"StatusNET":{"Hostname":"Display-Flur","IPAddress":"192.168.20.127","Gateway":"192.168.20.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.20.1","DNSServer2":"0.0.0.0","Mac":"94:E6:86:0C:ED:F8","IP6Global":"","IP6Local":"fe80::96e6:86ff:fe0c:edf8","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
22:28:58.864 MQT: stat/Display-Flur/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.20.4","MqttPort":1883,"MqttClientMask":"Display-Flur","MqttClient":"Display-Flur","MqttUser":"BassT","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
22:28:58.892 MQT: stat/Display-Flur/STATUS7 = {"StatusTIM":{"UTC":"2023-05-06T20:28:58","Local":"2023-05-06T22:28:58","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"06:21","Sunset":"21:11"}}
22:28:58.916 MQT: stat/Display-Flur/STATUS10 = {"StatusSNS":{"Time":"2023-05-06T22:28:58","ESP32":{"Temperature":53.3},"TempUnit":"C"}}
22:28:58.939 MQT: stat/Display-Flur/STATUS11 = {"StatusSTS":{"Time":"2023-05-06T22:28:58","Uptime":"0T00:38:34","UptimeSec":2314,"Heap":33,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"Berry":{"HeapUsed":10,"Objects":166},"POWER1":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Trinity","BSSId":"B0:7F:B9:A4:90:1C","Channel":11,"Mode":"11n","RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
I want to make my first Screen with multiple pages on AZ-Touch

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
I search a lot in lvgl pages, berry pages and google, but find no answers.
For now I had a page with 3 Buttons.

autoexec.be:
```
#- start LVGL and init environment -#
lv.start()

hres = lv.get_hor_res()       # should be 320
vres = lv.get_ver_res()       # should be 240

scr = lv.scr_act()            # default screan object
f20 = lv.montserrat_font(20)  # load embedded Montserrat 20
[... 52 lines cut ...]
power_label.set_text(lv.SYMBOL_POWER)
power_label.center()

#- callback function when a button is pressed, react to EVENT_CLICKED event -#
def btn_clicked_cb(obj, event)
    var btn = "Unknown"
    if     obj == home_btn  btn = "Home"  load("autoexec.be")
    elif   obj == next_btn  btn = "Next"  load("page2.be")
    elif   obj == power_btn  btn = "Power"
    end
    tasmota.cmd("buzzer")
    print(btn, "button pressed")
    tasmota.publish("tele/Display-Flur/SENSOR/Buttom",btn)
end

next_btn.add_event_cb(btn_clicked_cb, lv.EVENT_CLICKED, 0)
home_btn.add_event_cb(btn_clicked_cb, lv.EVENT_CLICKED, 0)
power_btn.add_event_cb(btn_clicked_cb, lv.EVENT_CLICKED, 0)
```
I also made a second page calles page2.be. with the "next" button I can call the second page, but all old buttons where there also, ...

so now my first question: how to make a second page without any content, and fill this with new button?
I tryed with `lv.obj_clean()`  but will give output: `lv has no member obj_clean`

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2023-05-06:
It would be simpler to use HASPmota. 

If you want to see the magic check here https://github.com/arendst/Tasmota/blob/e9d1e8c7250d89a24ade0c42a64731d6c492bbb2/tasmota/berry/haspmota_src/haspmota_core/haspmota.be#L1258

A new screen is created with an object with no parent `lv.obj(0)`

You can put buttons in a top layer with `lv.layer_top()`. It is always displayed on top of screens. 

`lv.scr_load_anim()` changes from one screen to another with animation 
https://github.com/arendst/Tasmota/blob/e9d1e8c7250d89a24ade0c42a64731d6c492bbb2/tasmota/berry/haspmota_src/haspmota_core/haspmota.be#L1349

s-hadinger, 2023-05-06:
https://tasmota.github.io/docs/HASPmota/

BassT23, 2023-05-06:
haspmota will not work, because my board has no psram :(

s-hadinger, 2023-05-07:
This constraint was when HASPmota needed to be compiled in RAM. Now that it is solidified in Flash it should work for not too complex setups

BassT23, 2023-05-07:
You mean, that haspmota will work with my "ESP32S Dev Kit C V4 NodeMCU - ESP32-D0WD-V3 rev.3)?
is there a pre compiled version for that?
How to self compile - what need to define?
Cant find any souch changes in changelog

BassT23, 2023-05-07:
Another Problem;
If i rotate the display wit :r,3 or displayrotate 3, the touchscreen is "mirrored". so, if i touch on the lift side, the real recognization is on the right side, ...
does it give a fix for that?

BassT23, 2023-05-07:
Ok - HASPmota without PSRAM work. Should be wrote in wiki. Only the demo pake don't.

s-hadinger, 2023-05-07:
> Another Problem;
> If i rotate the display wit :r,3 or displayrotate 3, the touchscreen is "mirrored". so, if i touch on the lift side, the real recognization is on the right side, ...
> does it give a fix for that?

Did you check the numerous parameters in display.ini?

BassT23, 2023-05-07:
in rotate 0 - the touch work like it should.
here my display.ini:
```
:H,ILI9341,320,240,16,SPI,1,*,*,*,*,*,*,*,20
:S,2,1,1,0,40,20
:I
EF,3,03,80,02
CF,3,00,C1,30
ED,4,64,03,12,81
E8,3,85,00,78
CB,5,39,2C,00,34,02
F7,1,20
EA,2,00,00
C0,1,23
C1,1,10
C5,2,3e,28
C7,1,86
36,1,48
37,1,00
3A,1,55
B1,2,00,18
B6,3,08,82,27
F2,1,00
26,1,01
E0,0F,0F,31,2B,0C,0E,08,4E,F1,37,07,10,03,0E,09,00
E1,0F,00,0E,14,03,11,07,31,C1,48,08,0F,0C,31,36,0F
11,80
29,80
:o,28
:O,29
:A,2A,2B,2C
:R,36
:0,28,00,00,05
:1,88,00,00,01
:2,E8,00,00,02
:3,48,00,00,03
:r,0
:i,20,21
:TS,*
:M,3946,401,241,3828
#
```

BassT23, 2023-05-07:
> Did you check the numerous parameters in display.ini?

please explain this more. Should I change the numbers?

s-hadinger, 2023-05-07:
The effect on touchscreen for rotate 3 is described in the 3rd number after `:3`.
Currently it is 3 which means: "swap and flip y".
I would try `80` which means "swap x,y"

So the complete line should be (if I understand well the doc):
`:3,48,00,00,80`

s-hadinger, 2023-05-07:
See here https://tasmota.github.io/docs/Displays/#descriptor-file

BassT23, 2023-05-07:
I read about that, but now I understand it better.

it should be:
:r,3
:M,3946,401,3828,241 <- here i must change the values

Now it work correctly :D