arendst/Tasmota #18659: Truncated BLE Beacon Advertisements

ajlennon, 2023-05-15:
### PROBLEM DESCRIPTION

I am experimenting with some BLE tags from HolyIoT. These send out a regular unencrypted beacon advertisement with sensor data depending on the type of sensor (e.g.currently pressure, temperature, humidity)

http://www.holyiot.com/eacpzs.asp?dlb_id=31

- when I use these with the provided iOS application it works fine

- when I look at the raw beacon data with an Android tool I see the full packet as per the protocol specification

- however when I use "bledetails2" to look at the raw beacon data in Tasmota I see a truncated advertisement

10:36:45.845 MQT: tele/SmartPlug_FactoryDefault/BLE = {"DetailsBLE":{"mac":"F131F5C217BA","RSSI":-22,"p":"0201061AFFFFFF021500000000000000000000000000000000271B4CB9CC"}}

I *think* this is because NimBLE as implemented in Tasmota is not handling "extended BLE advertisements" although I am not certain. 

I have looked at the received payload size out of NimBLE and it is correct for the truncated data displayed

I have looked at enabling extended advertising in `nimconfig.h` but it doesn't seem to make a difference

```
/** @brief Un-comment to enable extended advertising */
#define CONFIG_BT_NIMBLE_EXT_ADV 1

/** @brief Un-comment to set the max number of extended advertising instances (Range: 0 - 4) */
#define CONFIG_BT_NIMBLE_MAX_EXT_ADV_INSTANCES 4

/** @brief Un-comment to set the max extended advertising data size (Range: 31 - 1650) */
#define CONFIG_BT_NIMBLE_MAX_EXT_ADV_DATA_LEN 1024

/** @brief Un-comment to enable periodic advertising */
#define CONFIG_BT_NIMBLE_ENABLE_PERIODIC_ADV 1
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): _____ ESP32S3 based board we are designing
- [ ] Tasmota binary firmware version number used: _____ ESP32 master branch
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: _____ Platform.io esptool.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

10:41:29.982 CMD: Backlog Template; Module; GPIO 255
10:41:30.020 MQT: stat/SmartPlug_FactoryDefault/RESULT = {"NAME":"2NICE UP111","GPIO":[0,320,0,0,2720,2656,0,0,2624,576,224,0,0,0,1,1,1,1,1,1,1,1,0,0,0,0,0,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
10:41:30.038 MQT: tele/SmartPlug_FactoryDefault/BLE = {"DetailsBLE":{"mac":"F131F5C217BA","RSSI":-25,"p":"0201061AFFFFFF021500000000000000000000000000000000271B4CB9CC"}}
10:41:30.262 MQT: stat/SmartPlug_FactoryDefault/RESULT = {"Module":{"1":"ESP32S3"}}
10:41:30.466 MQT: stat/SmartPlug_FactoryDefault/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"},"GPIO40":{"0":"None"},"GPIO41":{"0":"None"},"GPIO42":{"0":"None"},"GPIO43":{"0":"None"},"GPIO44":{"0":"None"},"GPIO45":{"0":"None"},"GPIO46":{"0":"None"},"GPIO47":{"0":"None"},"GPIO48":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
status 0
10:42:06.911 MQT: stat/SmartPlug_FactoryDefault/STATUS = {"Status":{"Module":1,"DeviceName":"KC_SmartPlug","FriendlyName":["KC_SmartPlug"],"Topic":"SmartPlug_FactoryDefault","ButtonTopic":"0","Power":-1,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
10:42:06.960 MQT: stat/SmartPlug_FactoryDefault/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"smartplugs","OtaUrl":"http://dl.kettlecompanion.com/devel/tasmota-minimal-trampoline.bin.gz","RestartReason":"Vbat power on reset","Uptime":"0T17:59:29","StartupUTC":"2023-05-14T15:42:37","Sleep":50,"CfgHolder":4617,"BootCount":27,"BCResetTime":"2023-05-14T12:50:29","SaveCount":44}}
10:42:07.000 MQT: stat/SmartPlug_FactoryDefault/STATUS2 = {"StatusFWR":{"Version":"12.4.0(bluetooth)","KC_Version":"2.0.0.0","BuildDateTime":"2023-05-14T16:40:57","Core":"2_0_6","SDK":"4.4.3.20230129","CpuFrequency":240,"Hardware":"ESP32-S3","CR":"462/699"}}
[... 23 lines cut ...]
10:42:53.545 MQT: tele/SmartPlug_FactoryDefault/BLE = {"DetailsBLE":{"mac":"F131F5C217BA","RSSI":-25,"p":"0201061AFFFFFF021500000000000000000000000000000000271B4CB9CC"}}
10:42:56.379 BLE: payload 30
10:42:56.397 MQT: tele/SmartPlug_FactoryDefault/BLE = {"DetailsBLE":{"mac":"F131F5C217BA","RSSI":-21,"p":"0201061AFFFFFF021500000000000000000000000000000000271B4CB9CC"}}
10:42:56.908 EQ3 poll cycle starting
10:42:59.258 BLE: payload 30

```

### TO REPRODUCE

Use `bledetails2` to look at raw advertisement data from a HolyIoT tag (or presumably any tag with extended advertising)

### EXPECTED BEHAVIOUR

We should see all the advertising packet data for a tag, which should correspond to the raw data seem on e.g. an Android phone tool

### SCREENSHOTS

TBD

### ADDITIONAL CONTEXT

N/A

**(Please, remember to close the issue when the problem has been addressed)**

ajlennon, 2023-05-15:
> Since BLE 5.0 not supported

I'm sorry I don't understand - longer advertisements are not supported in BLE5.0 ?

ajlennon, 2023-05-15:
Ah you’re saying it needs BLE 5 for extended advertising. Yes the ESP32S3 should support this

Jason2866, 2023-05-15:
Yes, hardware does support. But it is not implemented in Tasmota software.

barbudor, 2023-05-15:
I think we can ask @h2zero some advice

h2zero, 2023-05-15:
Do you have active scanning enabled? The missing data appears to be in the scan response as the advertisement data is at maximum.

ajlennon, 2023-05-15:
> Do you have active scanning enabled? The missing data appears to be in the scan response as the advertisement data is at maximum.

I don't. I'll give it a go!

ajlennon, 2023-05-16:
OK this is fantastic thanks @h2zero, everybody . With active scanning enabled I now see all the advertisement data with a vanilla build with -DFIRMWARE_BLUEOOTH !!!

```
10:10:00.014 MQT: tele/SmartPlug_FactoryDefault/BLE = {"DetailsBLE":{"mac":"F131F5C217BA","RSSI":-36,"p":"0201061AFFFFFF021500000000000000000000000000000000271B4CB9CC0D09486F6C792D494F5400000000101642524162F131F5C217BA06060258C5"}}
```
I like these little HolyIoT tags a lot as there's - for example - temperature/pressure/humidity in a really small coin cell powered  form factor

I am unclear on whether their choice to advertise with packets that use the BLE 5.x extended advertising format causes potential issues.

Clearly I need to do some more research but can you offer any advice on what impact "active scanning" has to the Tasmota stack and/or to power consumption on the devices themselves? It seems like a better option would be for them to advertise in the legacy format that doesn't use active scanning but they are probably unwilling to change their firmware I would guess
 
[image]

ajlennon, 2023-05-16:
My HolyIoT beacon now reporting via BLE through Tasmota to an MQTT Mosquitto broker where I'm graphing some numbers :)

[image]

Jason2866, 2023-05-16:
If it is working without active scan, you are back at square one. Needs to be implemented.
Without being verified working / supported from underlying Nimble library. The train stops here.

ajlennon, 2023-05-16:
>If it is working without active scan, you are back at square one. Needs to be implemented.

OK so "active scan" is where we are connecting to the device to interrogate it for extended information (if I am understanding) and we want to avoid this

>Without being verified working / supported from underlying Nimble library. The train stops here.

So I need to confirm that extended advertising does not require active scan, and then work out why I'm not seeing the full packets through the NimBLE library. I'll try some basic (non-Tasmota) NimBLE examples to see what happens and if that doesn't get me anywhere I'll engage with the NimBLE folks and see if they can advise!

h2zero, 2023-05-16:
These devices require active scanning in legacy mode because they provide the sensor data in the scan response. This is an unfortunate design choice for battery life. 

If these devices can have extended advertising enabled then NimBLE will need to have it enabled as well to receive the extended advertisements. This can be done with the build config flags in your post above.

Jason2866, 2023-05-17:
@ajlennon  h2zero is the maintainer of the used NimBLE lib in Tasmota ;-)

ajlennon, 2023-05-18:
A little bit more research....

- I ran the NimBLE Arduino example code for the beacon scanner
- I fiddled with extended advertising options. I *seem* to be getting extended advertising GAP discovery events but I don't see the extra data
- I've made these points to the folks at HolyIoT and hopefully they are going to give me a revised firmware that will advertise all the sensor data in one legacy packet

joopheuvel, 2023-06-01:
Hi Alex / Jason.
I am experiencing the same issue. Was hoping you could help me.
I want to use/read the pushbutton on the HolyIOT beacon (ESp32, Arduino).
As you indicate, I am getting the string until the point you also had, and the stuff I need is not part of that as it seems truncated. Tries a call for a specific part verus the whole ble string:
sprintf (ble_string,"%s",advertisedDevice->toString().c_str());
and
sprintf (manufacturer_data_string,"%s",advertisedDevice->getManufacturerData().c_str());
but the last one is no readable text.
Desperate to find a solution as the board I am using for a gateopener system with beacon will make registration a 
lot easier versus having to push an extra button. Can you let me know how you think I can fix this...?
Thanks

Joop van den heuvel

ajlennon, 2023-06-07:
Hi @joopheuvel,

Sorry for the delay - I've been ill. As the others say above you can enable active scanning and you'll get the full message.

However I agree with the comments that this isn't what we want and really the HolyIoT firmware isn't fit for purpose.

They did say they'd do a redesign to fit everything into a single legacy advertisement but I haven't heard anything since.

Ultimately if they can't make this work as I think it should work for a battery powered beacon I won't be continuing to look at their devices :'(

joopheuvel, 2023-06-07:
Thanks Alex
No problem, hope you feel better,I did already  have active scanning on:

  #if defined(NimBLE_libraries)
    if (use_ble) {
      SerialPrintLogging ("Initializing NimBLE..\n");
      NimBLEDevice::init("");
      pBLEScan = NimBLEDevice::getScan(); //create new scan
      pBLEScan->setAdvertisedDeviceCallbacks(new
MyAdvertisedDeviceCallbacks(),TRUE); //
https://github.com/nkolban/esp32-snippets/issues/464
      pBLEScan->setActiveScan(TRUE);   //active scan uses more power, but
get results faster
      pBLEScan->setInterval(BLE_SCANINTERVAL_MS);       // was 1000 when
working
      pBLEScan->setWindow(BLE_SCANWINDOW_MS);         // less or equal
setInterval value was 500 when working
      pBLEScan->setMaxResults(0);      // do not store the scan results,
use callback only.
      lastlive_ble = millis();
    }
  #endif // NimBLE_libraries

See code above. Ths would do the trick right?  am still not seeing the full
string but a truncated one as indicated. Am I doing something  wrong. TRUE
has value 1, so should work, right?. Thanks!
Joop

ajlennon, 2023-06-08:
Just taking a step back here @joopheuvel - I think you should be able to build Tasmota with BLE or install the BLE OTA firmware, then configure active scan on in the configuration section. then up the DEBUG messages and see the packets received which should be the full ones? I might have had to enable bridging to see this? Is this what you're trying?

joopheuvel, 2023-06-09:
Thanks Alex,

This would be way too deep for me and beyond my technical capabilities.
I will leave it for the moment, maybe will look into this later.
Many thanks for your help and hope you feel better.

Another question is on how to update settings in these Holyiot beacons.
do you happen to know how to do this, do they have generic passwords or
specific?

Thanks!
Joop

ajlennon, 2023-06-09:
It's actually not too hard. I'm happy to help. What ESP32 device are you running ?

ajlennon, 2023-06-09:
Also this is the default password to connect (you can then change it) 

[image]

joopheuvel, 2023-06-09:
Great!
Thanks Alex, will try.
Joop

joopheuvel, 2023-06-09:
Hi Alex,

I have a multi-station (1-6), event (festival/mobile) beacon tracker and
gate opener system (for beacons on cords and on cars).
Have a mix of (Olimex ESP32-EVB's) that run from a (Teltonika) Router
supplying MQTT tags, and some work
autonomously running their BLE with the external antenna. They track the
beacons and as form of registration/deregistration
the button clock close to an ESP32 would be nice to register/unregister.

Thanks!
Joop

ajlennon, 2023-06-10:
> Have a mix of (Olimex ESP32-EVB's) that run from a (Teltonika) Router

OK! So if I assume this is a bog standard ESP32 then I suggest you program a standard Tasmota Bluetooth image in there

If you are doing an OTA update "Firmware Update" then you can use this image

http://ota.tasmota.com/tasmota32/release/tasmota32-bluetooth.bin

Or if you are programming with `esptool.py` then you can use this image

http://ota.tasmota.com/tasmota32/release/tasmota32-bluetooth.factory.bin

With

`esptool.py write_flash 0x0 tasmota32-bluetooth.factory.bin`

This will give you an extra configuration option for Bluetooth where you can enable active scanning

Try that and if that works for you I'll try to dig out the commands to enable Bluetooth beacon -> MQTT bridging

Cheers!

joopheuvel, 2023-06-11:
Hey Alex.

You are quick. Thanks, very helpful indeed. Let me wait with flashing until
I gave the next batch in, just to be safe.
I am sure we will get it to work. Was able to change settings already,
lower power and slower interval.
Many thanks again.
Note we were using teltonika beacons but batteries can not be replaced.
Bommer. These were 10 euro a piece, replacement batteries,  and seem to do
the job.

Joop

ajlennon, 2023-06-11:
> Let me wait with flashing until I gave the next batch in, just to be safe.

Sure thing - just @ me here when you're ready to go :)

ajlennon, 2023-06-12:
Update: The HolyIoT people have been back to me and are nearly ready with a new firmware build

"the firmware almost finished , we put the battery level , tempreture, humidity,air pressure in the scan mode"

So that sounds positive! Will keep you all updated on this thread

joopheuvel, 2023-06-12:
Ah. Top. Great Alex, keep me in the loop please. Very much appreciated.
Joop

rus-ik, 2024-05-08:
> Update: The HolyIoT people have been back to me and are nearly ready with a new firmware build
> 
> "the firmware almost finished , we put the battery level , tempreture, humidity,air pressure in the scan mode"
> 
> So that sounds positive! Will keep you all updated on this thread

Do you have any updates regarding the HolyIoT  beacon?

Rude999, 2025-03-15:
Dear all,

Is it possible to add some text into the HolyioT which others (nearby) can pick up with Bluetooth? 

I want to use this for creating a GeoCache someware outside in the woods.

Rude

Rude999, 2025-04-01:
Dear @h2zero, 

Is it possible to add some text into the HolyioT which others (nearby) can pick up with Bluetooth?

I want to use this for creating a GeoCache someware outside in the woods.

Rude

h2zero, 2025-04-01:
@Rude999 That would be a question for Holyiot, it would be a modification to their firmware. Unless of course you can upload custom firmware to their tags yourself, then you can add whatever you like :).

Rude999, 2025-04-02:
@h2zero  Thank you for your reply. 

Do you or anyone else in this group know if there is another brand (nRF) Beacon that I can add text to using an app? Holyiot indicates that they removed that feature because it 'damages' the Beacon.