arendst/Tasmota #18670: Shutter Interlocked relays (both) go ON after reboot

kbaras, 2023-05-16:
### PROBLEM DESCRIPTION
Hi!

I have a serious problem with the latest Tasmota release. 
I have a shutter configured with two relays, with  ShutterMode1 and the relays are interlocked (1,2). As I am using inverted relays the configuration has been set accordingly (with relay_i both for relay1 and relay2). It works fine at normal operation (the interlock as well, so if one of the relays is turned ON and the other is clicked to turn it ON than the interlock automatically turns OFF the other relay so you are unable to turn ON both - which is the expected behaviour).

Problem comes after I restart Tasmota (plugged out the power source and plugged in again) then both of the shutter relays are ON after startup! After clicking on either of the relays the interlock intervenes and it turns OFF the other relay so from this point it works fine until the next restart...

I have a separate inverted relay configured additionally and it does not start with ON state.

Initially the PowerOnState was 3, but I have also tried with 0 but it did not help...

  STATUS 0 output:
18:20:30.680 CMD: Status 0
18:20:30.687 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota Wemos Kitchen shutter & sinklight","FriendlyName":["Tasmota","Tasmota2","Tasmota3"],"Topic":"tasmotawemoskitchen","ButtonTopic":"0","Power":3,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:20:30.723 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:00:21","StartupUTC":"2023-05-16T17:20:09","Sleep":50,"CfgHolder":4617,"BootCount":17,"BCResetTime":"2023-05-16T17:34:21","SaveCount":44,"SaveAddress":"F8000"}}
18:20:30.755 RSL: STATUS2 = {"StatusFWR":{"Version":"12.5.0(tasmota)","BuildDateTime":"2023-04-17T08:03:54","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"438/699"}}
18:20:30.775 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HOME_WIRELESS","HOME_WIRELESS_FF"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C80001000600003C5A0A192800000000","40000080","00006000","00004000","00000000"]}}
18:20:30.808 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":632,"Free":368,"Heap":26,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
18:20:30.840 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmotawemoskitchen-6024","IPAddress":"192.168.1.116","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"CC:50:E3:69:57:88","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:20:30.865 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.28","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_695788","MqttUser":"kbaras","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:20:30.887 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-05-16T17:20:30","Local":"2023-05-16T18:20:30","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"05:07","Sunset":"20:25"}}
18:20:30.909 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-05-16T18:20:30","Switch3":"ON","Switch4":"ON","Switch5":"OFF","Switch6":"OFF","AM2301":{"Temperature":null,"Humidity":null,"DewPoint":null},"Shutter1":{"Position":2,"Direction":0,"Target":2,"Tilt":0},"TempUnit":"C"}}
18:20:30.932 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-05-16T18:20:30","Uptime":"0T00:00:21","UptimeSec":21,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"ON","POWER2":"ON","POWER3":"OFF","Wifi":{"AP":1,"SSId":"HOME_WIRELESS","BSSId":"18:FD:74:C5:42:94","Channel":11,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}}
18:20:30.963 RSL: STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":170,"Close":170,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"1"}}}

### TO REPRODUCE
- Configure two inverted relays into a shutter with interlock group
- Cut the power source and add power again (reboot)
- Both relays will be ON

### EXPECTED BEHAVIOUR
After powerup both relays should be in OFF state.

### SCREENSHOTS
[image]

kbaras, 2023-05-16:
Sorry, I forgot to mention that I am using the Generic (18) configuration with a Wemos D1 mini - but the hardware is probably not so important here...

kbaras, 2023-05-16:
Ok, I have recognized that the problem comes up when I configure D1 and D2 as the relays for shutter...

stefanbode, 2023-05-17:
hi, pls check that `savedata 1` is set. then first check on the `poweronstate`. this must be either 0 or 3, I do all my tests with 3, the situation with both relay on only can happen if the relay where ON and then you do `savedata 0`, in this case the state is always wrong on both ON and will not updated, The shutter driver cannot override the `interlock 1,2`. pls also check that the interlock is also set with `interlock on`. My command to get it fixed:
```
savedata 1
interlock 1,2
interlock on
poweronstate 0
restart 1
```

stefanbode, 2023-05-17:
this can also be a reason. pls try `so63 1`

SetOption63	Set relay state feedback scan at restart ([#5594](https://github.com/arendst/Tasmota/issues/5594), [#5663](https://github.com/arendst/Tasmota/issues/5663))
0 = Scan power state at restart (default)
1 = Disable power state scanning at restart

Jason2866, 2023-05-21:
Ping @kbaras to test latest development version with the mentioned settings.

stefanbode, 2023-05-21:
I have an idea where it comes from. Maybe the inverter relay is the root cause. If I think about it it's getting more likely. I need to setup a testbench to check it again. The use case is quite rare. And it was not part of the testing.

stefanbode, 2023-05-23:
Double checked the Relay_i and cannot reproduce the error. Device react as specified.

kbaras, 2023-05-23:
Hi! 
Thank you for handling the issue. 
In the meantime I was able to "workaround" it by using different pins for the shutter (D5 and D6). As I've mentioned I experienced the problem using D1 and D2 initially.

stefanbode, 2023-05-23:
Sorry to say, but this makes no sense at all. May be a pullup or pulldown might be the case. Here s`etoption63 1` can solve this one. I use this as a default.

Jason2866, 2023-05-26:
Closing. Using non restricted GPIOs is always better. Restricted GPIOs can generate not wanted side effects.

closed 2023-05-26 by Jason2866

stefanbode, 2025-07-17:
The interlock always has priority no matter if shutter or not. Please can you disable the shutter. Setoption80 off and check if the behavior is the same? There is one option that is not default but this set the pin to the status it see at boot. Looking forward to your investigation