arendst/Tasmota #18672: Shelly Plus 2PM kWh not saved in eprom

Zounich, 2023-05-16:
I have a Shelly Plus 2PM with PCB v0.1.9
The problem is that the kWh is lost after a power on reset.

[image]

The Tasmota version is: 12.5.0
The following template was used: {"NAME":"Shelly Plus 2PM PCB v0.1.9","GPIO":[320,0,0,0,32,192,0,0,225,224,0,0,0,0,193,0,0,0,0,0,0,608,640,3458,0,0,0,0,0,9472,0,4736,0,0,0,0],"FLAG":0,"BASE":1}

SetOption129 1 is aktiv

Savedata is 1

Is there any parameter to change that?

Jason2866, 2023-05-17:
Please provide logs.

sfromis, 2023-05-17:
Note that consumption data is not saved continuously, but only at midnight, or when settings are saved for other reasons.

Jason2866, 2023-05-19:
Closing, no feedback

closed 2023-05-19 by Jason2866

Zounich, 2023-05-19:
Please excuse the delay
here are the login data:

06:37:44.081 MQT: tele/Poolpumpe/SENSOR = {"Time":"2023-05-18T06:37:44","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature1":24.8},"ENERGY":{"TotalStartTime":"1970-01-01T00:00:00","Total":[0.042,0.000],"Yesterday":[0.042,0.000],"Today":[0.000,0.000],"Period":[0,0],"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":50,"Voltage":241,"Current":[0.000,0.000]},"TempUnit":"C"}
06:37:57.263 HTP: Console
06:39:36.289 HTP: Consoles
06:39:42.386 HTP: Manage File system
06:39:47.499 HTP: Consoles
06:39:49.038 HTP: Main Menu
06:40:02.091 HTP: Configuration
06:40:12.609 HTP: Configure Logging
06:40:41.605 HTP: Configuration
06:40:46.411 HTP: Main Menu
06:40:49.605 HTP: Consoles
06:40:50.996 HTP: Console
06:42:44.101 MQT: tele/Poolpumpe/STATE = {"Time":"2023-05-18T06:42:44","Uptime":"0T09:30:10","UptimeSec":34210,"Heap":167,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":45},"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"Draft","BSSId":"24:5A:4C:A7:FB:D2","Channel":1,"Mode":"11n","RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:03"}}
06:42:44.114 MQT: tele/Poolpumpe/SENSOR = {"Time":"2023-05-18T06:42:44","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature1":25.3},"ENERGY":{"TotalStartTime":"1970-01-01T00:00:00","Total":[0.042,0.000],"Yesterday":[0.042,0.000],"Today":[0.000,0.000],"Period":[0,0],"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":50,"Voltage":242,"Current":[0.000,0.000]},"TempUnit":"C"}
06:44:35.602 WIF: Network (re)scan started...
06:44:42.571 WIF: Network 0, AP1, SSId Draft, Channel 11, BSSId 50:E6:36:91:67:85, RSSI -50, Encryption 1
06:44:42.572 WIF: Network 1, AP-, SSId FRITZ!Box Gastzugang, Channel 11, BSSId CA:E6:36:91:67:86, RSSI -51, Encryption 1
06:44:42.573 WIF: Network 2, AP-, SSId DraftK, Channel 1, BSSId 26:5A:4C:97:FB:D2, RSSI -54, Encryption 1
06:44:42.574 WIF: Network 3, AP1, SSId Draft, Channel 1, BSSId 24:5A:4C:A7:FB:D2, RSSI -54, Encryption 1
06:44:42.575 WIF: Network 4, AP-, SSId WLAN-8K7MLN, Channel 6, BSSId 04:A2:22:8B:D1:71, RSSI -77, Encryption 1
06:44:42.576 WIF: Network 5, AP-, SSId DraftK, Channel 1, BSSId 76:AC:B9:B1:9A:08, RSSI -80, Encryption 1
06:44:42.577 WIF: Network 6, AP1, SSId Draft, Channel 1, BSSId 74:AC:B9:B1:9A:08, RSSI -80, Encryption 1
06:47:44.094 MQT: tele/Poolpumpe/STATE = {"Time":"2023-05-18T06:47:44","Uptime":"0T09:35:10","UptimeSec":34510,"Heap":170,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":45},"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"Draft","BSSId":"24:5A:4C:A7:FB:D2","Channel":1,"Mode":"11n","RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:03"}}
06:47:44.107 MQT: tele/Poolpumpe/SENSOR = {"Time":"2023-05-18T06:47:44","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature1":25.0},"ENERGY":{"TotalStartTime":"1970-01-01T00:00:00","Total":[0.042,0.000],"Yesterday":[0.042,0.000],"Today":[0.000,0.000],"Period":[0,0],"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":50,"Voltage":240,"Current":[0.000,0.000]},"TempUnit":"C"}
06:52:44.081 MQT: tele/Poolpumpe/STATE = {"Time":"2023-05-18T06:52:44","Uptime":"0T09:40:10","UptimeSec":34810,"Heap":171,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":45},"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"Draft","BSSId":"24:5A:4C:A7:FB:D2","Channel":1,"Mode":"11n","RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:03"}}
06:52:44.093 MQT: tele/Poolpumpe/SENSOR = {"Time":"2023-05-18T06:52:44","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature1":24.9},"ENERGY":{"TotalStartTime":"1970-01-01T00:00:00","Total":[0.042,0.000],"Yesterday":[0.042,0.000],"Today":[0.000,0.000],"Period":[0,0],"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":50,"Voltage":241,"Current":[0.000,0.000]},"TempUnit":"C"}
06:57:44.073 MQT: tele/Poolpumpe/STATE = {"Time":"2023-05-18T06:57:44","Uptime":"0T09:45:10","UptimeSec":35110,"Heap":171,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":45},"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"Draft","BSSId":"24:5A:4C:A7:FB:D2","Channel":1,"Mode":"11n","RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:03"}}
06:57:44.086 MQT: tele/Poolpumpe/SENSOR = {"Time":"2023-05-18T06:57:44","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature1":24.5},"ENERGY":{"TotalStartTime":"1970-01-01T00:00:00","Total":[0.042,0.000],"Yesterday":[0.042,0.000],"Today":[0.000,0.000],"Period":[0,0],"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":50,"Voltage":240,"Current":[0.000,0.000]},"TempUnit":"C"}
07:02:44.073 MQT: tele/Poolpumpe/STATE = {"Time":"2023-05-18T07:02:44","Uptime":"0T09:50:10","UptimeSec":35410,"Heap":171,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":45},"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"Draft","BSSId":"24:5A:4C:A7:FB:D2","Channel":1,"Mode":"11n","RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:03"}}
07:02:44.086 MQT: tele/Poolpumpe/SENSOR = {"Time":"2023-05-18T07:02:44","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature1":24.8},"ENERGY":{"TotalStartTime":"1970-01-01T00:00:00","Total":[0.042,0.000],"Yesterday":[0.042,0.000],"Today":[0.000,0.000],"Period":[0,0] [... cut ...]
07:10:30.049 MQT: tele/Poolpumpe/SENSOR = {"Time":"2023-05-18T07:10:30","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature1":25.1},"ENERGY":{"TotalStartTime":"1970-01-01T00:00:00","Total":[0.042,0.000],"Yesterday":[0.042,0.000],"Today":[0.000,0.000],"Period":[0,0],"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":50,"Voltage":240,"Current":[0.000,0.000]},"TempUnit":"C"}

It is really the case that the consumption data is only saved at night. It's the same with my Plug S. I just never noticed.
I can live with that.
Are you still finding problems in the log?

Jason2866, 2023-05-19:
With the command  `restart 2` you shutdown the esp correctly. In this state it can be powered off. Settings are saved. After applying power again all values should be there.