arendst/Tasmota #18702: SDS0X1 PM shows INVALID in Main Menu

mrMoe, 2023-05-20:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

`SDS0X1 PM 2.5 µm` always show `<--INVALID--> µg/m³` in main menue but seems to have valid values in sensor data in console.
`{"Time":"2023-05-20T15:19:50","BME280":{"Temperature":21.1,"Humidity":49.0,"DewPoint":10.0,"Pressure":978.1},"SDS0X1":{"PM2.5":7.2,"PM10":8.0},"PressureUnit":"hPa","TempUnit":"C"}`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Generic (0)
- [x] Tasmota binary firmware version number used: 12.5.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: `esptool.py write_flash -fs 1MB -fm dout 0x0 tasmota-sensors.bin`
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:34:06.530 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
15:34:06.759 RSL: RESULT = {"Module":{"0":"Generic"}}
15:34:06.966 RSL: RESULT = {"GPIO0":{"640":"I2C SDA"},"GPIO1":{"1728":"SDS0X1 Tx"},"GPIO2":{"608":"I2C SCL"},"GPIO3":{"1760":"SDS0X1 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
15:34:46.473 RSL: RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
15:34:46.704 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
15:34:46.908 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:35:05.906 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_D6938A","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
15:35:05.911 RSL: STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-sensors.bin.gz","RestartReason":"External System","Uptime":"0T00:19:26","StartupUTC":"2023-05-20T14:15:39","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2023-05-20T13:51:14","SaveCount":36,"SaveAddress":"F8000"}}
15:35:05.915 RSL: STATUS2 = {"StatusFWR":{"Version":"12.5.0(sensors)","BuildDateTime":"2023-04-17T08:03:55","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"389/699"}}
15:35:05.920 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["iot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
15:35:05.937 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":699,"Free":304,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9A8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","54000020","0008C081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34,62,65,66","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,98,103,105","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,69,76"}}
15:35:05.943 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-D6938A-5002","IPAddress":"192.168.xx.xx","Gateway":"192.168.xx.x","Subnetmask":"255.255.255.0","DNSServer1":"192.168.xx.x","DNSServer2":"0.0.0.0","Mac":"68:C6:3A:D6:93:8A","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
15:35:05.946 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D6938A","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:35:05.951 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-05-20T14:35:05","Local":"2023-05-20T15:35:05","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"05:02","Sunset":"20:30"}}
15:35:05.959 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-05-20T15:35:05","BME280":{"Temperature":21.1,"Humidity":48.2,"DewPoint":9.7,"Pressure":977.9},"SDS0X1":{"PM2.5":8.3,"PM10":9.0},"PressureUnit":"hPa","TempUnit":"C"}}
15:35:05.965 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-05-20T15:35:05","Uptime":"0T00:19:26","UptimeSec":1166,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"iot","BSSId":"XX:XX:XX:XX:XX:XX","Channel":6,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
15:38:32.860 CMD: weblog 4
15:38:32.865 RSL: RESULT = {"WebLog":4}
15:38:33.575 CFG: Saved to flash at F7, Count 37, Bytes 4096
15:38:41.781 WIF: Sending Gratuitous ARP
15:38:44.754 WIF: Checking connection...
15:39:04.798 WIF: Checking connection...
15:39:24.764 WIF: Checking connection...
15:39:41.788 WIF: Sending Gratuitous ARP
15:39:44.784 WIF: Checking connection...
15:39:57.247 HTP: Main Menu
15:40:04.754 WIF: Checking connection...
15:40:24.767 WIF: Checking connection...
15:40:41.769 WIF: Sending Gratuitous ARP
15:40:44.794 WIF: Checking connection...
15:40:49.862 HTP: Console
15:41:04.767 WIF: Checking connection...
15:41:12.565 HTP: Main Menu
15:41:13.807 DMP: AA C5 06 01 01 00 C0 0A D2 AB
15:41:24.791 WIF: Checking connection...
15:41:38.869 DMP: AA C0 4F 00 56 00 C0 0A 6F AB
15:41:39.867 DMP: AA C0 4E 00 55 00 C0 0A 6D AB
15:41:40.866 DMP: AA C0 4E 00 55 00 C0 0A 6D AB
15:41:41.776 WIF: Sending Gratuitous ARP
15:41:41.865 DMP: AA C0 4E 00 55 00 C0 0A 6D AB
15:41:42.864 DMP: AA C5 06 01 00 00 C0 0A D1 AB
15:41:43.781 RSL: STATE = {"Time":"2023-05-20T15:41:43","Uptime":"0T00:26:04","UptimeSec":1564,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"iot","BSSId":"XX:XX:XX:XX:XX:XX","Channel":6,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:04"}}
15:41:43.790 RSL: SENSOR = {"Time":"2023-05-20T15:41:43","BME280":{"Temperature":21.2,"Humidity":48.0,"DewPoint":9.7,"Pressure":977.7},"SDS0X1":{"PM2.5":7.8,"PM10":8.5},"PressureUnit":"hPa","TempUnit":"C"}
15:41:44.796 WIF: Checking connection...
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Power on NodeMCU with BME280 and SDS011 attached and have a look at the Main Menu.

```lua
BME280 Temperature	21.2 °C
BME280 Humidity	47.9 %
BME280 Dew point	9.7 °C
BME280 Pressure	977.8 hPa
SDS0X1 PM 2.5 µm	<--INVALID--> µg/m³
SDS0X1 PM 10 µm	<--INVALID--> µg/m³
```

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Valid values are shown for SDS0X1 as they are in the console.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
[image]
[image]

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2023-05-21:
Will fix today.

arendst, 2023-05-21:
Was fixed in v12.5.0.1 just after the release of v12.5.0

See changelog.

So I suggest you install latest dev branch (v12.5.0.3) to fix your issue.

Thx for reporting.

mrMoe, 2023-05-21:
confirm fixed in version 12.5.0.3

closed 2023-05-21 by mrMoe