arendst/Tasmota #18710: Web-UI displays wrong BMx280 results during deepsleep

FOTH156, 2023-05-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Results from an BME280 are reported flawlessly via the Web UI in normal mode. When operated in deepsleep mode (DST 20, TP 10), this module gives correct results for part of the active cycle – the rest of the time (mainly during deep sleep) wrong results are displayed (see screenshots below). This is a problem because one gets incorrect information most of the time, especially for long deep sleep phases. Nevertheless, DeepSleep behaves as expected, i.e. current drops to 3mA (D1 mini) during sleep and then returns to 70 mA.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): D1 mini, Lolin, ESP12E on adapter board
- [ ] Tasmota binary firmware version number used: 12.5.0 (sensors)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: Tasmotizer 1.2
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

10:33:49.087 CMD: Backlog Template; Module; GPIO 255
10:33:49.089 SRC: WebConsole from 192.168.1.2
10:33:49.090 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
10:33:49.185 SRC: Backlog
10:33:49.187 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
10:33:49.192 MQT: stat/tasmota_0C6AAB/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
10:33:49.398 SRC: Backlog
10:33:49.400 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
10:33:49.405 MQT: stat/tasmota_0C6AAB/RESULT = {"Module":{"18":"Generic"}}
10:33:49.650 SRC: Backlog
10:33:49.652 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here: 
n.a.

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
10:21:31.395 CMD: status 0
10:21:31.396 SRC: WebConsole from 192.168.1.2
10:21:31.398 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
10:21:31.405 MQT: stat/tasmota_0C6AAB/STATUS = {"Status":{"Module":18,"DeviceName":"T, rH, BP","FriendlyName":["Tasmota"],"Topic":"tasmota_0C6AAB","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
10:21:31.442 MQT: stat/tasmota_0C6AAB/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-sensors.bin.gz","RestartReason":"Deep-Sleep Wake","Uptime":"0T00:00:04","StartupUTC":"2023-05-21T08:21:27","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2023-05-20T10:55:53","SaveCount":24,"SaveAddress":"F4000"}}
10:21:31.476 MQT: stat/tasmota_0C6AAB/STATUS2 = {"StatusFWR":{"Version":"12.5.0(sensors)","BuildDateTime":"2023-04-17T08:03:55","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"405/699"}}
10:21:31.496 MQT: stat/tasmota_0C6AAB/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HUAWEI4G","Galaxy-S7"],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000088","00006000","00004000","00000000"]}}
10:21:31.540 MQT: stat/tasmota_0C6AAB/STATUS4 = {"StatusMEM":{"ProgramSize":699,"Free":304,"Heap":20,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9A8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","54000020","0008C081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34,62,65,66","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,98,103,105","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,69,76"}}
10:21:31.591 MQT: stat/tasmota_0C6AAB/STATUS5 = {"StatusNET":{"Hostname":"BME280-4-DS","IPAddress":"192.168.1.46","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"C8:C9:A3:0C:6A:AB","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
10:21:31.614 MQT: stat/tasmota_0C6AAB/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.8","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0C6AAB","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
10:21:31.641 MQT: stat/tasmota_0C6AAB/STATUS7 = {"StatusTIM":{"UTC":"2023-05-21T08:21:31","Local":"2023-05-21T10:21:31","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"06:01","Sunset":"21:31"}}
10:21:31.664 MQT: stat/tasmota_0C6AAB/STATUS10 = {"StatusSNS":{"Time":"2023-05-21T10:21:31","BME280":{"Temperature":20.6,"Humidity":40.1,"DewPoint":6.5,"Pressure":860.1,"SeaPressure":1022.3},"PressureUnit":"hPa","TempUnit":"C"}}
10:21:31.682 MQT: stat/tasmota_0C6AAB/STATUS11 = {"StatusSTS":{"Time":"2023-05-21T10:21:31","Uptime":"0T00:00:04","UptimeSec":4,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"HUAWEI4G","BSSId":"00:BE:3B:B5:50:9C","Channel":1,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F4, Count 24
00:00:00.070 ROT: Mode 1
00:00:00.073 SRC: Restart
00:00:00.075 Project tasmota - T, rH, BP Version 12.5.0(sensors)-2_7_4_9(2023-04-17T08:03:55)
00:00:00.088 SHT: Sensor did not ACK command
00:00:00.111 I2C: BME280 found at 0x76
00:00:00.221 WIF: Checking connection...
00:00:00.222 WIF: Attempting connection...
[... 29 lines cut ...]
Prepare a breadboard with an ESP8266 (FW 12.5.0) wired for deepsleep according to the docs and a BME280 (GPIO4-SDA, GPIO5-SCL) and verify proper operation. Now set DSL 20, TP 10 and monitor the Web-UI to observe the back and forth switching between correct and wrong values.  An instrument to monitor the supply current is helpful in determining begin and end of deepsleep. 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

Correct results
[image]

Wrong results
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

What I tried so far but to no effect:
Tried USB supply and LiIo 
Tried Brave and Firefox
Several combinations of telemetry period and DeepSleepTime made no difference 
Tried multiple ESPs (D1 mini, Lolin, adapter board) with a single BME280. In all cases I ended up with the identical set of wrong values in DeepSleep-mode. 
Tried several BME280, BMP280 and BMP180. The problem is not observed with BMP180. But each BME/P 280 sensor has its own characteristic set of wrong values, very much like a fingerprint.
Eventually I stepped back to 8.5.0 Hannah only to learn, that the problem existed with this version already.
The display of wrong values is shifted to the right by one position
The change from correct to wrong results occurs approx. 2 seconds before the supply current drops to zero. Unfortunately my software knowledge is far from being sufficient to further track down this problem. This is why every support in this matter is highly appreciated. Please let me know if additional information is required!

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-05-22:
During deep sleep there is no Web UI since the device is sleeping. So you can't get anything.

FOTH156, 2023-05-22:
Oops - forgot the expected behaviour:
I expected the last valid measurement to be displayed during deep sleep until after restart new data become available – exactly as it is with the BME180.

barbudor, 2023-05-22:
Whatever the sensor, when the ESP is in DeepSleep it is sleeping, so not responding to Web request
Trying to display the web page when the ESP is in DeepSleep, you should get an error in your browser and that's normal

FOTH156, 2023-05-24:
Please excuse me for being tedious, yet the problem is not resolved. I don't expect any values to be transmitted during DS. My question is, why the BMx280 switches to a wrong set of values approx. two seconds before entering DS. This is not observed with the BMP180. Sorry for any inconvenience this might have caused.

Jason2866, 2023-05-24:
Yes, thats why the issue is still open.

barbudor, 2023-05-24:
Ah that makes sense
If the web UI shows wrong values just before going to sleep it is maybe a problem of timing because the driver reacts to the event 'about to go to sleep' by putting the BMP in sleep mode
So the next refresh of the UI right before the esp goes in deep sleep would get wrong values

Should be easy to solve
I can try to have a look tonight

barbudor, 2023-05-24:
@FOTH156 
Can you please the below test version and confirm ?
[issue-18710_tasmota-sensors_test1.zip](https://github.com/arendst/Tasmota/files/11558105/issue-18710_tasmota-sensors_test1.zip)

As soon as the driver has received the information that the ESP is about to enter DeepSleep and as put the BMP in LowPower/Sleep mode, it disable reading/updating the values and keep the last ones until next restart.

barbudor, 2023-05-24:
I won't be availble in the next couple of days but I'm pretty confident on the results so I have already created the PR
Please do not merge until @FOTH156 has confirmed it's ok

FOTH156, 2023-05-24:
Will do some thorough testing tomorrow and report the results. Anyway a big THX to you for taking care of this issue!

FOTH156, 2023-05-25:
After extensive testing I can confirm that with the test version the problem has vanished and the issue can be closed. BTW, do I have to close the thread, or are you going to do this ? Thank you very much for your patience and efforts ! - Gerd

closed 2023-05-25 by arendst