arendst/Tasmota #18730: BME280 temperature readings too high compared to DS18B20 on same device and other references

schnudeldudel, 2023-05-28:
### PROBLEM DESCRIPTION
I have a Lingan SWA1 powerswitch to which i added 2 sensors: a DS18B20 sesnor and a BME280 sensorboard. The switch is plugged in on my balcony.
Both sensors are recognized well with tasmota.
But apparantly, the BME280 temperature readings are constantly way to high (see info below).
To get a reference value I build a ESP32 breadboard setup with another BME280 sensor on it and also a weather station (offline with display).

Result: 
ESP32 breadboard BME280: correct temp. values; correct pressure readings. 
DS18B20 temp. values from tasmota device: correct.
BME280 connected to tasmota device: temp. values too high; pressure readings correct

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Ligan SWA1 
- [ x] Tasmota binary firmware version number used: 12.5.0
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

07:30:40.173 MQT: stat/tasmota_swa1_01/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
07:30:40.385 MQT: stat/tasmota_swa1_01/RESULT = {"Module":{"18":"Generic"}}
07:30:40.592 MQT: stat/tasmota_swa1_01/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"640":"I2C SDA"},"GPIO3":{"0":"None"},"GPIO4":{"288":"Led1"},"GPIO5":{"224":"Relay1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"608":"I2C SCL"},"GPIO13":{"32":"Button1"},"GPIO14":{"1312":"DS18x201"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
07:31:12.993 CMD: Backlog Rule1; Rule2; Rule3
07:31:13.078 MQT: stat/tasmota_swa1_01/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
07:31:13.295 MQT: stat/tasmota_swa1_01/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
07:31:13.498 MQT: stat/tasmota_swa1_01/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
07:31:40.365 CMD: Status 0
07:31:40.373 MQT: stat/tasmota_swa1_01/STATUS = {"Status":{"Module":18,"DeviceName":"Balkon","FriendlyName":["Balkon"],"Topic":"tasmota_swa1_01","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
07:31:40.381 MQT: stat/tasmota_swa1_01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://192.168.0.59:9541/data/firmwares/tasmota-sensors.bin.gz","RestartReason":"Software/System restart","Uptime":"7T15:22:43","StartupUTC":"2023-05-20T15:08:57","Sleep":50,"CfgHolder":4617,"BootCount":70,"BCResetTime":"2020-12-25T11:06:05","SaveCount":305,"SaveAddress":"FB000"}}
07:31:40.387 MQT: stat/tasmota_swa1_01/STATUS2 = {"StatusFWR":{"Version":"12.5.0(sensors)","BuildDateTime":"2023-04-17T08:03:55","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"349/699"}}
07:31:40.394 MQT: stat/tasmota_swa1_01/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["edel",""],"TelePeriod":10,"Resolution":"55A180C0","SetOption":["00008009","2805C80001000600003C5A0A002800000000","00000080","00006000","00005000","00000000"]}}
07:31:40.415 MQT: stat/tasmota_swa1_01/STATUS4 = {"StatusMEM":{"ProgramSize":699,"Free":304,"Heap":16,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15400E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9A8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","54000020","0008C081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34,62,65,66","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,98,103,105","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,69,76"}}
07:31:40.423 MQT: stat/tasmota_swa1_01/STATUS5 = {"StatusNET":{"Hostname":"tasmota-swa1-01-5929","IPAddress":"192.168.0.20","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.50","DNSServer2":"0.0.0.0","Mac":"68:C6:3A:87:B7:29","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
07:31:40.429 MQT: stat/tasmota_swa1_01/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_87B729","MqttUser":"DVES_USER","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
07:31:40.438 MQT: stat/tasmota_swa1_01/STATUS7 = {"StatusTIM":{"UTC":"2023-05-28T06:31:40","Local":"2023-05-28T07:31:40","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"04:54","Sunset":"20:40"}}
07:31:40.451 MQT: stat/tasmota_swa1_01/STATUS10 = {"StatusSNS":{"Time":"2023-05-28T07:31:40","DS18B20":{"Id":"000005FA90D2","Temperature":14.9},"BME280":{"Temperature":27.2,"Humidity":28.4,"DewPoint":7.3,"Pressure":963.7},"PressureUnit":"hPa","TempUnit":"C"}}
07:31:40.459 MQT: stat/tasmota_swa1_01/STATUS11 = {"StatusSTS":{"Time":"2023-05-28T07:31:40","Uptime":"7T15:22:43","UptimeSec":660163,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"ON","Wifi":{"AP":1,"SSId":"edel","BSSId":"44:4E:6D:EE:B4:2A","Channel":1,"Mode":"11n","RSSI":92,"Signal":-54,"LinkCount":2,"Downtime":"0T00:01:52"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Connect BME280 and DS18B20 Sensor to tasmota device (here:Lingan SWA1 Power Switch) and compare sensor values.
I also have an external (with display and no esp) weather station to get reference values.
I also have another BME280 which runs on a breadboard with ESP32 and Adafruit driver for BME280.
 
The BME280-Tasmota (connected to Lingan SWA1 oer switch) pressure values are correct and similar to the values from the ESP32 breadboard and the weather station.

But the BME280 temperature values from tasmota device have a huge offset compared to the DS18B20 values and the values from the breadboard-BME280 and also the weather station.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
BME280 connected to tasmota device should provide "same" (=correct) temp. values.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

I read out both values (DS18B20 & BME280) via MQTT from the tasmota device and sent to home assistant for comparision.
The offset is apparent.
[image]

Here you see the offset of approx. 10 deg.:

[image]

Here I put on the different readings in a dashboard style. Pressure values are almost equal. BME280 tasmta temperature values are way off:

[image]

### ADDITIONAL CONTEXT
Like said, the prerssure readings from BME280 seem correct (at least they match the same reading from my ESP32 based breadboard setup), but temperature reading is wrong.

I also interchanged the BME280 hardware between the tasmota device and the ESP32-breadboard. Same result. So the issue seems not related to the sensor hardware.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-05-28:
Most DS18B20 are dubious clones of "questionable" (to be kind) quality, and not useful as a "reliable reference". Physical layout of your hardware can also make a major difference, as the ESP chip acts as a radiator, giving significant local heating.

schnudeldudel, 2023-05-28:
Hi,

Thanks for the comment. I know about the DS18B20 fake sensor issues but values from my DS18B20 sensors seem nevertheless correct.

Self heating of BME280 is also known ,but according to my setup ,it seems not possible to have 10deg deviation.

Below pictures of my hardware. The BME280 sensor is external accessible.

[image]

[image]

sfromis, 2023-05-28:
The BME280 self-heating is largely a myth without real-world significance.

I was **instead** talking about the radiator assembly you have:
1) The ESP chip generating a quite a bit of heat
2) The relay coil inside a plug also uses power, and generates heat when engaged
3) The power supply inside is not highly efficient; another heat source
4) Using the plug, expect some amount of heat from resistance when plugging in.
It does not magically solve the issue to have the BME280 poking out through a hole, it still depends on your hardware layout, and flow of heat from all that stuff inside. I observe what looks like your DS18B20 well outside the heat box, much better.

With good hardware layout, I'm using quite a few BME280 with Tasmota without any hint of "surprising" temperature values. Of course, I'm not making problematic designs like expecting a plug to be a good base for builds with temperature sensors.

schnudeldudel, 2023-05-28:
Hi,

good point. But I also tried the setup with kind of a breakout/extension cable to have the BME280 sensor several centimeters apart from the power plug. Same results.

As said, the same BME280 hardware board works without this huge temperature deviation in my other esp32 based setup.

btw: I double checked and I use the SPARKFUN BME280 library in my reference esp32 breadboard setup (not Adafruit as mentioned earlier).
Not sure about the different implementations of Adafruit vs SPARKFUN vs tasmota in regards to temperature reading ...

[image]

sfromis, 2023-05-28:
Well, you say that it works better on your breadboard. That is very much to be expected, and only makes the case for the plug-based design physically not working. You could flash Tasmota on that ESP32 to verify that Tasmota is not the source of much different readings.

My experience with Tasmota+BME280 with both ESP32 and ESP8266 is that it just works reasonably well, even coupled with other sensors on the same board for direct comparison. [image]