arendst/Tasmota #18740: Berry script adding a button to the configuration menu works but event isn´t working

nobodyman1, 2023-05-29:
### PROBLEM DESCRIPTION
I want to have an additional button in the config menu.
Therefore I tried the example [Adding a button to the Main menu](https://tasmota.github.io/docs/Berry-Cookbook/#adding-a-button-to-the-main-menu).
The button is visible in the config menu. But if I press the button nothing happens (no output in the log).
The button in the main menu works as expected.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32C3
- [x] Tasmota binary firmware version number used: 12.5.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
1. Insert the simplified script from documentation into berry scripting console.
2. Open config menu and press "Toggle Conf" button.
3. Open console and check log. 

```
import webserver # import webserver class

class MyButtonMethods : Driver

  #- create a method for adding a button to the main menu -#
  def web_add_main_button()
    webserver.content_send("<p></p><button onclick='la(\"&m_toggle_main=1\");'>Toggle Main</button>")
  end

  #- create a method for adding a button to the configuration menu-#
  def web_add_config_button()
    #- the onclick function "la" takes the function name and the respective value you want to send as an argument -#
    webserver.content_send("<p></p><button onclick='la(\"&m_toggle_conf=1\");'>Toggle Conf</button>")
  end

  #- As we can add only one sensor method we will have to combine them besides all other sensor readings in one method -#
  def web_sensor()

    if webserver.has_arg("m_toggle_main")
      print("main button pressed")
    end

    if webserver.has_arg("m_toggle_conf") # takes a string as argument name and returns a boolean
      print("config button pressed")
    end

    end
  end
d1 = MyButtonMethods()
tasmota.add_driver(d1)
```

### EXPECTED BEHAVIOUR
"config button pressed" visible in the log.

### SCREENSHOTS

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-05-29:
From what I can see, using `web_sensor()` is only useful for the main page, as the purpose of that callback is to allow you to add sensor values rendered on the main page. And I confirmed that `web_sensor()` is not called for the configuration page button. Not sure what the intention for such a button as in the example is, but a tool like Auto-configration is instead handling the configuration page button via `webserver.on()`, and a corresponding form action in the button.

Maybe @s-hadinger can suggest how configuration page buttons are supposed to work? Looking for pertinent callback, I did not spot a good candidate... It did not work in `every_second()`; was not really expecting that anyway.

s-hadinger, 2023-05-29:
For the configuration page, you need to define an HTML form.

Example from Matter UI:
```berry
  def web_add_config_button()
    import webserver
    webserver.content_send("<p><form id=ac action='matterc' style='display: block;' method='get'><button>Configure Matter</button></form></p>")
  end
```

You also need to register to the sub-url:
```berry
  #- ---------------------------------------------------------------------- -#
  # respond to web_add_handler() event to register web listeners
  #- ---------------------------------------------------------------------- -#
  #- this is called at Tasmota start-up, as soon as Wifi/Eth is up and web server running -#
  def web_add_handler()
    import webserver
    #- we need to register a closure, not just a function, that captures the current instance -#
    webserver.on("/matterc", / -> self.page_part_mgr(), webserver.HTTP_GET)
    webserver.on("/matterc", / -> self.page_part_ctl(), webserver.HTTP_POST)
  end
```