arendst/Tasmota #18743: SR04 shows wrong distance on ESP32-S2

christianh17, 2023-05-29:
### PROBLEM DESCRIPTION
The distance sensor (ultrasound) gives back too long distances. e.g. I expect around 44 cm, but it says 172 cm.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): dev Board with ESP32-S2
- [X] Tasmota binary firmware version number used: 12.5.0.3(tasmota32) built: 2023-05-22T10:56:02
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: esptool.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:56:35.124 RSL: RESULT = {"NAME":"Wemos Teleinfo","GPIO":[1,1,1,1,1,1,1,1376,1,1,1,5632,1,1,1,1,1,1,1,1,1,1,640,1,608,1,0,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":1}
18:56:35.354 RSL: RESULT = {"Module":{"1":"ESP32S2"}}
18:56:35.602 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"1888":"SR04 Ech/RX"},"GPIO17":{"0":"None"},"GPIO18":{"1856":"SR04 Tri/TX"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"},"GPIO40":{"0":"None"},"GPIO41":{"0":"None"},"GPIO42":{"0":"None"},"GPIO43":{"0":"None"},"GPIO44":{"0":"None"},"GPIO45":{"0":"None"},"GPIO46":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
18:57:50.071 CMD: Backlog Rule1; Rule2; Rule3
18:57:50.106 RSL: RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
18:57:50.321 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
18:57:50.551 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:58:30.197 CMD: status 0
18:58:30.201 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_46F5C6","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:58:30.235 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32s2.bin","RestartReason":"Software reset digital core","Uptime":"0T00:09:22","StartupUTC":"2023-05-29T17:49:08","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2023-05-29T17:15:31","SaveCount":39}}
18:58:30.269 RSL: STATUS2 = {"StatusFWR":{"Version":"12.5.0.3(tasmota32)","BuildDateTime":"2023-05-22T10:56:02","Core":"2_0_9","SDK":"4.4.4.20230310","CpuFrequency":240,"Hardware":"ESP32-S2FNR2","CR":"412/699"}}
18:58:30.282 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Herrmann_19c",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
18:58:30.323 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1416,"Free":1463,"Heap":115,"StackLowMark":4,"PsrMax":2045,"PsrFree":2028,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":80,"FlashMode":"DIO","Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","080840D2","20200000","D4BC482D","000A80B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,68,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77"}}
18:58:30.380 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-46F5C6-5574","IPAddress":"192.168.17.106","Gateway":"192.168.17.254","Subnetmask":"255.255.255.0","DNSServer1":"192.168.17.55","DNSServer2":"192.168.17.1","Mac":"48:27:E2:46:F5:C6","IP6Global":"2a02:8108:8f80:3f3c:4a27:e2ff:fe46:f5c6","IP6Local":"fe80::4a27:e2ff:fe46:f5c6","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:58:30.416 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_46F5C6","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:58:30.442 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-05-29T17:58:30","Local":"2023-05-29T18:58:30","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"04:53","Sunset":"20:41"}}
18:58:30.460 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-05-29T18:58:30","SR04":{"Distance":172.6}}}
18:58:30.466 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-05-29T18:58:30","Uptime":"0T00:09:22","UptimeSec":562,"Heap":115,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":36},"Wifi":{"AP":1,"SSId":"Herrmann_19c","BSSId":"CC:CE:1E:83:17:8E","Channel":1,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
no output generated

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
attach a SR04 sensor to GPIOs 16 and 18, configure and measure

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I'd like to have correct distances displayed

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
n/a

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I think the function "millis" does not work correct on my device, perhaps the CPU runs 4 times faster and so it measures the wrong time (4 times too long). That could be the reason why the distance ist wrong.

**(Please, remember to close the issue when the problem has been addressed)**

christianh17, 2023-06-04:
I asked teckel12 (he maintains the library) and his answer was:
4 times the distance sounds like your microcontroller timer is set to the wrong multiplier. Either resolve the multiplier, or divide the result by 4 until such time you have it resolved.
Does anyone have an idea how to do this?

sfromis, 2023-06-04:
Just made a small test with ESP32-S2 + HC-SR04, without seeing any anomalies in indicated distance, when making sure to aim at a sufficiently large target. That toy sensor is not great at disambiguating a scenario with multiple distances.

Did you do a proper hardware setup? Like 5V power instead of the usual 3.3V.