arendst/Tasmota #18749: SetOption13 on/off for individual buttons

Jonsonmon, 2023-05-29:
I Have a switch that including curtain and 1 GANG light switch.
I Have to set setoption13 0, because of the curtain.
but my light button is suffering because of that. there is also a delay.
is there a way to set setoption13 1, only for the light button ?

sfromis, 2023-05-29:
No, SetOption13 is for all button pins. But you can likely get what you want by configuring as a Switch input instead, and selecting a good SwitchMode.

And questions should not be asked as an "issue", but in "Discussions".

Jonsonmon, 2023-05-29:
Thanks
I'm not finding the correct combination for my situation... Could you please help me a little further?
And how can I move this to the 'Discussion' section?

sfromis, 2023-05-29:
Just reconfig the input gpio as a Switch instead of Button. Default [SwitchMode](https://tasmota.github.io/docs/Buttons-and-Switches/#switchmode) of 0 (toggle) is probably what you want.

Someone will probably move the thread tomorrow ;-)

Jonsonmon, 2023-05-29:
> 

`SwitchMode` is already configured as 0