arendst/Tasmota #18759: ESP-C3-12F "invalid header 0xffffffff"

pploume, 2023-05-31:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After flashing with the lastest available tasmota32c3.bin, the chip don't boot with "invalid header 0xffffffff" message in serial console output

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____
- [ x] Tasmota binary firmware version number used: _____
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
use "tasmotizer1-2.exe" (not problem to upload the tasmota bin to the chip

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
error message : invalid header 0xffffffff

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
console output :
[12:04:11]invalid header: 0xffffffff
[12:04:11]invalid header: 0xffffffff
[12:04:11]invalid header: 0xffffffff
[12:04:14]invalid header: 0xfESP-ROM:esp32c3-api1-20210207
[12:04:14]Build:Feb  7 2021
[12:04:14]rst:0x1 (POWERON),boot:0x4 (DOWNLOAD(USB/UART0/1))
[12:04:14]waiting for download

download is OK but same problem 

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

chip alone, not on a board (to be soldered to replace a Tuya on a powermeter ... later)

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-05-31:
For initial wired flashing, `tasmota32c3.bin` cannot work, you have to use `tasmota32c3.factory.bin` or `tasmota32c3cdc.factory.bin`. It is recommended to use the [Tasmota Web installer](https://tasmota.github.io/install/) to get the details right, Tasmotizer does not support ESP32.

pploume, 2023-06-01:
@sfromis 
Thanks for the comment.
I (previously) upload the tasmota32c3.factory.bin via epstool/tasmotizer ... with the same error.
...or I've made a mistake somewhere/somehow
:)

sfromis, 2023-06-01:
Your steps "to reproduce" says that you used Tasmotizer, which cannot be used for ESP32.

In general, be very exact about details of the specific steps which would reproduce the error, it is not useful to just waffle about a list of things you may have tried.

And maybe follow the recommendation I already gave.