arendst/Tasmota #18782: Console not working when Web log level is set to 0-None or 1-Error

telece2, 2023-06-02:
### PROBLEM DESCRIPTION
Console is either:
- blank (no green text at all)
- showing only 1 row of text:
`00:00:00.050 CFG: Loaded from flash at F9, Count 37`
Commands given to the console seems to execute correctly (e.g. telemetry 60 - you can see the new value in Configure Logging) but there is no output in the multi-line console green text frame.
Please note that all below required console commands were executed with Web log level set to >= 2, otherwise there is no output.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
01:39:05.451 CMD: Backlog Template; Module; GPIO 255
01:39:05.518 MQT: tasmota/bw-shp6-6/stat/RESULT = {"NAME":"BW-SHP6 10A","GPIO":[576,1,320,1,0,2720,0,0,2624,32,2656,224,0,0],"FLAG":0,"BASE":45}
01:39:05.736 MQT: tasmota/bw-shp6-6/stat/RESULT = {"Module":{"0":"BW-SHP6 10A"}}
01:39:05.991 MQT: tasmota/bw-shp6-6/stat/RESULT = {"GPIO0":{"576":"LedLink_i"},"GPIO1":{"0":"None"},"GPIO2":{"320":"Led_i1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"32":"Button1"},"GPIO14":{"2656":"HLWBL CF1"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
23:32:04.760 CMD: Backlog Rule1; Rule2; Rule3
23:32:04.828 MQT: tasmota/ikea-1/stat/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
23:32:05.040 MQT: tasmota/ikea-1/stat/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
23:32:05.242 MQT: tasmota/ikea-1/stat/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
23:33:34.301 CMD: Status 0
23:33:34.309 MQT: tasmota/ikea-1/stat/STATUS = {"Status":{"Module":1,"DeviceName":"Prelungitor 1 IKEA 6 prize","FriendlyName":["Prelungitor 1 IKEA 6 prize"],"Topic":"ikea-1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
23:33:34.316 MQT: tasmota/ikea-1/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-lite.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:22:45","StartupUTC":"2023-06-02T22:10:49","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2023-06-02T23:05:51","SaveCount":30,"SaveAddress":"F6000"}}
23:33:34.324 MQT: tasmota/ikea-1/stat/STATUS2 = {"StatusFWR":{"Version":"12.5.0(lite)","BuildDateTime":"2023-04-17T08:03:57","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285N08","CR":"464/699"}}
23:33:34.329 MQT: tasmota/ikea-1/stat/STATUS3 = {"StatusLOG":{"SerialLog":1,"WebLog":2,"MqttLog":1,"SysLog":1,"LogHost":"10.10.0.250","LogPort":514,"SSId":["tlc-private2","tlc-private"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A00192800000000","00000080","00006000","00004000","00000000"]}}
23:33:34.339 MQT: tasmota/ikea-1/stat/STATUS4 = {"StatusMEM":{"ProgramSize":493,"Free":508,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","0F000782","04108001","00000002","00000000","00000000","00000000","00000000","04000000","00000080"],"Drivers":"1,2,4,9,10,12,16,20,21,62","Sensors":""}}
23:33:34.347 MQT: tasmota/ikea-1/stat/STATUS5 = {"StatusNET":{"Hostname":"ikea-1","IPAddress":"10.10.0.241","Gateway":"10.10.0.1","Subnetmask":"255.255.255.0","DNSServer1":"10.10.0.1","DNSServer2":"0.0.0.0","Mac":"60:01:94:E1:18:E9","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
23:33:34.352 MQT: tasmota/ikea-1/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.10.0.250","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E118E9","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
23:33:34.358 MQT: tasmota/ikea-1/stat/STATUS7 = {"StatusTIM":{"UTC":"2023-06-02T22:33:34","Local":"2023-06-02T23:33:34","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"04:50","Sunset":"20:45"}}
23:33:34.362 MQT: tasmota/ikea-1/stat/STATUS10 = {"StatusSNS":{"Time":"2023-06-02T23:33:34"}}
23:33:34.368 MQT: tasmota/ikea-1/stat/STATUS11 = {"StatusSTS":{"Time":"2023-06-02T23:33:34","Uptime":"0T00:22:45","UptimeSec":1365,"Vcc":3.466,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"tlc-private2","BSSId":"64:70:02:E7:39:C2","Channel":1,"Mode":"11n","RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
[not applicable, the issue appears only when web log is set to 0 or 1]

```

### TO REPRODUCE
1. Go to Configuration/Configure Logging/Web log level --> set to 0 or 1
2. Go to Console and you will notice there is either no output, or only one row of text:
`00:00:00.050 CFG: Loaded from flash at F9, Count 37`

### EXPECTED BEHAVIOUR
Console should work properly regardless of Web log level setting

### SCREENSHOTS
01_configure_logging.jpg
[image]

02_Console_loaded_from_flash.jpg
[image]

03_Console_empty.jpg
[image]

04_Console_OK_after_setting_Web_log_level_to_2_3_4.jpg
[image]

### ADDITIONAL CONTEXT
None.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-06-02:
Seems to be work as designed. If you ask for no info messages, you get no info messages. Thus, it is working properly. Sure, the console may get less useful for working with commands, but the primary control methods is intended to be MQTT, and if you want to work with manual commands, just do not go below log level 2. Most of the time, I do not see log level 1 as very useful, anyway.

telece2, 2023-06-03:
On second thought, you are right, it seems to work as designed. I was expecting to see at least the response messages when running commands in the console, regardless of the Web log level setting. Thanks.

stefanbode, 2023-06-03:
No, as @sfromis mentioned any message has it's connected log level. And 0 and 1 is reserved for errors and warnings. Therefore an empty log if there is no one of this is the intended behavior.

closed 2023-06-03 by telece2