arendst/Tasmota #18783: reset esp32 con tasmota 12.5

mascolo74, 2023-06-03:


mascolo74, 2023-06-03:
chiedo scusa errore di tastiera per commento sopra.
installazione tasmota 12.5 su esp 32, comportamento strano su due schedine.
installazione ok , inserimento wifi e collegamento wifi ok, inserimento gpio 2/4/21/23 pulsante gpio 4/ 25/26/27 ok riavvio del modulo si perde tutta la configurazione bisogna rifarla compreso il wifi.

barbudor, 2023-06-03:
No comprendo

This is an English only forum
Please use Google translate before posting as other users do

stefanbode, 2023-06-03:
I apologize keyboard error for comment above.
tasmota 12.5 installation on esp 32, strange behavior on two cards.
installation ok, wifi insertion and wifi connection ok, gpio 2/4/21/23 insertion gpio 4/25/26/27 button ok restarting the module the whole configuration is lost, you have to redo it including the wifi.

Hmm even the translate is not clear for me.... looks like it did not save the config. Pls check "savedata" in the console and if it is 0 set it to 1. Also multiple restarts because of a electrical issue will reset the device after some restarts.

closed 2023-06-03 by Jason2866