arendst/Tasmota #18802: ESP32 + Bluetooth - Disabling bluetooth does not revert WiFi.setSleep to what it should be according to SLEEP / SETOPTION127

anonymous-one, 2023-06-06:
Please excuse my terminology here, I will do my best to explain:

**ESP32 without bluetooth compile OR bluetooth disabled via web panel behavoir:**

WiFi.setSleep is set according to SLEEP OR SETOPTION127.

**This works exactly as it should.**

SLEEP== 0? WiFi.setSleep(false)
SETOPTION127 == 1? WiFi.setSleep(false)

---

**ESP32 with bluetooth compiled in AND bluetooth disabled via webpanel:**

The behavior is as above basically, WiFi.setSleep is set according to SLEEP and SETOPTION127.

**This works exactly as it should.**

---

**ESP32 with bluetooth compiled in AND bluetooth enabled:**

The bluetooth driver will set WiFi.setSleep(true) regardless of what SLEEP and SETOPTION127 is set to. This is correct as well.

**Here is the issue:**

If bluetooth is enabled, and later disabled within the web panel WiFi.setSleep is not revered to what ever the state should be as per SLEEP / SETOPTION127.

**Expected behavior:**

Setting bluetooth to false via the web panel (and I guess console, mqtt etc)... Should revert WiFi.setSleep to whatever SLEEP / SETOPTION127 dictate.

**Why this matters:**

Being able to dynamically turn bluetooth on and off, and have wifi.setSleep follow this status will enable me to choose what matters more to me:

Are we home? Let make the lights which are purely wifi controlled (eg: no physical switch directly connected), slightly more responsive.

Are we away? Lets make these lights pick up my ibeacon's as we are not home, and all "you have arrived" actions are more likely to be picked up quicker by having these ble gateways on.

Hope this makes sense!

stefanbode, 2023-06-07:
I would assume this should be easy to fix if we use the `static int StopBLE(void){` function and inside this function reset the sleep status to where it should be. I will not do a PR because the chances to mess up are quite high. I do not have insights into the dependencies in the xdrv_79_esp32_ble.ino file.

stefanbode, 2023-06-07:
Maybe you can test if this additional lien solves your problem:
[image]

stefanbode, 2023-06-07:
https://github.com/arendst/Tasmota/compare/development...stefanbode:Sonoff-Tasmota:patch-5 here a patch you can compile

anonymous-one, 2023-06-07:
> Please excuse my terminology here, I will do my best to explain:
> 
> **ESP32 without bluetooth compile OR bluetooth disabled via web panel behavior on fresh restart:**
> 
> WiFi.setSleep is set according to SLEEP OR SETOPTION127.
> 
> **This works exactly as it should.**
> 
> ```
> SLEEP == 0? WiFi.setSleep(false)
> SETOPTION127 == 1? WiFi.setSleep(false)
> Else : WiFi.setSleep(true)
> ```
> 
> **ESP32 with bluetooth compiled in AND bluetooth enabled:**
> 
> The bluetooth driver will set WiFi.setSleep(true) regardless of what SLEEP and SETOPTION127 is set to. This is correct as well.
> 
> **Here is the issue:**
> 
> If bluetooth is enabled, and later disabled within the web panel WiFi.setSleep is not revered to what ever the state should be as per SLEEP / SETOPTION127. The only way to revert it is to change bluetooth to disabled, and restart the device. At which point it will be set correctly.
> 
> **Expected behavior:**
> 
> Setting bluetooth to false via the web panel (and I guess console, mqtt etc)... Should revert WiFi.setSleep to whatever SLEEP / SETOPTION127 dictate.
> 
> **Why this matters:**
> 
> Being able to dynamically turn bluetooth on and off, and have wifi.setSleep follow this status will enable me to choose what matters more to me:
> 
> Are we home? Let make the lights which are purely wifi controlled (eg: no physical switch directly connected), slightly more responsive. _(Please lets not make this a discussion about if wifi.setSleep(false/true) makes a difference... If it is wifi.setSleep or the fact that the ESP32 has ONE 2.4ghz radio for both wifi and bluetooth that is shared when bluetooth enabled... One thing is for sure, disabling bluetooth makes network control of the lights slightly more responsive... There is no doubt about this.)_
> 
> Are we away? Lets make these lights pick up my ibeacon's as we are not home, and all "you have arrived" actions are more likely to be picked up quicker by having these ble gateways on. Plus we are not home, so network control of the lights being slightly more delayed is not an issue as no one is there to press one of the wifi enabled physical switches which control the light via ip control.
> 
> Hope this makes sense!

I have slotted this in for sometime Friday (~48 hours from now). Will post as soon as I get a chance to compile a build with this.

Thank you!

anonymous-one, 2023-06-07:
I have slotted this in for sometime Friday (~48 hours from now). Will post as soon as I get a chance to compile a build with this.

Thank you!

> [development...stefanbode:Sonoff-Tasmota:patch-5](https://github.com/arendst/Tasmota/compare/development...stefanbode:Sonoff-Tasmota:patch-5) here a patch you can compile

Jason2866, 2023-06-15:
@anonymous-one Is there a status update?

closed 2023-07-22 by github-actions[bot]