arendst/Tasmota #18828: ZB-GW03 cannot add zigbee devices through routing nodes

xsp1989, 2023-06-09:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): ZB-GW03
- [x] Tasmota binary firmware version number used: 12.3.1/12.4.0/12.5.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
{"NAME":"ZB-GW03-V1.2","GPIO":[0,0,3552,0,3584,0,0,0,5793,5792,320,544,5536,0,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,608,640,32,0,0,0,0,0],"FLAG":0,"BASE":1}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

When using ZB-GW03, when adding a zigbee device, if the network access is requested through a routing node, an additional wrong link key will be sent, and then the correct key will be transmitted through the tuner, but the routing device cannot process the key correctly at this time

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Through the packet capture analysis of ubiqua software, a new device applies for network access through another route (0xB35D), and 0xB35D assigns the network address 0xB5B0, and then the coordinator sends a wrong link key to 0xB35D (all key values are 0), and then Continue to transmit the correct key to the 0xB5B0 device through the tunnel, but because the router (0xB35D) cannot correctly parse the tunnel key, it cannot transmit the correct key to 0xB5B0.

The correct interaction process should be to directly send the key to 0xB5B0 through the tunnel.

I am attaching the packet I grabbed via ubiqua, the decryption key is 00:00:E4:96:A0:B7:65:57:00:00:E4:96:A0:B7:65:57 and
5A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39

This data packet is redundant and erroneous data, but I am not completely sure whether this redundant data is sent by tasmota or NCP itself, but I speculate that it should have something to do with tasmota
[image]

This zip file is a special format file of ubiqua. Although you can open it with some database software, you cannot analyze the data inside.
[join fail.zip](https://github.com/arendst/Tasmota/files/11699548/join.fail.zip)

**(Please, remember to close the issue when the problem has been addressed)**

xsp1989, 2023-06-12:
I used the tcp method to transparently transmit the ezsp command, and connected to the Z2M. Through the packet capture analysis, the Z2M can access the network normally. Maybe some configurations are different, resulting in different results. Because tasmota and Z2M use the same ncp firmware.

[image]

MattWestb, 2023-06-12:
Tasmota is not using standard Zigbee 3 security that Z2M and ZHA is using and therefor is messing up the network and TC-Link key. More no Zigbee 3 devices like it but more strict Ziogbee 3 devices is not working at all then the link keys is not working.

I have also seen some sniffs then coordinator is sending one not valid key and later sending one OK one and some device is it OK other is leaving the network.

xsp1989, 2023-06-12:
This wrong key has nothing to do with zigbee 3.0. Zigbee 3.0 security is the process after the "device announce" message. After zigbee3.0 sends the "device announce" on the device side, it will continue to send "request key", and the coordinator will Resend the key, and then send a confirmation message. According to the packet capture data analysis, it should be that the coordinator performed error handling after receiving the "update device" message.

xsp1989, 2023-06-12:
In EZSP, there is an important command, which is related to network processing, but it is a little strange that I did not find any processing code related to setpolicy in the source code of NCP. Of course, it may not be exposed to application processing.

[image]

xsp1989, 2023-06-14:
This problem needs to use the 0x00AB command and configure the 0x0A parameter to 0x0110, which can solve this problem

closed 2023-06-14 by xsp1989

s-hadinger, 2023-06-29:
@xsp1989 Sorry I was busy with Matter and only check this one now. Do I need to change these options in Tasmota? Honestly configuring EZSP security options was a painful process of trial and error.

xsp1989, 2023-07-01:
I think the option of 0x0068 should be modified. Although zigbee 3.0 supports many secure network access methods, I think that most of the standard equipment can access the network in the traditional way: that is, the coordinator allows access to the network, the device enters the network access mode, and the coordinator Through 5A:69... this key is used to transmit the key to the newly connected device. If you don't want to make more extensive changes, I think you can delete the option EMBER_PRECONFIGURED_NETWORK_KEY_MODE. But I used the parameters of z3gateway and it works great so far. I can push a modified branch of code if needed.