arendst/Tasmota #18840: Security issue in AP-mode

scanheads, 2023-06-10:
As the link for security-related problems only leads to nowhere (got a "404 Object not found!
The requested URL was not found on this server. If you entered the URL manually please check your spelling and try again."), I report this here. There is a serious problem in Tasmota's AP-mode which makes it easy as fuck to take over an existing Tasmota device. This can be done quite easy:

1. Jam all existing WiFi networks close to the Tasmota-device to be attacked (this is extremely simple as there are ready-to-use applications available e.g. for the ESP32)
2. Now Tasmota comes up in AP-mode
3. Exclude the Tasmota-AP from the WiFi networks to be jammed
4. Connect to the Tasmota-device and do whatever evil you want to do (turn it on/off, change configuration, damage configuration, change stored values, configure it to connect to an other WiFi network so take over control permanently, set some Web-passwords,...)

The problem with Tasmota is, there is no possibility to set some password for the AP-mode. Whenever the AP-mode is enabled, it comes as open network without any protection and gives full access to everything.

So the solution is also very simple: give the possibility to configure a password for the default/fallback AP so that it no longer comes up as open network.

Yes, I'm aware of the web admin password, but this is not a solution but a pain in the a**. It turns on a password protection for normal operation, so whenever one wants to access the web interface regularly, one has to enter this password. Or in other words: even people that are willing to configure Tasmota in a secure way will not make use of this "feature" as it disturbs them way too much during regular operation.

barbudor, 2023-06-10:
> The problem with Tasmota is, there is no possibility to set some password for the AP-mode

Wrong
```
#define WIFI_AP_PASSPHRASE     ""                // AccessPoint passphrase. For WPA2 min 8 char, for open use "" (max 63 char).
```
Documented in the changelog

> As the link for security-related problems only leads to nowhere (got a "404 Object not found

If you found a problem in the docs, please open an issue in the doc repository at https://github.com/tasmota/docs

Better, you can also contribute to the doc to improve it by clicking on the pencil button on the top right of the doc page.

Thanks in advance

sfromis, 2023-06-10:
Additionally, with the intended AP unavailable, Tasmota does not come up in AP mode, if using the recommended `WifiConfig 4` (or 5) - which you really should if wanting rudimentary security.