arendst/Tasmota #18841: Shutter not working as expected, after update to latest (12.5)

HerrFrutti, 2023-06-10:
### PROBLEM DESCRIPTION
I had working shutters, that can be open and closed via mqtt. I used the `ShutterPosition` command with the `OPEN` and `CLOSE` commands. With the update to 12.5 those commands are always opening the shutter. As workaround I changed those commands to `0` and `100` which is working.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32
- [x] Tasmota binary firmware version number used: 12.5.0(tasmota32)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:

### EXPECTED BEHAVIOUR
ShutterPosition1 UP -> Shutter opens
ShutterPosition1 DOWN -> Shutter closes

ShutterPosition1 OPEN -> Shutter opens
ShutterPosition1 CLOSE -> Shutter closes

### ADDITIONAL CONTEXT
```
15:34:57.340 CMD: ShutterPosition1 STOP
15:34:57.353 MQT: stat/ESP-Velux/RESULT = {"POWER1":"OFF"}
15:34:57.360 MQT: stat/ESP-Velux/POWER1 = OFF
15:34:57.368 MQT: stat/ESP-Velux/SHUTTER1 = 83
15:34:57.375 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":83,"Direction":0,"Target":83,"Tilt":0}}
15:35:03.523 CMD: ShutterPosition1 CLOSE
15:35:03.541 MQT: stat/ESP-Velux/RESULT = {"POWER1":"ON"}
15:35:03.548 MQT: stat/ESP-Velux/POWER1 = ON
15:35:03.556 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":83,"Direction":1,"Target":0,"Tilt":0}}
15:35:03.922 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":83,"Direction":1,"Target":0,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
15:35:04.890 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":81,"Direction":1,"Target":0,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
15:35:05.911 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":79,"Direction":1,"Target":0,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
15:35:06.154 CMD: ShutterPosition1 STOP
15:35:06.168 MQT: stat/ESP-Velux/RESULT = {"POWER1":"OFF"}
15:35:06.170 MQT: stat/ESP-Velux/POWER1 = OFF
15:35:06.175 MQT: stat/ESP-Velux/SHUTTER1 = 78
15:35:06.180 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":78,"Direction":0,"Target":78,"Tilt":0}}
15:35:10.649 CMD: ShutterPosition1 100
15:35:10.662 MQT: stat/ESP-Velux/RESULT = {"POWER2":"ON"}
15:35:10.664 MQT: stat/ESP-Velux/POWER2 = ON
15:35:10.667 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":78,"Direction":-1,"Target":100,"Tilt":0}}
15:35:10.922 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":79,"Direction":-1,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
15:35:11.913 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":81,"Direction":-1,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
15:35:12.927 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":83,"Direction":-1,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
15:35:13.892 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":85,"Direction":-1,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
15:35:14.912 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":87,"Direction":-1,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
15:35:15.926 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":89,"Direction":-1,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
15:35:16.833 CMD: ShutterPosition1 STOP
15:35:16.845 MQT: stat/ESP-Velux/RESULT = {"POWER2":"OFF"}
15:35:16.848 MQT: stat/ESP-Velux/POWER2 = OFF
15:35:16.850 MQT: stat/ESP-Velux/SHUTTER1 = 91
15:35:16.853 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":91,"Direction":0,"Target":91,"Tilt":0}}
15:35:20.480 CMD: ShutterPosition1 0
15:35:20.495 MQT: stat/ESP-Velux/RESULT = {"POWER1":"ON"}
15:35:20.498 MQT: stat/ESP-Velux/POWER1 = ON
15:35:20.502 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":91,"Direction":1,"Target":0,"Tilt":0}}
15:35:20.919 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":90,"Direction":1,"Target":0,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
15:35:21.889 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":88,"Direction":1,"Target":0,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
15:35:22.905 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":86,"Direction":1,"Target":0,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
15:35:23.932 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":84,"Direction":1,"Target":0,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
15:35:24.891 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":82,"Direction":1,"Target":0,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
15:35:25.848 CMD: ShutterPosition1 100
15:35:25.861 MQT: stat/ESP-Velux/RESULT = {"POWER1":"OFF"}
15:35:25.864 MQT: stat/ESP-Velux/POWER1 = OFF
15:35:26.375 MQT: stat/ESP-Velux/RESULT = {"POWER2":"ON"}
15:35:26.378 MQT: stat/ESP-Velux/POWER2 = ON
15:35:26.381 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":80,"Direction":-1,"Target":100,"Tilt":0}}
15:35:26.905 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":81,"Direction":-1,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
15:35:27.921 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":83,"Direction":-1,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
15:35:28.915 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":85,"Direction":-1,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
15:35:29.906 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":87,"Direction":-1,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
15:35:30.921 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":89,"Direction":-1,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
15:35:31.883 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":91,"Direction":-1,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
15:35:32.910 MQT: stat/ESP-Velux/RESULT = {"Shutter1":{"Position":93,"Direction":-1,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
[... cut ...] 100,"Tilt":0}}
15:37:09.460 CMD: STATUS 13 
15:37:09.468 MQT: stat/ESP-Velux/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":496,"Close":495,"50perc":50,"Delay":0,"Opt":"0001","Calib":[300,500,700,900,1000],"Mode":"1"},"SHT1":{"Relay1":3,"Relay2":4,"Open":496,"Close":495,"50perc":50,"Delay":0,"Opt":"0001","Calib":[300,500,700,900,1000],"Mode":"1"},"SHT2":{"Relay1":5,"Relay2":6,"Open":496,"Close":498,"50perc":50,"Delay":0,"Opt":"0001","Calib":[300,500,700,900,1000],"Mode":"1"}}}
```

stefanbode, 2023-06-12:
Bug accepted. Thx

kugelkopf123, 2023-06-12:
Same here. When set the HW switch to OFF it still stays on till its on its endposition. Its not possible to stop the shutter from moving. 
When setting setoption80 0 the HW Switch works fine.

stefanbode, 2023-06-12:
The error was introduced in dec 10th. Looks like it was accidental removed. 12.3 was the last version where the shutterposition1 up/down/open/close/stop was working

stefanbode, 2023-06-12:
@kugelkopf123 your issue seems something different. Please more logs and as always a high loglevel and the shutter settings help fixing. Switching the relay directly is still not recommended but supported. Shutterbuttons got great enhancements.

kugelkopf123, 2023-06-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Shelly 2.5
- [ ] Tasmota binary firmware version number used: 12.5.0.4(tasmota) 2_7_4_9/2.2.2-dev(38a443e) 2023.06.13 00:57:55
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Upload via Devicewebsite
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua

Configuration output here:
01:04:43.085 CMD: Backlog Template; Module; GPIO 255
01:04:43.113 MQT: stat/Shelly25_Bad/RESULT = {"NAME":"Shelly 2.5","GPIO":[320,0,32,0,224,193,0,0,640,192,608,225,3456,4736],"FLAG":0,"BASE":18}
01:04:43.335 MQT: stat/Shelly25_Bad/RESULT = {"Module":{"0":"Shelly 2.5"}}
01:04:43.590 MQT: stat/Shelly25_Bad/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"32":"Button1"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"193":"Switch_n2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"192":"Switch_n1"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"225":"Relay2"},"GPIO16":{"3456":"ADE7953 IRQ1"},"GPIO17":{"4736":"ADC Temp"}}
  

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
01:05:30.172 CMD: Status 0
01:05:30.181 MQT: stat/Shelly25_Bad/STATUS = {"Status":{"Module":0,"DeviceName":"ShutterBad","FriendlyName":["Bad_1","Bad_2"],"Topic":"Shelly25_Bad","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
01:05:30.230 MQT: stat/Shelly25_Bad/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:06:12","StartupUTC":"2023-06-12T22:59:18","Sleep":50,"CfgHolder":4617,"BootCount":20,"BCResetTime":"2023-06-08T21:12:00","SaveCount":109,"SaveAddress":"FD000"}}
01:05:30.262 MQT: stat/Shelly25_Bad/STATUS2 = {"StatusFWR":{"Version":"12.5.0.4(tasmota)","BuildDateTime":"2023.06.13 00:57:55","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"398/699"}}
01:05:30.285 MQT: stat/Shelly25_Bad/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Was ein Drama!",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C80001000600003C4B00002800000000","400000C8","00006000","00004000","00000000"]}}
01:05:30.316 MQT: stat/Shelly25_Bad/STATUS4 = {"StatusMEM":{"ProgramSize":590,"Free":432,"Heap":23,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000407","8F00058E","04000001","00000007","010000C0","00000100","00004080","00200000","50000020","00000080"],"Drivers":"1,2,3,4,9,10,27,38,50,68","Sensors":"1,2,3","I2CDriver":"7"}}
01:05:30.350 MQT: stat/Shelly25_Bad/STATUS5 = {"StatusNET":{"Hostname":"Shelly25-Bad","IPAddress":"192.168.0.202","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"E0:98:06:94:F4:AA","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
01:05:30.376 MQT: stat/Shelly25_Bad/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.100","MqttPort":1883,"MqttClientMask":"Shelly25_Bad","MqttClient":"Shelly25_Bad","MqttUser":"tasmota","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
01:05:30.399 MQT: stat/Shelly25_Bad/STATUS7 = {"StatusTIM":{"UTC":"2023-06-12T23:05:30","Local":"2023-06-13T01:05:30","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"05:17","Sunset":"21:45"}}
01:05:30.422 MQT: stat/Shelly25_Bad/STATUS10 = {"StatusSNS":{"Time":"2023-06-13T01:05:30","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":55.8},"ENERGY":{"TotalStartTime":"2021-08-22T20:33:12","Total":0.003,"Yesterday":0.003,"Today":0.000,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":0,"Current":[0.000,0.000]},"Shutter1":{"Position":9,"Direction":0,"Target":9,"Tilt":0},"TempUnit":"C"}}
01:05:30.459 MQT: stat/Shelly25_Bad/STATUS11 = {"StatusSTS":{"Time":"2023-06-13T01:05:30","Uptime":"0T00:06:12","UptimeSec":372,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Was ein Drama!","BSSId":"74:42:7F:02:21:B3","Channel":11,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:06"}}}
[... cut ...]
I have 3 other Shelly 2.5 in the flat with the identical setting. The problem only occurs since the update to the latest version. All other Shellys still running with version 12.3.1.3 have no problem. 

I made a backup of the configuration before the update. 
I reset the Shelly and imported the configuration. The error remains.
I reset the configuration again and did the configuration as described in the documentation. The error remains. 

Do you need any more information?

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2023-06-13:
Thanks look great so far. let me see if I can find the error. There were some recent changes to get away one bug with switching the relay directly. I assume there is a side effect with the switch:
`01:15:03.496 SHT: FUNC_SET_POWER Relaymask 0 SwitchedRelay:2 by **Switch**, payload 1, powermask 2`
Should be easy to fix. I have to create exceptions for every input that in the background try to change the relay.

stefanbode, 2023-06-13:
Ok, i found the error but need maybe your help:
```
01:15:03.496 SHT: FUNC_SET_POWER Relaymask 0 SwitchedRelay:2 by Switch, payload 1, powermask 2
01:15:03.498 SHT: Stop 1 Mode 0 time 4
01:15:03.499 SRC: Shutter
01:15:03.501 SHT: FUNC_SET_POWER Relaymask 0 SwitchedRelay:0 by Shutter, payload 1, powermask 0
```
Everything is fine if you switch relay2. this executes as expected. But if you switch relay1 this is reported as "SwitchedRelay:0" and not 1. This is strange and unexpected. I checked your config with "power1 on/off" and "power2 on/off" in the console because missing the switches and this does work. is it really a switch or more a button? == momentary switch

stefanbode, 2023-06-13:
Ok then please provide switchmode and the gpio it is connected. Maybe a picture of the config page is also fine. I have to physically build it here.

kugelkopf123, 2023-06-13:
As also mentioned in my previous post: Switchmode 1

[image]
[image]

What else do you need?

stefanbode, 2023-06-14:
Ok. Got it. In this case I do know what to build. Will post later today.

kugelkopf123, 2023-06-14:
Thanks! Unfortunately, I can't test it until Friday. I'll let you know as soon as I've tried it.

kugelkopf123, 2023-06-15:
Problem solved! Everything is back to the way it used to be. 
Thank you very much!

closed 2023-06-15 by Jason2866

mkjustuk, 2023-06-28:
13.0.0 appears to still have the (or very related) issue.

If I send a msg via mqtt
cmnd/tasmota_F594DD/SHUTTER1/shutterstopopen
the shutter closes. Sending again stops it, but a third time closes again. shutterstopclose works fine.

Jason2866, 2023-06-28:
@mkjustuk  Please open a new issue.