arendst/Tasmota #18852: tasmota32-webcam not working  12.5.0 no recording.

hessel-a, 2023-06-11:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
  When using firmware 12.5.0 there is no recording/image. So I went back to firmware 11.1.0 and the image is back

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): __ESP32-Cam___
- [ X] Tasmota binary firmware version number used: ___12.5.0__
  - [X ] Pre-compiled
  - [ ] Self-compiled
- [X ] Flashing tools used: ___in app__
- [X ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
11:32:41.521 RSL: RESULT = {"NAME":"AITHINKER CAM","GPIO":[4992,1,672,1,416,5088,1,1,1,6720,736,704,1,1,5089,5090,0,5091,5184,5152,0,5120,5024,5056,0,0,0,0,4928,576,5094,5095,5092,0,0,5093],"FLAG":0,"BASE":2}
11:32:41.742 RSL: RESULT = {"Module":{"2":"ESP32-Cam"}}
11:32:42.002 RSL: RESULT = {"GPIO0":{"4992":"CAM_XCLK"},"GPIO1":{"0":"None"},"GPIO2":{"672":"SPI MISO1"},"GPIO3":{"0":"None"},"GPIO4":{"416":"PWM1"},"GPIO5":{"5088":"CAM_DATA1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"6720":"SDCard CS"},"GPIO14":{"736":"SPI CLK1"},"GPIO15":{"704":"SPI MOSI1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"5089":"CAM_DATA2"},"GPIO19":{"5090":"CAM_DATA3"},"GPIO20":{"0":"None"},"GPIO21":{"5091":"CAM_DATA4"},"GPIO22":{"5184":"CAM_PCLK"},"GPIO23":{"5152":"CAM_HREF"},"GPIO24":{"0":"None"},"GPIO25":{"5120":"CAM_VSYNC"},"GPIO26":{"5024":"CAM_SIOD"},"GPIO27":{"5056":"CAM_SIOC"},"GPIO32":{"4928":"CAM_PWDN"},"GPIO33":{"0":"None"},"GPIO34":{"5094":"CAM_DATA7"},"GPIO35":{"5095":"CAM_DATA8"},"GPIO36":{"5092":"CAM_DATA5"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"5093":"CAM_DATA6"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
11:38:13.513 RSL: STATUS = {"Status":{"Module":2,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_867690","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
11:38:13.549 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-webcam.bin","RestartReason":"Software reset CPU","Uptime":"0T00:06:42","StartupUTC":"2023-06-11T09:31:31","Sleep":50,"CfgHolder":4617,"BootCount":34,"BCResetTime":"2023-03-17T09:24:01","SaveCount":104}}
11:38:13.583 RSL: STATUS2 = {"StatusFWR":{"Version":"12.5.0(webcam)","BuildDateTime":"2023-04-17T08:06:59","Core":"2_0_7","SDK":"4.4.4.20230310","CpuFrequency":240,"Hardware":"ESP32-D0WD rev.1","CR":"375/699"}}
11:38:13.596 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Iot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C80001000600003C5A0A192800000000","000000C0","00006000","00004000","00000000"]}}
11:38:13.622 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1257,"Free":1856,"Heap":125,"StackLowMark":3,"PsrMax":4093,"PsrFree":3337,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","0F0005CA","00040001","00000001","00000000","00000880","80000080","00200000","54000008","00000080"],"Drivers":"1,2,4,9,10,12,29,38,50,52,62,81","Sensors":"1,127"}}
11:38:13.657 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-867690-5776","IPAddress":"10.0.0.221","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer1":"10.0.0.1","DNSServer2":"0.0.0.0","Mac":"8C:CE:4E:86:76:90","IP6Global":"","IP6Local":"fe80::8ece:4eff:fe86:7690","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
11:38:13.693 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-06-11T09:38:13","Local":"2023-06-11T11:38:13","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+02:00","Sunrise":"05:46","Sunset":"21:52"}}
11:38:13.710 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-06-11T11:38:13"}}
11:38:13.716 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-06-11T11:38:13","Uptime":"0T00:06:42","UptimeSec":402,"Heap":125,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":34},"POWER":"OFF","Dimmer":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Iot","BSSId":"76:AC:B9:2A:00:D6","Channel":1,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Switch back to version 11.1.0

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
 Image picture is gone and Webcam Frame rate = 0 FPS

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-06-11:
Try latest development version. Changes are done in Webcam driver since release v12.5.0

Jason2866, 2023-06-15:
@hessel-a Status?

Jason2866, 2023-06-23:
Closing since no feedback.

closed 2023-06-23 by Jason2866