arendst/Tasmota #18854: No tuyareceived for Duux Whisper Flex Smart Fan (DXCF10)

barrymossel, 2023-06-11:
### PROBLEM DESCRIPTION
I installed Tasmota through tuya-convert on a Duux Whisper Flex Smart Fan (DXCF10). Upgraded to latest NL version (12.5.0). Used the template from https://templates.blakadder.com/duux_DXCF10.html. Unfortunately no tuyareceived is received. 

```
14:02:06.388 TYA: Send "55aa001c00080117060b0e02060769"
14:02:14.355 TYA: Send "55aa00000000ff"
14:02:25.375 TYA: Send "55aa00000000ff"
14:02:36.382 TYA: Send "55aa00000000ff"
14:02:44.660 HTP: Hoofdmenu
14:02:47.376 TYA: Send "55aa00000000ff"
14:02:58.378 TYA: Send "55aa00000000ff"
14:03:06.350 TYA: Send "55aa001c00080117060b0e0306076a"
14:03:09.374 TYA: Send "55aa00000000ff"
14:03:20.388 TYA: Send "55aa00000000ff"
14:03:31.394 TYA: Send "55aa00000000ff"
14:03:42.368 TYA: Send "55aa00000000ff"
14:03:53.367 TYA: Send "55aa00000000ff"
14:04:04.368 TYA: Send "55aa00000000ff"
14:04:06.358 TYA: Send "55aa001c00080117060b0e0406076b"
```

Tried both baud rates as well as default GPIOs for TuyaTx and TuyaRx. Also tried to bruteforce GPIOs by trying all possible combinations and baud rates. Still no tuyareceived.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Duux Whisper Flex Smart Fan (DXCF10)
- [X] Tasmota binary firmware version number used: 12.5.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: tuya-convert
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
14:07:13.307 CMD: Backlog Template; Module; GPIO 255
14:07:13.309 SRC: WebConsole from 192.168.2.28
14:07:13.313 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
14:07:13.402 SRC: Backlog
14:07:13.405 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
14:07:13.410 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
14:07:13.615 SRC: Backlog
14:07:13.618 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
14:07:13.622 RSL: RESULT = {"Module":{"54":"Tuya MCU"}}
14:07:13.865 SRC: Backlog
14:07:13.868 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
14:07:13.875 RSL: RESULT = {"GPIO0":{"0":"Geen"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"Geen"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"Geen"},"GPIO5":{"0":"Geen"},"GPIO9":{"0":"Geen"},"GPIO10":{"0":"Geen"},"GPIO12":{"0":"Geen"},"GPIO13":{"0":"Geen"},"GPIO14":{"0":"Geen"},"GPIO15":{"0":"Geen"},"GPIO16":{"0":"Geen"},"GPIO17":{"0":"Geen"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
No rules

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:08:23.136 CMD: Status 0
14:08:23.138 SRC: WebConsole from 192.168.2.28
14:08:23.140 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
14:08:23.147 RSL: STATUS = {"Status":{"Module":54,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_BEBF53","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
14:08:23.155 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-NL.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:09:17","StartupUTC":"2023-06-11T12:59:06","Sleep":50,"CfgHolder":4617,"BootCount":3,"BCResetTime":"2023-06-11T13:55:32","SaveCount":13,"SaveAddress":"F7000"}}
14:08:23.161 RSL: STATUS2 = {"StatusFWR":{"Version":"12.5.0(tasmota)","BuildDateTime":"2023-04-17T08:10:13","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"377/699"}}
14:08:23.167 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["REDACTED",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","0000E000","00004000","00000000"]}}
14:08:23.179 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":633,"Free":368,"Heap":25,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000413","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
14:08:23.184 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-BEBF53-8019","IPAddress":"192.168.2.27","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer1":"1.1.1.1","DNSServer2":"8.8.8.8","Mac":"50:02:91:BE:BF:53","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:08:23.189 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_BEBF53","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:08:23.196 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-06-11T13:08:23","Local":"2023-06-11T14:08:23","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:52"}}
14:08:23.202 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-06-11T14:08:23"}}
14:08:23.208 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-06-11T14:08:23","Uptime":"0T00:09:17","UptimeSec":557,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"REDACTED","BSSId":"6A:86:60:D5:1B:48","Channel":1,"Mode":"11n","RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
14:02:06.388 TYA: Send "55aa001c00080117060b0e02060769"
14:02:14.355 TYA: Send "55aa00000000ff"
14:02:25.375 TYA: Send "55aa00000000ff"
14:02:36.382 TYA: Send "55aa00000000ff"
14:02:44.660 HTP: Hoofdmenu
14:02:47.376 TYA: Send "55aa00000000ff"
14:02:58.378 TYA: Send "55aa00000000ff"
14:03:06.350 TYA: Send "55aa001c00080117060b0e0306076a"
14:03:09.374 TYA: Send "55aa00000000ff"
14:03:20.388 TYA: Send "55aa00000000ff"
14:03:31.394 TYA: Send "55aa00000000ff"
14:03:42.368 TYA: Send "55aa00000000ff"
14:03:53.367 TYA: Send "55aa00000000ff"
14:04:04.368 TYA: Send "55aa00000000ff"
14:04:06.358 TYA: Send "55aa001c00080117060b0e0406076b"

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
See above.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Receive tuyareceived data

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barrymossel, 2023-06-11:
Donwgraded to 9.1.0 and still (almost) the same:

```
14:21:38 TYA: Send "55aa00000000ff"
14:21:49 TYA: Send "55aa00000000ff"
14:22:00 TYA: Send "55aa00000000ff"
14:22:11 TYA: Send "55aa00000000ff"
14:22:22 TYA: Send "55aa00000000ff"
14:22:33 TYA: Send "55aa00000000ff"
14:22:44 TYA: Send "55aa00000000ff"
14:22:55 TYA: Send "55aa00000000ff"
14:23:06 TYA: Send "55aa00000000ff"
14:23:17 TYA: Send "55aa00000000ff"
14:23:28 TYA: Send "55aa00000000ff"
14:23:39 TYA: Send "55aa00000000ff"
```

sfromis, 2023-06-11:
That is failure to communicate with the TuyaMCU, assuming that device is (still) using such. Potential causes include wrong baud rate or wrong gpio pins, and of course any hardware issues, like "accidents" during flashing.

barrymossel, 2023-06-11:
> That is failure to communicate with the TuyaMCU, assuming that device is (still) using such. Potential causes include wrong baud rate or wrong gpio pins, and of course any hardware issues, like "accidents" during flashing.

Tried alle GPIOs and baud rates. So I guess that's not the issue. How could I "debug" hardware issues? And any way to fix those?

sfromis, 2023-06-11:
First step would be to carefully check details of what you may have done while flashing. A little bit of solder the wrong place can be pretty bad....

barrymossel, 2023-06-11:
> First step would be to carefully check details of what you may have done while flashing. A little bit of solder the wrong place can be pretty bad....

Ah like that. I didn't solder anything, as I used tuya-convert. So hardware issue aren't likely either...

barbudor, 2023-06-11:
Is that this model? https://templates.blakadder.com/duux_DXCF10.html

A careful look at the board could help confirm that the expected connections between the ESP and the TuyaMCU are GPIO14 and 16 (which are not the best as this will require software Serial)

An oscilloscope or logic analyser would be helpful to verify that something is actually coming from the MCU and at which baudrate

barrymossel, 2023-06-11:
I am not sure if I want to open up the device, as it still works with the normal remote. And I don't have an oscilloscope. And as I tried all GPIO combinations I don't see why looking at the traces would help.

sfromis, 2023-06-11:
That serial pin assignments on gpio 14/16 is **highly unusual**, and I suspect that some error may have happened. Try with this template instead:
`{"NAME":"Duux Fan","GPIO":[0,0,0,0,0,0,0,0,0,2304,0,2272,0,0],"FLAG":0,"BASE":54}`

barrymossel, 2023-06-11:
> 

You are like fantastic. 

```
18:29:55 TYA: Send "55aa00000000ff"
18:29:55 {"TuyaReceived":{"Data":"55AA030000010003","Cmnd":0,"CmndData":"00"}}
18:29:55 TYA: Heartbeat
18:29:55 TYA: Detected MCU restart
18:29:56 TYA: Set WiFi LED 3 (0)
18:29:56 TYA: Send "55aa000300010306"
18:29:56 {"TuyaReceived":{"Data":"55AA0303000005","Cmnd":3}}
18:29:56 TYA: RX WiFi LED set ACK
18:30:06 TYA: Send "55aa00000000ff"
18:30:06 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
18:30:06 TYA: Heartbeat
18:30:07 WIF: Checking connection...
18:30:17 TYA: Send "55aa00000000ff"
18:30:17 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
18:30:17 TYA: Heartbeat
18:30:27 WIF: Checking connection...
18:30:28 TYA: Send "55aa00000000ff"
18:30:28 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
18:30:28 TYA: Heartbeat
```

barrymossel, 2023-06-11:
> That serial pin config using gpio 14/16 is **highly unusual**, and I suspect that some error may have happened. Try with this template instead: `{"NAME":"Duux Fan","GPIO":[0,0,0,0,0,0,0,0,0,2304,0,2272,0,0],"FLAG":0,"BASE":54}`

What is exactly changed btw? Only 13/15? Because I tried that as well before??

sfromis, 2023-06-11:
Yes, only 13/15 instead of 14/16. I may be a bit cynical, but when I see people claiming to have "tried all", I instead read it as "tried to try all", without assuming it to be 100% :grin: 

Thanks for submitting the correction to the templates repository.

barrymossel, 2023-06-11:
Might be because I had the wrong template active and tried to overrule it?

sfromis, 2023-06-11:
Such depends on details of how you tried to overrule the already loaded template. Any actual update of the template should trigger a restart at the same time.

barrymossel, 2023-06-11:
Oh well, it worked. And that's great. Now off to getting all in HA as a fan!

closed 2023-06-11 by barrymossel