arendst/Tasmota #18865: v12.5.0.4 - Script: Slider and permanent variables not working

SHU-red, 2023-06-12:
### PROBLEM DESCRIPTION
- Successfully build Tasmota from development branch with additional definitions (see below)
- Programmed Script using `>W` for sliders --> If sliders are dragged and released, the connected variables do not change
- Programmed Script using permanent variables with `p:*` --> Permanent variables are always 0 and can not be used

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [x] Tasmota binary firmware version number used: 12.5.0.4
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Built with VSCode(PlatformIO), flashed on Linux with esptool.py
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

16:36:20.813 CMD: Backlog Template; Module; GPIO 255
16:36:20.830 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
16:36:21.059 RSL: RESULT = {"Module":{"0":"Generic"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:36:43.117 CMD: Status 0
16:36:43.124 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_7AD804","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
16:36:43.159 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:01:06","StartupUTC":"2023-06-12T15:35:37","Sleep":50,"CfgHolder":4617,"BootCount":3,"BCResetTime":"2023-06-11T14:41:04","SaveCount":12,"SaveAddress":"F9000"}}
16:36:43.190 RSL: STATUS2 = {"StatusFWR":{"Version":"12.5.0.4(tasmota)","BuildDateTime":"2023-06-12T17:29:40","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"396/699"}}
16:36:43.207 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ShUte___2GHz",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
16:36:43.240 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":704,"Free":296,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"QIO","Features":["00000809","879AC787","043E8001","000002CF","010013C0","C000F981","00004004","00201000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,50,62,68","Sensors":"1,2,3,4,5,6,8","I2CDriver":"7,9"}}
16:36:43.276 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-7AD804-6148","IPAddress":"192.168.178.102","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"48:55:19:7A:D8:04","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
16:36:43.299 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7AD804","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:36:43.319 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-06-12T15:36:43","Local":"2023-06-12T16:36:43","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:53"}}
16:36:43.337 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-06-12T16:36:43","HTU21":{"Temperature":24.0,"Humidity":47.2,"DewPoint":12.1},"TempUnit":"C"}}
16:36:43.350 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-06-12T16:36:43","Uptime":"0T00:01:06","UptimeSec":66,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"ShUte___2GHz","BSSId":"18:FD:74:1B:C2:FA","Channel":9,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
nothing to see

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Slider: Drag the slider --> Slider snaps back, no value changed
Permanent Variables: Defining a variable as permanent via adding `p:*` in >D section makes variable not usable

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Slider: Draging the slider should set variable to new value and sliders stays at dragged position
Permanent Variables: Behave like any other variable but being permanent

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

https://postimg.cc/bZLvWczx

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

My not yet completed script below
There may be issues in my code --> If this is the case, i want to excuse in advance
Basically the main-mechanism of the script is working without errors

Current script
```
>D
target_hum=70
time_wait_h=20
time_humidify_s=2
countdown_wait=0
t:countdown_humidify=0
act_hum=100

>T
act_hum=HTU21#Humidity

>S
; Check for ended countdown
if countdown_wait<=0
then
  countdown_humidify=time_humidify_s
  countdown_wait=time_wait_h
endif

; Check for humidification
if countdown_humidify>0
then
  print HUMIDIFY
  spin(4 1)
else
  print NO HUMIDIFICATION
  spin(4 0)
endif
  print level of gpio4 %pin[4]%

; Hour Countdown
hour=int(time/60)
;if chg[hour]>0 {
  countdown_wait-=1
;}

>W
sl(65 80 target_hum "65" "Target Humidity" "80")
sl(1 48 time_wait_h "1" "Waiting Time  "48")
sl(1 5 time_humidify_s "1" "Humidification Time" "5")
Countdown Waiting{m}%countdown_wait% h
Countdown Humnidification{m}%countdown_humidify% s
```

Test with making variables permanent
```
>D
p:target_hum=70
p:time_wait_h=20
p:time_humidify_s=2
countdown_wait=0
t:countdown_humidify=0
act_hum=100
...
```

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2023-06-12:
ok, there is an important info missing in the docs.  in section >W USR IO variables may not contain an underscore.

use hum instead of act_hum and it will work

SHU-red, 2023-06-12:
Do you also know why im not able to use permanent variables?

Is it cuz my hardware (Wemos D1 Mini / ESP8266) does not support something like this?

gemu2015, 2023-06-12:
It supports permanent variables on any hardware
Setup a value with slider and restart device. It should restore that value

SHU-red, 2023-06-12:
Thank you
I had to also use the `svars` command

Now it seems to work

Thanks a lot