arendst/Tasmota #18871: Neopool - NPFiltration incorrect behavior

fdebrus, 2023-06-13:
### PROBLEM DESCRIPTION
When using Tasmota with Neopool, the NPFiltration command accept 0 "OFF" 1 "ON" or 2 "Toggle" as values.

The opposite is reported in MQTT stat

Calling NPFiltration with 0 reports the below in MQTT, the Filtration pool controller is OFF
{
  "NPFiltration": "ON",
  "Speed": "2"
}
Calling NPFiltration with 1 reports the below in MQTT, the Filtration pool controller is ON
{
  "NPFiltration": "OFF",
  "Speed": "2"
}

So stat/.../NPFilration is incorrectly set when calling NPFiltration with 0 or 1

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
send NPFiltration with 0 or 1 over the MQTT command channel and check stat channel 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-06-13:
Please add the missing Infos asked in the template.

barbudor, 2023-06-13:
@fdebrus 
What is the result of command `StateText` ?

fdebrus, 2023-06-14:
StateText
{
  "StateText1": "OFF",
  "StateText2": "ON",
  "StateText3": "TOGGLE",
  "StateText4": "HOLD"
}

fdebrus, 2023-06-14:
I tried to call
 NPFiltration 0 mutiple times in a row
1 call -> ON
2 call -> OFF
3 call -> OFF
 NPFiltration 1 mutiple times in a row
1 call -> OFF
2 call -> ON
3 call -> ON

So the first call to NPFiltration gives a wrong status while nexts are good.

barbudor, 2023-06-14:
Looking at the code I don't see any problem with the logic
The value that is displayed as the status from the command is read back to the device
So I see 2 possible issues:
- The device take some time to take the command into account 
- The device is missing some commands due to the communication problems (wiring quality ?)

You can interrogate the device with command `NPFiltration` without any argument
Starting with a state OFF, what are the result for the following sequence of commands executed one after the other leaving 5-10 seconds between each command:
```
NPFiltration
NPFiltration 1
NPFiltration
NPFiltration
NPFiltration
NPFiltration 1
NPFiltration
NPFiltration
```

barbudor, 2023-06-14:
Hi @curzon01 I think you were the main contributor on this code
My opinion would be a communication problem which makes the NeoPool device miss commands
What do you think ?

fdebrus, 2023-06-14:
Here you go

12:03:18.561 CMD: **NPFiltration**
12:03:18.563 SRC: WebConsole from 192.168.2.120
12:03:18.565 CMD: Grp 0, Cmd 'NPFILTRATION', Idx 1, Len 0, Pld -99, Data ''
12:03:18.585 MQT: stat/SmartPool/NPFILTRATION = {"NPFiltration":"OFF","Speed":"2"}
12:03:28.126 CMD: **NPFiltration 1**
12:03:28.128 SRC: WebConsole from 192.168.2.120
12:03:28.130 CMD: Grp 0, Cmd 'NPFILTRATION', Idx 1, Len 1, Pld 1, Data '1'
12:03:28.198 MQT: stat/SmartPool/NPFILTRATION = {"NPFiltration":"OFF","Speed":"2"}
12:03:34.788 WIF: Checking connection...
12:03:35.447 CMD: **NPFiltration**
12:03:35.449 SRC: WebConsole from 192.168.2.120
12:03:35.452 CMD: Grp 0, Cmd 'NPFILTRATION', Idx 1, Len 0, Pld -99, Data ''
12:03:35.473 MQT: stat/SmartPool/NPFILTRATION = {"NPFiltration":"ON","Speed":"2"}
12:03:42.066 CMD: **NPFiltration**
12:03:42.068 SRC: WebConsole from 192.168.2.120
12:03:42.071 CMD: Grp 0, Cmd 'NPFILTRATION', Idx 1, Len 0, Pld -99, Data ''
12:03:42.092 MQT: stat/SmartPool/NPFILTRATION = {"NPFiltration":"ON","Speed":"2"}
12:03:42.769 WIF: Sending Gratuitous ARP
12:03:48.051 CMD: **NPFiltration**
12:03:48.053 SRC: WebConsole from 192.168.2.120
12:03:48.056 CMD: Grp 0, Cmd 'NPFILTRATION', Idx 1, Len 0, Pld -99, Data ''
12:03:48.077 MQT: stat/SmartPool/NPFILTRATION = {"NPFiltration":"ON","Speed":"2"}
12:03:54.782 WIF: Checking connection...
12:03:57.176 CMD: **NPFiltration 1**
12:03:57.178 SRC: WebConsole from 192.168.2.120
12:03:57.181 CMD: Grp 0, Cmd 'NPFILTRATION', Idx 1, Len 1, Pld 1, Data '1'
12:03:57.249 MQT: stat/SmartPool/NPFILTRATION = {"NPFiltration":"ON","Speed":"2"}
12:04:03.776 CMD: **NPFiltration** 
12:04:03.779 SRC: WebConsole from 192.168.2.120
12:04:03.781 CMD: Grp 0, Cmd 'NPFILTRATION', Idx 1, Len 0, Pld -99, Data ''
12:04:03.802 MQT: stat/SmartPool/NPFILTRATION = {"NPFiltration":"ON","Speed":"2"}
12:04:10.193 CMD: **NPFiltration** 
12:04:10.196 SRC: WebConsole from 192.168.2.120
12:04:10.198 CMD: Grp 0, Cmd 'NPFILTRATION', Idx 1, Len 0, Pld -99, Data ''
12:04:10.221 MQT: stat/SmartPool/NPFILTRATION = {"NPFiltration":"ON","Speed":"2"}
12:04:15.034 WIF: Checking connection...

Note:
When issuing NPFiltration 1 or 0 the pump do react accordingly 1- ON 0-OFF so the command is not "lost" only state is not updated accordingly.

curzon01, 2023-06-14:
It seems @barbudor are right, the Auarite system might be very slow updating their internal register.

Pls provide "NPControl" and "Teleperiod" (topic .../SENSOR) result to see the system info and FW version number

(pls be patient, I'm busy 'till ne t week)

fdebrus, 2023-06-14:
14:10:56.256 MQT: stat/SmartPool/NPCONTROL = {"Modules":{"pH":1,"Redox":1,"Hydrolysis":1,"Chlorine":0,"Conductivity":0,"Ionization":0},"Relay":{"Acid":1,"Base":0,"Redox":0,"Chlorine":0,"Conductivity":0,"Heating":7,"UV":0,"Valve":0}}

14:11:21.778 MQT: stat/SmartPool/TELEPERIOD = {"TelePeriod":300}

tele/SmartPool/SENSOR = {"Time":"2023-06-14T14:11:22","NeoPool":{"Time":"2023-06-14T15:09:49","Type":"Generic","Modules":{"pH":1,"Redox":1,"Hydrolysis":1,"Chlorine":0,"Conductivity":0,"Ionization":0},"Temperature":30.2,"Power":{"Version":"V3.57","NodeID":"05DE 2D35 3837 574E 4318 0937","5V":5,"12V":14,"24-30V":32,"4-20mA":0.120},"pH":{"Data":7.2,"Min":7.0,"Max":7.4,"State":3,"Pump":2,"FL1":0,"Tank":1},"Redox":{"Data":663,"Setpoint":700},"Hydrolysis":{"Data":15,"Unit":"g/h","Runtime":{"Total":"42T17:52:10","Part":"42T17:52:10","Pol1":"21T03:40:54","Pol2":"21T14:11:16","Changes":492},"State":"Pol1","Cover":0,"Boost":0,"Low":0},"Filtration":{"State":1,"Speed":3,"Mode":0},"Light":0,"Relay":{"State":[0,1,0,0,0,0,0],"Aux":[0,0,0,0],"Acid":0,"Heating ":0}},"TempUnit":"C"}

curzon01, 2023-06-15:
may be I can fix it, let me check next week

meantime workaround:
for switching pump use `Backlog NPFiltration x;Delay 10;NPFiltration` - x = 0 or 1, adapt Delay period if necessary

fdebrus, 2023-06-19:
thanks, not sure how to use that workaround from HA but I can code something similar. For the time being this is how I call the command

`mqtt:
  switch:
    - unique_id: "pool_pump_switch"
      name: "Pool Pump Switch"
      device_class: switch
      retain: true
      command_topic: "cmnd/SmartPool/NPFiltration"
      state_topic: "stat/SmartPool/NPFILTRATION"
      value_template:  >- 
        {% set values = { 'OFF':'0', 'ON':'1' } %}
        {{ values[value_json["NPFiltration"]] if value_json["NPFiltration"] in values.keys() else 'Error' }}
      payload_on: "1"
      payload_off: "0"`

curzon01, 2023-06-19:
you're not patient enough :-) 
fix is still in test and already on the way...

PR #18908 (2023-06-19): Fix NeoPool NPFiltration switch result
## Description:

Due to the slow processing on some devices NPFiltration with switch parameter now returns given parameter as result instead of the device register state. NPFiltration without switch parameter always returns the device register state.

**Related issue (if applicable):** fixes #18871

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and work [... cut]
closed 2023-06-19 by s-hadinger