arendst/Tasmota #18876: no websend result with 12.5

karoCB, 2023-06-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### NO websend result with compiler options
#ifdef USE_WEBSEND_RESPONSE
#undef USE_WEBSEND_RESPONSE
#endif
#define USE_WEBSEND_RESPONSE

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ x] Tasmota binary firmware version number used: 12.5____
  - [ ] Pre-compiled
  - [x ] Self-compiled
- [ x] Flashing tools used: _Update (esptool.exe)____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
RESULT = {"NAME":"Generic","GPIO":[1,1,224,1,1,1,1,1,225,226,1,1,1,4736],"FLAG":0,"BASE":18}
RESULT = {"Module":{"0":"Generic"}}
RESULT = {"Module":{"0":"Generic"}}
```
- [X ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":170,"Free":341,"Rules":"on VAR2#State do websend [feiertage-api.de] /api/?jahr=%VAR2%&nur_land=BE endon  on Pfingstmontag#datum do var1 %value% endon on VAR1#state do Pfingsten===>> %var1% endon"}}
RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","",""],"Topic":"tasmota_591198","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:07:07","StartupUTC":"2023-06-14T11:01:08","Sleep":50,"CfgHolder":4617,"BootCount":41,"BCResetTime":"2023-04-13T15:59:07","SaveCount":372,"SaveAddress":"F7000"}}
RSL: STATUS2 = {"StatusFWR":{"Version":"12.5.0(TasmoCompiler-esp8266generic)","BuildDateTime":"2023-06-11T20:04:34","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"367/699"}}
RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ddd1",""],"TelePeriod":20,"Resolution":"55818040","SetOption":["00008009","2805C80001000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
RSL: STATUS4 = {"StatusMEM":{"ProgramSize":531,"Free":472,"Heap":28,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","0F804582","04260001","00000003","00000000","C000E881","00000004","00001000","54000000","00000080"],"Drivers":"1,2,4,6,8,9,10,12,18,19,22,24,26,29,30,35,37,45,62","Sensors":"1,2"}}
RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-591198-4504","IPAddress":"192.168.178.201","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"EC:FA:BC:59:11:98","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_591198","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-06-14T11:08:16","Local":"2023-06-14T12:08:16","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:54"}}
RSL: STATUS10 = {"StatusSNS":{"Time":"2023-06-14T12:08:16","ANALOG":{"Temperature":193.1},"TempUnit":"C"}}
RSL: STATUS11 = {"StatusSTS":{"Time":"2023-06-14T12:08:16","Uptime":"0T00:07:08","UptimeSec":428,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"ON","POWER3":"OFF","Wifi":{"AP":1,"SSId":"ddd1","BSSId":"FC:F4:FA:F3:F4:F8","Channel":6,"Mode":"11n","RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
12:13:09.670 CMD: var2 2023
12:13:09.672 SRC: WebConsole from 192.168.178.57
12:13:09.674 CMD: Grp 0, Cmd 'VAR', Idx 2, Len 4, Pld 2023, Data '2023'
12:13:09.679 RSL: RESULT = {"Var2":"2023"}
12:13:09.776 RUL: VAR2#STATE performs "websend [feiertage-api.de] /api/?jahr=2023&nur_land=BE"
12:13:09.779 SRC: Rule
12:13:09.781 CMD: Grp 0, Cmd 'WEBSEND', Idx 1, Len 46, Pld -99, Data '[feiertage-api.de] /api/?jahr=2023&nur_land=BE'
12:13:09.993 RSL: RESULT = {"WebQuery":{"Neujahrstag":{"datum":"2023-01-01","hinweis":""},"Frauentag":{"datum":"2023-03-08","hinweis":""},"Karfreitag":{"datum":"2023-04-07","hinweis":""},"Ostermontag":{"datum":"2023-04-10","hinweis":""},"Tag der Arbeit":{"datum":"2023-05-01","hinweis":""},"Christi Himmelfahrt":{"datum":"2023-05-18","hinweis":""},"Pfingstmontag":{"datum":"2023-05-29","hinweis":""},"Tag der Deutschen Einheit":{"datum":"2023-10-03","hinweis":""},"1. Weihnachtstag":{"datum":"2023-12-25","hinweis":""},"2. Weihnachtstag":{"datum":"2023-12-26","hinweis":""}}}
12:13:09.999 RSL: RESULT = {"WebSend":"Done"}
...

### TO REPRODUCE
_Steps to reproduce the behavior:_
input year in variable VAR2 creates webservice without result in VAR1 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
19:13:13.179 RUL: VAR1#STATE performs "Pfingsten==>>2024 2024-05-20"

log from 9.5 
19:13:12.947 CMD: Grp 0, Cmnd 'WEBSEND', Idx 1, Len 46, Data '[feiertage-api.de] /api/?jahr=2024&nur_land=BE'
19:13:13.139 RSL: RESULT = {"Neujahrstag":{"datum":"2024-01-01","hinweis":""},"Frauentag":{"datum":"2024-03-08","hinweis":""},"Karfreitag":{"datum":"2024-03-29","hinweis":""},"Ostermontag":{"datum":"2024-04-01","hinweis":""},"Tag der Arbeit":{"datum":"2024-05-01","hinweis":""},"Christi Himmelfahrt":{"datum":"2024-05-09","hinweis":""},"Pfingstmontag":{"datum":"2024-05-20","hinweis":""},"Tag der Deutschen Einheit":{"datum":"2024-10-03","hinweis":""},"1. Weihnachtstag":{"datum":"2024-12-25","hinweis":""},"2. Weihnachtstag":{"datum":"2024-12-26","hinweis":""}}
19:13:13.152 RUL: PFINGSTMONTAG#DATUM performs "var1 2024-05-20"
19:13:13.153 SRC: Rule
19:13:13.156 CMD: Grp 0, Cmnd 'VAR', Idx 1, Len 10, Data '2024-05-20'
19:13:13.159 RSL: RESULT = {"Var1":"2024-05-20"}
19:13:13.166 RSL: RESULT = {"�?�?��?":"Done"}
19:13:13.179 RUL: VAR1#STATE performs "Pfingsten==>>2024 2024-05-20"

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
in release 9, ABC was needed to succeed !

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-06-14:
A rule with illegal syntax won't be executed
Fix it

barbudor, 2023-06-14:
And you are missing `WebQuery#` in the trigger for the 2nd rule