arendst/Tasmota #18886: GT911 - offset issue

OperatorB, 2023-06-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sunton 3.5" Capacitive https://macsbug.wordpress.com/2022/10/02/esp32-3248s035/
- [x] Tasmota binary firmware version number used: 12.5.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: ESP-IDF 4.4 and esptool.py
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

Template:
{"NAME":"mako1","GPIO":[0,1,800,6210,320,1,1,1,672,704,736,768,321,322,1,1,0,0,1,1,0,1,1,992,0,0,0,0,1,1,1,1,1,0,1,1],"FLAG":0,"BASE":1}

Module:
{"Module":{"0":"mako1"}}

GPIO 255:
{"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"800":"SPI DC1"},"GPIO3":{"6210":"Option A3"},"GPIO4":{"320":"Led_i1"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"672":"SPI MISO1"},"GPIO13":{"704":"SPI MOSI1"},"GPIO14":{"736":"SPI CLK1"},"GPIO15":{"768":"SPI CS1"},"GPIO16":{"321":"Led_i2"},"GPIO17":{"322":"Led_i3"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"992":"Backlight"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

n/a

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

19:48:36.430 CMD: status 0
19:48:36.433 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_44B724","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
19:48:36.467 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-lvgl.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:07:34","StartupUTC":"2023-06-15T18:41:02","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2023-06-15T19:39:03","SaveCount":13}}
19:48:36.502 RSL: STATUS2 = {"StatusFWR":{"Version":"12.5.0(lvgl-haspmota)","BuildDateTime":"2023-06-15T20:30:34","Core":"2_0_7","SDK":"4.4.4.20230310","CpuFrequency":240,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"392/699"}}
19:48:36.515 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Monero",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
19:48:36.553 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1886,"Free":993,"Heap":103,"StackLowMark":19,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","0FA287CE","00048001","000007C7","01000000","40000881","40004080","00200000","5400001D","000200A0"],"Drivers":"1,2,3,4,5,8,9,10,12,13,16,20,24,29,38,50,52,54,55,56,62,67,82,88","Sensors":"1,2,3,5,6,7,8,9,85,127","I2CDriver":"7,8,9,10,58,59,77"}}
19:48:36.602 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-44B724-5924","IPAddress":"192.168.2.118","Gateway":"192.168.2.111","Subnetmask":"255.255.255.0","DNSServer1":"9.9.9.9","DNSServer2":"8.8.8.8","Mac":"54:43:B2:44:B7:24","IP6Global":"","IP6Local":"fe80::5643:b2ff:fe44:b724","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"9.9.9.9","DNSServer2":"8.8.8.8","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:48:36.651 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-06-15T18:48:36","Local":"2023-06-15T19:48:36","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:54"}}
19:48:36.668 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-06-15T19:48:36"}}
19:48:36.675 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-06-15T19:48:36","Uptime":"0T00:07:34","UptimeSec":454,"Heap":101,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"Berry":{"HeapUsed":5,"Objects":68},"POWER":"ON","Wifi":{"AP":1,"SSId":"SecretSSID","BSSId":"C0:C9:E3:51:32:86","Channel":4,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

Touched near the corners, clockwise starting from top left corner:

19:54:14.355 TS : TSGlobal.touched x=5 y=-152 (raw x=472 y=5)
19:54:14.653 TS : released x=5 y=-152 (raw x=472 y=5)
19:54:18.555 TS : TSGlobal.touched x=313 y=-125 (raw x=445 y=313)
19:54:18.654 TS : released x=313 y=-125 (raw x=445 y=313)
19:54:21.752 TS : TSGlobal.touched x=314 y=311 (raw x=9 y=314)
19:54:21.851 TS : released x=314 y=311 (raw x=9 y=314)
19:54:24.655 TS : TSGlobal.touched x=3 y=311 (raw x=9 y=3)
19:54:24.854 TS : released x=3 y=310 (raw x=10 y=3)

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Build firmware  with PlatformIO from tasmota-master branch. Change only 'platformio_override.ini' select 'tasmota32-lvgl' from 'default_envs'.
Erase_flash then write_flash. (esptool.py --port com11 write_flash -fs 4MB -fm dout 0x0 firmware.factory.bin)
Configure device with template then upload 'display.ini' (see below) then upload 'autoexec.be' (see below) to the file system.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Button should react to touch where the button is visible not outside the button.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

n/a

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

display.ini:
```lua
:H,ST7796,480,320,16,SPI,1,*,*,*,*,*,*,*,40
:S,2,1,1,0,40,20
:I
EF,3,03,80,02
CF,3,00,C1,30
ED,4,64,03,12,81
E8,3,85,00,78
CB,5,39,2C,00,34,02
F7,1,20
EA,2,00,00
C0,1,23
C1,1,10
C5,2,3e,28
C7,1,86
36,1,48
37,1,00
3A,1,55
B1,2,00,18
B6,3,08,82,27
F2,1,00
26,1,01
E0,0F,0F,31,2B,0C,0E,08,4E,F1,37,07,10,03,0E,09,00
E1,0F,00,0E,14,03,11,07,31,C1,48,08,0F,0C,31,36,0F
11,80
29,80
:o,28
:O,29
:A,2A,2B,2C
:R,36
:0,28,00,00,01
:1,88,00,00,02
:2,E8,00,00,03
:3,48,00,00,00
:r,0
:i,20,21
:TI1,5d,32,33,21,25
#
```
autoexec.be:
```lua
lv.start()
tasmota.cmd("DisplayDimmer 100")
hres = lv.get_hor_res()       # should be 480
vres = lv.get_ver_res()       # should be 320
scr = lv.scr_act()            # default screean object
f20 = lv.montserrat_font(20)  # load embedded Montserrat 20
scr.set_style_bg_color(lv.color(0x0000A0), lv.PART_MAIN | lv.STATE_DEFAULT)
#- create a style for the buttons -#
btn_style = lv.style()
btn_style.set_radius(10)                        # radius of rounded corners
btn_style.set_bg_opa(lv.OPA_COVER)              # 100% background opacity
if f20 != nil btn_style.set_text_font(f20) end  # set font to Montserrat 20
btn_style.set_bg_color(lv.color(0x1fa3ec))      # background color #1FA3EC (Tasmota Blue)
btn_style.set_border_color(lv.color(0x0000FF))  # border color #0000FF
btn_style.set_text_color(lv.color(0xFFFFFF))    # text color white #FFFFFF
#- create buttons -#
prev_btn = lv.btn(scr)                            # create button with main screen as parent
prev_btn.set_pos(150,vres-150)                      # position of button
prev_btn.set_size(200, 50)                         # size of button
prev_btn.add_style(btn_style, lv.PART_MAIN | lv.STATE_DEFAULT)   # style of button
prev_label = lv.label(prev_btn)                   # create a label as sub-object
prev_label.set_text("<")                          # set label text
prev_label.center()
```

**(Please, remember to close the issue when the problem has been addressed)**

OperatorB, 2023-06-17:
I have finally found a way to eliminate this issue but it might not be the best way.  This modification might screw up other type of touch enabled displays so anyone with better arithmetic and programming skills should take a more professional look for a long term fix.

What I did:
edit 'GT911.h'
From
int32_t begin(TwoWire *use_wire, int8_t pin_int, int8_t pin_res, uint16_t xs, uint16_t ys);
To
int32_t begin(TwoWire *use_wire, int8_t pin_int, int8_t pin_res, uint16_t ys, uint16_t xs);

Any other tests or whatever needs to be done to verify certain things please let me know.
Thank you!

Jason2866, 2023-06-18:
@gemu2015 You know this stuff best. Correct fix?

gemu2015, 2023-06-19:
i am currently traveling will check this next week

gemu2015, 2023-06-25:
i don't know exactly what is wrong with your setup.
but i assume case 2

1. either x and y of touch is flipped

flipping x and y is already supported via bit 7 of touch rotation opcode.
like this
:0,28,00,00,81
:1,88,00,00,82
:2,E8,00,00,83
:3,48,00,00,80

2. or x and y sizes don't match touch. on gt 911 x and y screen sizes are programmed to eeprom from screen x and y definition.
in this case flip x and y in definition 
:H,ST7796,320,480,16,SPI,1,*,*,*,*,*,*,*,40

and set the desired rotation by console cmd

OperatorB, 2023-06-26:
Understood. Thank you for your input!

I have restored the source code recompiled, loaded etc. and switched X and Y in the display.ini.
The result was that the touch positions were correct however now the display is messed up(cropped).

I realized that I have also changed one more thing back then in the GT911.h file that was:
From
#define GT911_X_OUTPUT_MAX_LOW         (uint16_t)0x8048
#define GT911_X_OUTPUT_MAX_HIGH        (uint16_t)0x8049
#define GT911_Y_OUTPUT_MAX_LOW         (uint16_t)0x804A
#define GT911_Y_OUTPUT_MAX_HIGH        (uint16_t)0x804B

To
#define GT911_Y_OUTPUT_MAX_LOW         (uint16_t)0x8048
#define GT911_Y_OUTPUT_MAX_HIGH        (uint16_t)0x8049
#define GT911_X_OUTPUT_MAX_LOW         (uint16_t)0x804A
#define GT911_X_OUTPUT_MAX_HIGH        (uint16_t)0x804B

Good luck!

closed 2023-06-26 by OperatorB

RaW79, 2024-09-30:
Dear @OperatorB kindly ask if you can provide your diplay.ini and autoexec.be? I´m a absolute beginner in this and just flashed Tasmota lgvl to my ESP32-3248S035C (capacitive touch) and by reusing your files my screen works (on/off) but cannot use/test the touch button on it you created. Gues you fixed this finally and kindly ask to share. Big Thx!!!