arendst/Tasmota #18923: MiElHVAC Power Toggle and Power cmnd over MQTT won't work.

wegw3rf, 2023-06-21:
### PROBLEM DESCRIPTION
The air conditioner cannot be switched on via MQTT or the power toggle. The power cmnd is interpreted correctly according to the log, but the serial output to the air conditioner is not sent correctly.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): wemos d1 mini
- [ ] Tasmota binary firmware version number used: 12.5.0.4
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: ota
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
07:30:00.827 CMD: Backlog Template; Module; GPIO 255
07:30:00.899 MQT: stat/tasmota_EG_SZ_Klima/RESULT = {"NAME":"MiEL_Template","GPIO":[1,5376,1,5408,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
07:30:01.102 MQT: stat/tasmota_EG_SZ_Klima/RESULT = {"Module":{"18":"Generic"}}
07:30:01.339 MQT: stat/tasmota_EG_SZ_Klima/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"5376":"MiEl HVAC Tx"},"GPIO2":{"0":"None"},"GPIO3":{"5408":"MiEl HVAC Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
07:31:54.836 MQT: stat/tasmota_EG_SZ_Klima/STATUS = {"Status":{"Module":18,"DeviceName":"EG_SZ_Klima","FriendlyName":["Klimaanlage Schlafzimmer EG"],"Topic":"tasmota_EG_SZ_Klima","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
07:31:54.844 MQT: stat/tasmota_EG_SZ_Klima/STATUS1 = {"StatusPRM":{"Baudrate":2400,"SerialConfig":"8E1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz	","RestartReason":"Software/System restart","Uptime":"0T10:00:02","StartupUTC":"2023-06-20T20:31:52","Sleep":50,"CfgHolder":4617,"BootCount":24,"BCResetTime":"2023-06-18T18:37:08","SaveCount":106,"SaveAddress":"F7000"}}
07:31:54.851 MQT: stat/tasmota_EG_SZ_Klima/STATUS2 = {"StatusFWR":{"Version":"12.5.0.4(tasmota)","BuildDateTime":"2023.06.20 22:28:21","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"399/699"}}
07:31:54.857 MQT: stat/tasmota_EG_SZ_Klima/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["smcdh",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00808009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
07:31:54.871 MQT: stat/tasmota_EG_SZ_Klima/STATUS4 = {"StatusMEM":{"ProgramSize":641,"Free":360,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000407","8F8AC787","04368001","000000CF","010013C0","C000F981","04004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,44,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
07:31:54.878 MQT: stat/tasmota_EG_SZ_Klima/STATUS5 = {"StatusNET":{"Hostname":"tasmota-EG-SZ-Klima-7542","IPAddress":"192.168.1.141","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"30:93:98:B5:FC:76","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
07:31:54.884 MQT: stat/tasmota_EG_SZ_Klima/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B5FD76","MqttUser":"DVES_USER","MqttCount":9,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
07:31:54.891 MQT: stat/tasmota_EG_SZ_Klima/STATUS7 = {"StatusTIM":{"UTC":"2023-06-21T06:31:54","Local":"2023-06-21T07:31:54","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:56"}}
07:31:54.900 MQT: stat/tasmota_EG_SZ_Klima/STATUS10 = {"StatusSNS":{"Time":"2023-06-21T07:31:54","MiElHVAC":{"Temperature":26.5,"Operation":"OFF","Compressor":"OFF","roomtemp":"030000100001B5B50042000000820000","status":"06000000000001000A00004200000000","stage":"09000000000000000000000000000000"},"TempUnit":"C"}}
07:31:54.907 MQT: stat/tasmota_EG_SZ_Klima/STATUS11 = {"StatusSTS":{"Time":"2023-06-21T07:31:54","Uptime":"0T10:00:02","UptimeSec":36002,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":32,"MqttCount":9,"POWER":"OFF","Wifi":{"AP":1,"SSId":"smh","BSSId":"BA:79:F4:20:D7:EB","Channel":6,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
07:39:05.153 SRC: MQTT
07:39:05.156 CMD: Grp 0, Cmd 'POWER', Idx 1, Len 2, Pld 1, Data 'ON'
07:39:05.161 MQT: stat/tasmota_EG_SZ_Klima/RESULT = {"POWER":"ON"}
07:39:05.166 MQT: stat/tasmota_EG_SZ_Klima/POWER = ON
07:39:05.197 MiElHVAC: response data 06000000000001000A00004200000000
07:39:05.250 MiElHVAC: sending FC41013010 01010000000000000000000000000000 7c
07:39:05.461 MiElHVAC: sending FC42013010 02000000000000000000000000000000 7b
07:39:05.673 MiElHVAC: response updated 00000000000000000000000000000000
07:39:05.728 MiElHVAC: sending FC42013010 06000000000000000000000000000000 77
07:39:05.931 MiElHVAC: response data 06000000000001000A00004200000000
07:39:05.051 CFG: in Flash gespeichert an F5, Anzahl 108, Bytes 4096

```

### TO REPRODUCE
AC ist set OFF
Send over MQTT power/on
OR
Toggle Switch in Webui

### EXPECTED BEHAVIOUR
AC should be start

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
In my opinion the Line
07:39:05.250 MiElHVAC: sending FC41013010 01010000000000000000000000000000 7c
isn't correct.
I think the 8th byte should be 0x01 instead of 0x00

**(Please, remember to close the issue when the problem has been addressed)**

bobbytables2048, 2023-07-01:
Same issue when I upgraded from 12.4.0 to 13.0.0. Flashed back down to 12.4.0 and power commands are sent successfully.

s-hadinger, 2023-07-02:
I don't have the device, not know how the driver works. But I don't see any significant change that would explain the bug. Do you know which version broke it?

bobbytables2048, 2023-07-02:
Here is what we know so far
12.4.0 - OK (compiled with tasmocompiler)
12.5.0.4 - Issue (as reported by @wegw3rf)
13.0.0 - Issue (compiled with tasmocompiler)

arendst, 2023-07-03:
Try latest change in dev branch and report back.

wegw3rf, 2023-07-03:
@arendst it looks like it works. I've tested the webui and mqtt, both will start and stop the airconditioner correct. Thank you for your help!

arendst, 2023-07-03:
No problem. Was my fault anyway ;-)

closed 2023-07-03 by arendst