arendst/Tasmota #18931: TuyaMCU not communicating after init sequence on smoke detector with ESP12F replacement for CB3S

belveder79, 2023-06-22:
### PROBLEM DESCRIPTION

I got a smoke detector from AliExpress and replaced the CB3S module with an ESP12F module with stock tasmota firmware on it. The problem I'm facing is that it initializes right as you can see from the log
```lua
MCU Product ID: {"p":"fzb5up2iehjcrnqy","v":"1.0.0"}
```
but from there on nothing happens or works. The entire assembly is admittedly supposed to be super low power and I power the ESP externally to not go to sleep straight after the initialization, but I'd expect the system to be active at least for a couple of seconds (not just 2-3 or so). Anyway, the overall issue is that as far as I can see the MCU does not respond straight after the product ID. What is suspicious to me is this:

```lua
14:07:27.094 TYA: Send "55aa0002000001"
14:07:27.102 {"TuyaReceived":{"Data":"55AA0002000001","Cmnd":2}}
14:07:27.103 TYA: RX MCU configuration Mode=0
14:07:27.104 TYA: Read MCU state
14:07:27.112 TYA: Send "55aa0008000007"
```

The read from TuyaReceived is the same as the Send. I looked into the Tuya Serial Documentation but it did not really shed some insight (it should return MCU GPIO config, but it apparently does not). 

I also tried to run with the V2 version of the Tuya driver with a custom tasmota build, but it did not work at all...

Any hint would be helpful (going to sniff the serial on the original CB3S chip now to see what is communicated).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): TuyaMCU
- [x] Tasmota binary firmware version number used: 12.5.0.4
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: web/esptool/ota
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
No rules defined...

14:04:01.851 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 19, Pld -99, Data 'Rule1; Rule2; Rule3'
14:04:01.867 SRC: Backlog
14:04:01.869 CMD: Grp 0, Cmd 'RULE', Idx 1, Len 0, Pld -99, Data ''
14:04:01.874 RSL: RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
14:04:02.093 SRC: Backlog
14:04:02.096 CMD: Grp 0, Cmd 'RULE', Idx 2, Len 0, Pld -99, Data ''
14:04:02.101 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
14:04:02.345 SRC: Backlog
14:04:02.347 CMD: Grp 0, Cmd 'RULE', Idx 3, Len 0, Pld -99, Data ''
14:04:02.352 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
```
- [ ] Provide the output of this command: `Status 0`:
```lua
14:04:35.842 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
14:04:35.847 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"firealarm","FriendlyName":["firealarm2"],"Topic":"tasmota_D8CC16","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
14:04:35.853 RSL: STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:02:02","StartupUTC":"2023-06-22T13:02:33","Sleep":50,"CfgHolder":4617,"BootCount":86,"BCResetTime":"2022-10-03T11:23:35","SaveCount":135,"SaveAddress":"F9000"}}
14:04:35.856 RSL: STATUS2 = {"StatusFWR":{"Version":"12.5.0.4(tasmota)","BuildDateTime":"2023-06-21T16:34:43","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"345/699"}}
14:04:35.860 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NETGEAR81",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
14:04:35.871 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":634,"Free":368,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164068","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
14:04:35.877 RSL: STATUS5 = {"StatusNET":{"Hostname":"firealarm","IPAddress":"10.0.0.41","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer1":"10.0.0.1","DNSServer2":"0.0.0.0","Mac":"AC:0B:FB:D8:CC:16","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
14:04:35.881 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D8CC16","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:04:35.887 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-06-22T13:04:35","Local":"2023-06-22T14:04:35","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:57"}}
14:04:35.893 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-06-22T14:04:35"}}
14:04:35.898 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-06-22T14:04:35","Uptime":"0T00:02:02","UptimeSec":122,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"NETGEAR81","BSSId":"10:DA:43:84:25:B5","Channel":4,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
[... 51 lines cut ...]
14:08:45.774 WIF: Checking connection...
14:08:51.396 TYA: Send "55aa00000000ff"
14:08:51.847 HTP: Console```

### TO REPRODUCE

1) Buy https://www.aliexpress.com/item/1005005480282632.html?spm=a2g0o.order_list.order_list_main.5.6da01802l7Wht1
1) replace CB3S with ESP12 and tasmota, 
1) try...

### EXPECTED BEHAVIOUR

1) MCU communicating stuff back to tasmota, or tasmota running some init sequence to the MCU?

### SCREENSHOTS

![IMG_7620](https://github.com/arendst/Tasmota/assets/112880/b2942f1b-66d3-47df-b73b-c196256410d8)
![IMG_7622](https://github.com/arendst/Tasmota/assets/112880/c221a0c4-4d3e-4e3e-9cc6-4f0a90f801ea)
![IMG_7621](https://github.com/arendst/Tasmota/assets/112880/dd17ac72-dfd2-45c2-85a8-025e7460242b)

### ADDITIONAL CONTEXT

I know that the Tuya protocol is a work in progress and I'm happy to contribute to that if necessary. It seems that the current status is partially working for this MCU at least, but I couldn't figure out what's wrong currently...

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-06-22:
> The read from TuyaReceived is the same as the Send.
That could be suspicious and you'd better check for an permanent or intermittent contact between TX and RX.

However other logs doesn't seem to confirm
```
14:07:27.037 TYA: Send "55aa0001000000"
14:07:27.084 {"TuyaReceived":{"Data":"55AA000100247B2270223A22667A62357570326965686A63726E7179222C2276223A22312E302E30227DFA","Cmnd":1,"CmndData":"7B2270223A22667A62357570326965686A63726E7179222C2276223A22312E302E30227D"}}
```

Is this device documented in the Template repository ?
If not you may be missing some configuration
What is it supposed to do ?
Have you tried to set smoke on it to see if that triggers some messages ?

sfromis, 2023-06-22:
With a smoke detector, I'd expect battery operation, which normally means that the TuyaMCU chip as the main processor only expects the Wifi chip to be woken up when there is a need for Wifi. Existing smoke detector configs do have quite a bit more than just configuring the pins needed. Here's an example (which may not at all apply to your device).
https://templates.blakadder.com/YG400A.html

belveder79, 2023-06-22:
@barbudor actually the device is not documented yet actually. I will move on with checking on the connections between the ESP side and the MCU side. I did not put something on flames to test it, no :-) but there is a "WIFI enable" button and a "Test" button that really beeps if you press it and the one supplies power to the ESP again, which makes total sense to me. However, there is nothing happening in the Serial or nothing received if I do either of that...

@sfromis indeed the MCU puts the ESP to sleep about 30 seconds after initial power up, while "putting to sleep" refers to "do not provide any power to the ESP any more". In this respect the TuyaMCU behaves exactly the same as e.g. this one I successfully fixed to use an ESP as well: https://templates.blakadder.com/PA-210WYS.html

The bottom line right now is, "it is a smoke detector, but it is not smart (right now)". I did not test, but if I provide smoke to it, it will certainly start making noise, I'm pretty sure of that (but stopped smoking some time ago, so I need to really burn something to test)...

sfromis, 2023-06-22:
Spying on serial communication while operating with factory firmware can be a good way of learning what happens.