arendst/Tasmota #19013: When user sets color temperature to 500 (100% warm), TuyaMCU sends command to set it to approx. 90 %

kswt, 2023-07-02:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When user sets color temperature to 500 (100% warm), TuyaMCU sends command to set it to approx. 90 %. TuyaMCU sends the value "20", but "0" expected.
### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Node MCU v2 controlling Remez Leda ceiling light via TuyaMCU protocol
- [x] Tasmota binary firmware version number used:  13.0.0.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: esptool.py for first flashing, ota
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
23:50:44.976 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
23:50:45.177 RSL: RESULT = {"Module":{"54":"Tuya MCU"}}
23:50:45.385 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
23:51:32.557 RSL: RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
23:51:32.760 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
23:51:32.963 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
```
- [x] Provide the output of this command: `Status 0`:
```lua

23:56:29.329 RSL: STATUS = {"Status":{"Module":54,"DeviceName":"Remez Leda","FriendlyName":["Remez Leda"],"Topic":"tasmota_94C6D6","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
23:56:29.334 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:03:06","StartupUTC":"2023-07-02T22:53:23","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2023-07-02T00:09:11","SaveCount":53,"SaveAddress":"F8000"}}
23:56:29.338 RSL: STATUS2 = {"StatusFWR":{"Version":"13.0.0.1(tasmota)","BuildDateTime":"2023-07-01T12:33:18","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"342/699"}}
23:56:29.343 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["\\0/",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","0000E000","00004000","00000000"]}}
23:56:29.353 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":634,"Free":368,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
23:56:29.358 RSL: STATUS5 = {"StatusNET":{"Hostname":"<removed>","IPAddress":"192.168.0.165","Gateway":"192.168.1.1","Subnetmask":"255.255.0.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"<removed>","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
23:56:29.362 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_94C6D6","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
23:56:29.367 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-07-02T22:56:29","Local":"2023-07-02T23:56:29","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"04:51","Sunset":"20:56"}}
23:56:29.372 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-07-02T23:56:29"}}
23:56:29.378 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-07-02T23:56:29","Uptime":"0T00:03:06","UptimeSec":186,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":5,"Color":"000D","White":5,"CT":500,"Channel":[0,5],"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"<removed>","BSSId":"removed","Channel":11,"Mode":"11n","RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
00:00:17.959 CMD: ct 500
00:00:17.961 SRC: WebConsole from 192.168.0.125
00:00:17.964 CMD: Grp 0, Cmd 'CT', Idx 1, Len 3, Pld 500, Data '500'
00:00:17.970 RSL: RESULT = {"POWER":"ON","Dimmer":5,"Color":"000D","White":5,"CT":500,"Channel":[0,5]}
00:00:17.987 TYA: Send "55aa00060008160200040000003f68"
00:00:17.990 TYA: Send "55aa00060008170200040000000a34"
00:00:17.995 {"TuyaReceived":{"Data":"55AA03070008160200040000003F6C","Cmnd":7,"CmndData":"160200040000003F","DpType2Id22":63,"22":{"DpId":22,"DpIdType":2,"DpIdData":"0000003F"}}}
00:00:17.998 TYA: fnId=21 is set for dpId=22
00:00:17.000 TYA: RX value 63 from dpId 22 
00:00:18.002 {"TuyaReceived":{"Data":"55AA03070008170200040000001442","Cmnd":7,"CmndData":"1702000400000014","DpType2Id23":20,"23":{"DpId":23,"DpIdType":2,"DpIdData":"00000014"}}}
00:00:18.004 TYA: fnId=23 is set for dpId=23
00:00:18.006 TYA: RX value 20 from dpId 23 
00:00:18.189 {"TuyaReceived":{"Data":"55AA03070005140100010125","Cmnd":7,"CmndData":"1401000101","DpType1Id20":1,"20":{"DpId":20,"DpIdType":1,"DpIdData":"01"}}}
00:00:18.192 TYA: fnId=11 is set for dpId=20
00:00:18.194 TYA: RX Relay-1 --> MCU State: On Current State:On
00:00:18.199 {"TuyaReceived":{"Data":"55AA03070005150400010028","Cmnd":7,"CmndData":"1504000100","DpType4Id21":0,"21":{"DpId":21,"DpIdType":4,"DpIdData":"00"}}}
00:00:18.202 TYA: fnId=0 is set for dpId=21
00:00:18.205 {"TuyaReceived":{"Data":"55AA03070008160200040000003C69","Cmnd":7,"CmndData":"160200040000003C","DpType2Id22":60,"22":{"DpId":22,"DpIdType":2,"DpIdData":"0000003C"}}}
00:00:18.208 TYA: fnId=21 is set for dpId=22
00:00:18.209 TYA: RX value 60 from dpId 22 
00:00:18.212 {"TuyaReceived":{"Data":"55AA03070008170200040000001442","Cmnd":7,"CmndData":"1702000400000014","DpType2Id23":20,"23":{"DpId":23,"DpIdType":2,"DpIdData":"00000014"}}}
00:00:18.215 TYA: fnId=23 is set for dpId=23
00:00:18.216 TYA: RX value 20 from dpId 23 
00:00:18.219 {"TuyaReceived":{"Data":"55AA030700081A0200040000000031","Cmnd":7,"CmndData":"1A02000400000000","DpType2Id26":0,"26":{"DpId":26,"DpIdType":2,"DpIdData":"00000000"}}}
00:00:18.222 TYA: fnId=0 is set for dpId=26
00:00:18.224 TYA: RX value 0 from dpId 26 
00:00:20.428 TYA: Send "55aa00000000ff"
00:00:20.433 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
00:00:20.435 TYA: Heartbeat
00:00:22.412 TYA: Send "55aa001c000801170703000016015c"
00:00:24.407 WIF: Sending Gratuitous ARP
00:00:27.754 WIF: Checking connection...
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Configure dlpid 23 to function id 23:
`TuyaMCU 23, 23`
Set CT to some value:
`CT 200`
Set CT to 500 - maximum warm value:
`CT 500`
TuyaMCU will send value 20 to dlp id 23

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
TuyaMCU must send value 0x00 to dlp id 23. If user sends it manually, the color temperature of the light sets to 100 % warm value
### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

jonschz, 2023-07-08:
Is this a regression in v13.0.0, or have you encountered this issue in earlier versions as well?

jonschz, 2023-07-08:
To me it looks like the following lines are to blame (not sure if v1 or v2 is being used at the moment):
https://github.com/arendst/Tasmota/blob/d9d9ca2651363ac364f720403657e557e8a0efe8/tasmota/tasmota_xdrv_driver/xdrv_16_tuyamcu_v1.ino#L737-L741
https://github.com/arendst/Tasmota/blob/d9d9ca2651363ac364f720403657e557e8a0efe8/tasmota/tasmota_xdrv_driver/xdrv_16_tuyamcu_v2.ino#L1401-L1410
It looks like the bug was introduced in https://github.com/arendst/Tasmota/commit/e45a5f33234bf05b70427a3b6fee586b2bf3f899 three years ago. Intuitively I think the code should look like
```c
 if (TuyaIdx == 2 && CTLight) { 
   duty = changeUIntScale(duty, pTuya->CTMin, pTuya->CTMax, Settings->dimmer_hw_max, 0); 
 } else { 
   duty = changeUIntScale(duty, 0, 100, Settings->dimmer_hw_min, Settings->dimmer_hw_max); 
   // dimming acts odd below 25(10%) - this mirrors the threshold set on the faceplate itself 
   if (duty < Settings->dimmer_hw_min) { 
     duty = Settings->dimmer_hw_min; 
   } 
 }
```
but I would ask for a double check from someone else as I don't use TuyaMCU.

kswt, 2023-07-08:
> Is this a regression in v13.0.0, or have you encountered this issue in earlier versions as well?

13 is the first version that I used.

Yesterday I built Tasmota from source and this bug disappeared.

jonschz, 2023-07-08:
Maybe `DimmerRange` was set to some non-default value previously, and flashing your version reset it.

kswt, 2023-07-08:
> Maybe `DimmerRange` was set to some non-default value previously, and flashing your version reset it.

You are right:
DimmerRange 10, 1000 - bug present
DimmerRange 0, 1000 - no bug

closed 2023-08-07 by github-actions[bot]