arendst/Tasmota #19045: Tasmota32 does not apply template after long time with no power

msetina, 2023-07-06:
Using tasmota32 prebuild 13.0  () and applying basic settings and template. After a day off, turning the boad on I was surprised the template did not apply. I have an 8266 prepared with similar procedure and I am not seeing this issue.
This is from two command requested:
09:44:43.778 CMD: Backlog Template; Module; GPIO 255
09:44:43.815 MQT: stat/small_tasmota32/RESULT = {"NAME":"8relays8switches","GPIO":[230,0,231,0,229,164,0,0,0,0,0,0,0,0,161,162,0,225,224,163,0,227,160,226,0,0,0,0,228,166,167,165,0,0,0,0],"FLAG":0,"BASE":1}
09:44:44.047 MQT: stat/small_tasmota32/RESULT = {"Module":{"1":"ESP32-DevKit"}}
09:44:44.303 MQT: stat/small_tasmota32/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
09:45:56.928 CMD: Status0
09:45:56.965 MQT: stat/small_tasmota32/STATUS0 = {"Status":{"Module":1,"DeviceName":"ESP32 enota","FriendlyName":["rele1"],"Topic":"small_tasmota32","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:02:09","StartupUTC":"2023-07-06T08:43:47","Sleep":50,"CfgHolder":4617,"BootCount":27,"BCResetTime":"2023-06-29T13:38:57","SaveCount":65},"StatusFWR":{"Version":"13.0.0(tasmota32)","BuildDateTime":"2023-06-26T10:54:38","Core":"2_0_10","SDK":"4.4.5.230614","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"548/699"},"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Borealis",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000088","00006000","00004000","00000000"]},"StatusMEM":{"ProgramSize":1870,"Free":1009,"Heap":113,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","480840D2","20200000","D4BC482D","000A80B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,68,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77"},"StatusNET":{"Hostname":"small-tasmota32-4864","IPAddress":"192.168.44.31","Gateway":"192.168.44.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"7C:87:CE:30:33:00","IP6Global":"","IP6Local":"fe80::7e87:ceff:fe30:3300","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0},"StatusMQT":{"MqttHost":"192.168.1.66","MqttPort":1883,"MqttClientMask":"big_relay_box","MqttClient":"big_relay_box","MqttUser":"transmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2023-07-06T08:45:56","Local":"2023-07-06T09:45:56","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"04:54","Sunset":"20:55"},"StatusSNS":{"Time":"2023-07-06T09:45:56"},"StatusSTS":{"Time":"2023-07-06T09:45:56","Uptime":"0T00:02:09","UptimeSec":129,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":45},"Wifi":{"AP":1,"SSId":"Borealis","BSSId":"94:A6:7E:B1:72:2A","Channel":13,"Mode":"11n","RSSI":38,"Signal":-81,"LinkCount":1,"Downtime":"0T00:00:03"}}}

msetina, 2023-07-06:
Board was flashed with tasmota32.factory.bin .
After reloading template commands return this:
09:58:12.380 CMD: Backlog Template; Module; GPIO 255
09:58:12.408 MQT: stat/small_tasmota32/RESULT = {"NAME":"8relays8switches","GPIO":[230,0,231,0,229,164,0,0,0,0,0,0,0,0,161,162,0,225,224,163,0,227,160,226,0,0,0,0,228,166,167,165,0,0,0,0],"FLAG":0,"BASE":1}
09:58:12.636 MQT: stat/small_tasmota32/RESULT = {"Module":{"0":"8relays8switches"}}
09:58:12.845 MQT: stat/small_tasmota32/RESULT = {"GPIO0":{"230":"Relay7"},"GPIO1":{"0":"None"},"GPIO2":{"231":"Relay8"},"GPIO3":{"0":"None"},"GPIO4":{"229":"Relay6"},"GPIO5":{"164":"Switch5"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"161":"Switch2"},"GPIO19":{"162":"Switch3"},"GPIO20":{"0":"None"},"GPIO21":{"225":"Relay2"},"GPIO22":{"224":"Relay1"},"GPIO23":{"163":"Switch4"},"GPIO24":{"0":"None"},"GPIO25":{"227":"Relay4"},"GPIO26":{"160":"Switch1"},"GPIO27":{"226":"Relay3"},"GPIO32":{"228":"Relay5"},"GPIO33":{"166":"Switch7"},"GPIO34":{"167":"Switch8"},"GPIO35":{"165":"Switch6"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
09:58:30.381 CMD: Status0
09:58:30.420 MQT: stat/small_tasmota32/STATUS0 = {"Status":{"Module":0,"DeviceName":"ESP32 enota","FriendlyName":["rele1","rele2","rele3","rele4","rele5","rele6","rele7","rele8"],"Topic":"small_tasmota32","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:01:27","StartupUTC":"2023-07-06T08:57:03","Sleep":50,"CfgHolder":4617,"BootCount":29,"BCResetTime":"2023-06-29T13:38:57","SaveCount":68},"StatusFWR":{"Version":"13.0.0(tasmota32)","BuildDateTime":"2023-06-26T10:54:38","Core":"2_0_10","SDK":"4.4.5.230614","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"548/699"},"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Borealis",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000088","00006000","00004000","00000000"]},"StatusMEM":{"ProgramSize":1870,"Free":1009,"Heap":111,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","480840D2","20200000","D4BC482D","000A80B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,68,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77"},"StatusNET":{"Hostname":"small-tasmota32-4864","IPAddress":"192.168.44.31","Gateway":"192.168.44.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"7C:87:CE:30:33:00","IP6Global":"","IP6Local":"fe80::7e87:ceff:fe30:3300","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0},"StatusMQT":{"MqttHost":"192.168.1.66","MqttPort":1883,"MqttClientMask":"big_relay_box","MqttClient":"big_relay_box","MqttUser":"transmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2023-07-06T08:58:30","Local":"2023-07-06T09:58:30","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"04:54","Sunset":"20:55"},"StatusSNS":{"Time":"2023-07-06T09:58:30","stikalo1":"ON","stikalo2":"ON","stikalo3":"ON","stikalo4":"ON","stikalo5":"ON","stikalo6":"OFF","stikalo7":"ON","stikalo8":"OFF"},"StatusSTS":{"Time":"2023-07-06T09:58:30","Uptime":"0T00:01:27","UptimeSec":87,"Heap":110,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":45},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","Wifi":{"AP":1,"SSId":"Borealis","BSSId":"94:A6:7E:B1:72:2A","Channel":13,"Mode":"11n","RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:02"}}}

sfromis, 2023-07-06:
For readability, I'd suggest to wrap those copy/paste text blocks in triple backtick before and after.

Anyway the key thing is that the template is not activated in the first one, as `Module` being 0 means template activated, and thus with Module 1, not so. What you post does not give any indication of cause, or what any prior state even was. However, in case of boot loops (for whatever reason), Tasmota may "scale back" features until able to boot without crashing. This could include deactivating the template.

msetina, 2023-07-06:
Thank you for pointing this out about the Module 1.
Restart from WebUI does not turn off template.
Neither does reset on the board.
I have noticed that I used 2 GPIO thet are input only and have no pullup resistor, but in template I used normal Switch component. For those two WebUI and SENSOR on MQTT do not report the same state.
First I will try to find a replicable procedure.