arendst/Tasmota #19068: PWM1 seems to be not defined.

bennettopaul, 2023-07-08:
I've created a template for a Kogan moon lamp, and I've found all the channels, but I'm having some weird behavior with the PWM channels.

I noticed I needed to set red to PWM2 for color=#aa00000000 to illuminate it, and I worked my way through all the other channels until I got to WW. I noticed that there was no way to set warm white, as I'd effectively ran out of PWM channels. So why does PWM1 not work?

Strange sample output from the console is below:
05:01:13.778 CMD: color1=#aa000000
05:01:13.788 RSL: RESULT = {"POWER":"ON","Dimmer":67,"Color":"00AA000000","HSBColor":"120,100,67","White":0,"CT":500,"Channel":[0,67,0,0,0]}

barbudor, 2023-07-08:
Hi
You don't give much details on your actual findings such as your template and if you have open the device or just tried different templates. 

Have you actually identified 5 individualy working PWM?
If you define only PWM1 and move it across the 5 identified gpios, can you control each color on it's own?

I'm asking that because some lights are using mixed control with some channels directly on PWM gpios and some on a separated led driver chip (often i2c)

bennettopaul, 2023-07-08:
Template is currently:
{"NAME":"Kogan Moon","GPIO":[0,0,0,0,417,420,0,0,418,416,419,0,0,0],"FLAG":0,"BASE":18}
Pretty sure it's not a problem with the template. There are no extra driver chips and I can control each of the LEDs individually by assigning them to PWM2-5. The trouble is that red starts at PWM2, and PWM1 doesn't seem to be assigned to any of the R G B CW WW.
Look at the weird output of the color command. The channels get shifted by one. My input was AA00000000 but the output was 00AA000000.

bennettopaul, 2023-07-08:
Ah, my bad. I was using "color=#AA00000000" instead of "color AA00000000"

closed 2023-07-08 by bennettopaul