arendst/Tasmota #19084: DeepSleepTime documentation MQTT topic is case sensitive

OldSurferDude, 2023-07-09:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
The [documentation](https://tasmota.github.io/docs/DeepSleep/#using-mqtt), just below **Example: Sending using mosquitto utilities** shows "DeepsleepTime"  In other places the topic is "DeepSleepTime"  The former doesn't work, the latter does.  Just a documentation fix.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP-12F
- [x] Tasmota binary firmware version number used: 13.0.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: esptool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
16:44:32.972 MQT: stat/tasmota_3257B3/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1184,1,1,1,1],"FLAG":0,"BASE":18}
16:44:33.208 MQT: stat/tasmota_3257B3/RESULT = {"Module":{"0":"Generic"}}
16:44:33.461 MQT: stat/tasmota_3257B3/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"1184":"DHT11"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
16:46:09.632 CMD: Backlog Rule1; Rule2; Rule3
16:46:09.702 MQT: stat/tasmota_3257B3/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
16:46:09.912 MQT: stat/tasmota_3257B3/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
16:46:10.164 MQT: stat/tasmota_3257B3/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:47:17.963 CMD: Status 0
16:47:17.972 MQT: stat/tasmota_3257B3/STATUS = {"Status":{"Module":0,"DeviceName":"ESP-12F","FriendlyName":["ESP-12F"],"Topic":"tasmota_3257B3","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
16:47:17.980 MQT: stat/tasmota_3257B3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Deep-Sleep Wake","Uptime":"0T00:28:12","StartupUTC":"2023-07-09T23:19:05","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2023-07-06T21:06:33","SaveCount":23,"SaveAddress":"F6000"}}
16:47:17.985 MQT: stat/tasmota_3257B3/STATUS2 = {"StatusFWR":{"Version":"13.0.0(tasmota)","BuildDateTime":"2023-06-26T10:42:07","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"356/699"}}
16:47:17.990 MQT: stat/tasmota_3257B3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Automation",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
16:47:18.000 MQT: stat/tasmota_3257B3/STATUS4 = {"StatusMEM":{"ProgramSize":634,"Free":368,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
16:47:18.008 MQT: stat/tasmota_3257B3/STATUS5 = {"StatusNET":{"Hostname":"tasmota-3257B3-6067","IPAddress":"192.168.1.226","Gateway":"192.168.1.193","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.193","DNSServer2":"0.0.0.0","Mac":"84:F3:EB:32:57:B3","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
16:47:18.013 MQT: stat/tasmota_3257B3/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.142","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3257B3","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:47:18.020 MQT: stat/tasmota_3257B3/STATUS7 = {"StatusTIM":{"UTC":"2023-07-09T23:47:18","Local":"2023-07-09T16:47:18","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"-07:00","Sunrise":"20:56","Sunset":"12:53"}}
16:47:18.026 MQT: stat/tasmota_3257B3/STATUS10 = {"StatusSNS":{"Time":"2023-07-09T16:47:18","DHT11":{"Temperature":24.6,"Humidity":75.0,"DewPoint":19.9},"TempUnit":"C"}}
16:47:18.032 MQT: stat/tasmota_3257B3/STATUS11 = {"StatusSTS":{"Time":"2023-07-09T16:47:18","Uptime":"0T00:28:13","UptimeSec":1693,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Automation","BSSId":"04:D9:F5:96:DD:D0","Channel":1,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Sent to the MQTT broker the topic 
`cmnd/tasmota_3257B3/DeepsleepTime
`payload
`0
`and the DeepSleepTime was not set to zero
Sent to the MQTT broker the topic 
`cmnd/tasmota_3257B3/DeepSleepTime
`payload
`0
`and the DeepSleepTime was set to zero

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Expected DeepSleepTime to be set to 0 as the documentation showed

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Whew!  a lot of info just for a doc bug

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-07-10:
Doing a simple test, I found it to be working when publishing a command with `DeepSleepTime` as part of the topic, as expected. Do note that if you publish the command while the device is already in deep sleep, it will not be "seen", unless you publish it as retained. But you did not report any details of what/how you specifically made the commands.

barbudor, 2023-07-10:
Missing logs:
[image]

All tasmota commands are case insensitive so you probably have made a typo somewhere but as you didn't provided any logs there is nowhere to tell where