arendst/Tasmota #19090: Cannot update from 12.5.0(tasmota32) to 13.0.0 via OTA

jheisel01, 2023-07-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Cannot update from 12.5.0(tasmota32) to 13.0.0 via OTA
update via Fileupload: tasmota32.bin says:
**Upload Failed**
Program flash size is larger than real flash size

OTA updates from 12.5.0 to 13.0.0 on esp8266 devices works.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Dual R3 ZCD
- [x ] Tasmota binary firmware version number used: 13.0.0
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:28:06.113 CMD: Backlog Template; Module; GPIO 255
21:28:06.155 MQT: sonoff/sonoff_009/stat/RESULT = {"NAME":"Sonoff Dual R3 (ZCD)","GPIO":[32,0,0,0,7552,0,0,0,0,576,225,0,0,0,0,0,0,0,0,0,0,3200,8128,224,0,0,0,0,160,161,0,0,0,0,0,0],"FLAG":0,"BASE":1}
21:28:06.383 MQT: sonoff/sonoff_009/stat/RESULT = {"Module":{"0":"Sonoff Dual R3 (ZCD)"}}
21:28:06.597 MQT: sonoff/sonoff_009/stat/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"7552":"ZC Pulse"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"225":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"3200":"Serial Tx"},"GPIO26":{"8128":"BL0939 Rx"},"GPIO27":{"224":"Relay1"},"GPIO32":{"160":"Switch1"},"GPIO33":{"161":"Switch2"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
21:28:22.724 MQT: sonoff/sonoff_009/tele/STATE = {"Time":"2023-07-10T21:28:22","Uptime":"70T10:26:12","UptimeSec":6085572,"Heap":148,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":8,"Berry":{"HeapUsed":3,"Objects":44},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"mywlR2400","BSSId":"D2:32:E5:7C:A2:6E","Channel":11,"Mode":"11n","RSSI":68,"Signal":-66,"LinkCount":4,"Downtime":"0T00:00:22"}}
21:28:22.740 MQT: sonoff/sonoff_009/tele/SENSOR = {"Time":"2023-07-10T21:28:22","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"1970-01-01T00:00:00","Total":0.131,"Yesterday":0.000,"Today":0.000,"Period":[0,0],"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":0,"Current":[0.000,0.000],"BL09XX":{"Temperature":48.0}},"TempUnit":"C"}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:28:54.071 CMD: Status 0
21:28:54.079 MQT: sonoff/sonoff_009/stat/STATUS = {"Status":{"Module":0,"DeviceName":"Sonoff_009","FriendlyName":["Sonoff_009","Sonoff_009"],"Topic":"sonoff_009","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
21:28:54.086 MQT: sonoff/sonoff_009/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"70T10:26:44","StartupUTC":"2023-05-01T10:02:10","Sleep":50,"CfgHolder":4617,"BootCount":24,"BCResetTime":"2022-08-27T16:25:05","SaveCount":170}}
21:28:54.091 MQT: sonoff/sonoff_009/stat/STATUS2 = {"StatusFWR":{"Version":"12.5.0(tasmota32)","BuildDateTime":"2023-04-17T08:07:16","Core":"2_0_7","SDK":"4.4.4.20230310","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"426/699"}}
21:28:54.096 MQT: sonoff/sonoff_009/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mywlR2400",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
21:28:54.115 MQT: sonoff/sonoff_009/stat/STATUS4 = {"StatusMEM":{"ProgramSize":1525,"Free":1856,"Heap":142,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","480840D2","20200000","D4BC482D","000A80B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77"}}
21:28:54.127 MQT: sonoff/sonoff_009/stat/STATUS5 = {"StatusNET":{"Hostname":"Sonoff-D-009","IPAddress":"192.168.168.43","Gateway":"192.168.168.112","Subnetmask":"255.255.255.0","DNSServer1":"8.8.8.8","DNSServer2":"8.8.4.4","Mac":"94:3C:C6:C6:59:0C","IP6Global":"","IP6Local":"fe80::963c:c6ff:fec6:590c","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"8.8.8.8","DNSServer2":"8.8.4.4","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
21:28:54.132 MQT: sonoff/sonoff_009/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.168.12","MqttPort":1883,"MqttClientMask":"Sonoff_009","MqttClient":"Sonoff_009","MqttUser":"DVES_USER","MqttCount":8,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:28:54.138 MQT: sonoff/sonoff_009/stat/STATUS7 = {"StatusTIM":{"UTC":"2023-07-10T20:28:54","Local":"2023-07-10T21:28:54","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"04:57","Sunset":"20:52"}}
21:28:54.144 MQT: sonoff/sonoff_009/stat/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0,0,0,0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
21:28:54.162 MQT: sonoff/sonoff_009/stat/STATUS10 = {"StatusSNS":{"Time":"2023-07-10T21:28:54","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"1970-01-01T00:00:00","Total":0.131,"Yesterday":0.000,"Today":0.000,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":0,"Current":[0.000,0.000],"BL09XX":{"Temperature":48.2}},"TempUnit":"C"}}
21:28:54.173 MQT: sonoff/sonoff_009/stat/STATUS11 = {"StatusSTS":{"Time":"2023-07-10T21:28:54","Uptime":"70T10:26:44","UptimeSec":6085604,"Heap":144,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":8,"Berry":{"HeapUsed":3,"Objects":44},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"mywlR2400","BSSId":"D2:32:E5:7C:A2:6E","Channel":11,"Mode":"11n","RSSI":68,"Signal":-66,"LinkCount":4,"Downtime":"0T00:00:22"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Firmwareupdate via OTA

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

<html>
<body>

ESP Chip Id | 12998924 (ESP32-D0WD-V3 rev.3)
-- | --
Flash Chip Id  | 0x16405E (DOUT)
Flash size | 4096 KB
Program Flash Size | 4096 KB
Program Size | 1525 KB
Free Program Space | 1856 KB
Free Memory | 144.8 KB (frag. 26%)
 
Partition app0 | 1856 KB
Partition app1* | 1856 KB (used 82%)
Partition fs | 320 KB

</body>
</html>

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-07-10:
See the solution in https://github.com/arendst/Tasmota/discussions/18983

jheisel01, 2023-07-11:
> You "forgot" to search in discussions. See the solution in #18983

Hi, maybe it was too late for thinking 😴. I upload the Partion Wizzard let it do some magic and now I have version 13.0.0.
Thanks for your answer !

jheisel01, 2023-07-11:
Have to reopen the issue.
One device was sucessfully updated to 13.0.0 with the solution  #18983 
The second device doesn't. 
I have tried several times with the Partition Wizard.
[image]