arendst/Tasmota #19114: [Shutter] Inverted shutter behaves not like expected

SteWers, 2023-07-14:
### PROBLEM DESCRIPTION
Inverted shutter behaves not like expected.

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Designator3 and other ESP 8266 devices
- [x] Tasmota binary firmware version number used: 13.0.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
{"NAME":"Designator3","GPIO":[0,0,0,0,225,0,0,0,32,224,33,0,544,0],"FLAG":0,"BASE":18}
{"Module":{"0":"Designator3"}}
{"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"225":"Relay2"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"32":"Button1"},"GPIO13":{"224":"Relay1"},"GPIO14":{"33":"Button2"},"GPIO15":{"0":"None"},"GPIO16":{"544":"LedLink"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
No rules
```
- [x] Provide the output of this command: `Status 0`:
```lua
STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota",""],"Topic":"Shutter2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-DE.bin.gz","RestartReason":"Power On","Uptime":"0T18:32:52","StartupUTC":"2023-07-13T14:14:31","Sleep":50,"CfgHolder":4617,"BootCount":20,"BCResetTime":"2023-01-17T13:07:37","SaveCount":217,"SaveAddress":"F6000"}}
STATUS2 = {"StatusFWR":{"Version":"13.0.0(tasmota)","BuildDateTime":"2023.06.26 10:58:29","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"354/699"}}
STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["JRF-73529421",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C80001000600003C5A0A192800000000","40000080","00006000","00004000","00000000"]}}
STATUS4 = {"StatusMEM":{"ProgramSize":635,"Free":368,"Heap":21,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000407","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
STATUS5 = {"StatusNET":{"Hostname":"Shutter2-4508","IPAddress":"10.100.200.106","Gateway":"10.100.200.1","Subnetmask":"255.255.255.0","DNSServer1":"10.100.200.1","DNSServer2":"0.0.0.0","Mac":"84:CC:A8:B7:F1:9C","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
STATUS6 = {"StatusMQT":{"MqttHost":"10.100.200.222","MqttPort":1883,"MqttClientMask":"Shutter2_%06X","MqttClient":"Shutter2_B7F19C","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
STATUS7 = {"StatusTIM":{"UTC":"2023-07-14T08:47:23","Local":"2023-07-14T10:47:23","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"06:01","Sunset":"21:50"}}
STATUS10 = {"StatusSNS":{"Time":"2023-07-14T10:47:23","Shutter1":{"Position":0,"Direction":0,"Target":100,"Tilt":0}}}
STATUS11 = {"StatusSTS":{"Time":"2023-07-14T10:47:23","Uptime":"0T18:32:52","UptimeSec":66772,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"JRF-73529421","BSSId":"09:ab:2F:07:76:15","Channel":6,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:05"}}}
STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0001","Calib":[300,500,700,900,1000],"Mode":"1"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- setup a device as shutter
- set ShutterInvert1 1

### EXPECTED BEHAVIOUR
When a shutter is used in inverted mode it behaves not like expected. In version 12 everything was OK. I tried to show it in the following table:

Test|ShutterInvert<x> 0|ShutterInvert<x> 1 (expected behavior)|ShutterInvert<x> 1 (real behavior)
---|---|---|---
Relay 1|open shutter|open shutter|close shutter
Relay 2|close shutter|close shutter|open shutter
Webbutton UP|open shutter|open shutter|closing shutter (Relay 2), slider moves to CLOSE
Webbutton DOWN|close shutter|close shutter|opening shutter (Relay 1), slider moves to OPEN
Slider OPEN|open shutter|open shutter|closing shutter (Relay 2); click on OPEN, when it is OPEN(!?); slider moves to CLOSE
Slider CLOSE|close shutter|close shutter|opening shutter (Relay 1); click on CLOSE, when it is CLOSE(!?); slider moves to OPEN
ShutterOpen<x>|open shutter|open shutter|closing shutter (Relay 2), slider moves to CLOSE
ShutterClose<x>|close shutter|close shutter|opening shutter (Relay 1), slider moves to OPEN
ShutterPosition<x> 100|open shutter|close shutter|opening shutter (Relay 1), slider moves to OPEN
ShutterPosition<x> 0|close shutter|open shutter|closing shutter (Relay 2), slider moves to CLOSE
Reported position when open|100|0|Not clear; slider OPEN -> 0; shutter open -> 100
Reported position when close|0|100|Not clear; slider CLOSE -> 100; shutter close -> 0

NoName1311, 2023-07-14:
For me, the issue also occurs. Version 12.5.0 works as expected.

stefanbode, 2023-08-02:
uups missed that conversation. Let me check. where is get messed up. The idea of INVERTED that ONLY the reporting 0..100% is inverted. and that shutterposition 20 for example executes a shutterposition 80. There was some updates on the WEBUI that now updates during the move. And the WebUI was not correct with the INVERTED. Let me double check on this.

stefanbode, 2023-08-02:
<html>
<body>
<p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px; color: rgb(31, 35, 40); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">When a shutter is used in inverted mode it behaves not like expected. In version 12 everything was OK. I tried to show it in the following table:</p>

Test | ShutterInvert 0 | ShutterInvert 1 (expected behavior) | ShutterInvert 1 (real behavior)
-- | -- | -- | --
Relay 1 | open shutter | open shutter | close shutter (fixed/checked)
Relay 2 | close shutter | close shutter | open shutter (fixed/checked)
Webbutton UP | open shutter | open shutter | closing shutter (Relay 2), slider moves to CLOSE (fixed/checked)
Webbutton DOWN | close shutter | close shutter | opening shutter (Relay 1), slider moves to OPEN (fixed/checked)
Slider OPEN | open shutter | open shutter | closing shutter (Relay 2); click on OPEN, when it is OPEN(!?); slider moves to CLOSE (fixed/checked)
Slider CLOSE | close shutter | close shutter | opening shutter (Relay 1); click on CLOSE, when it is CLOSE(!?); slider moves to OPEN (fixed/checked)
ShutterOpen | open shutter | open shutter | closing shutter (Relay 2), slider moves to CLOSE (fixed/checked)
ShutterClose | close shutter | close shutter | opening shutter (Relay 1), slider moves to OPEN (fixed/checked)
ShutterPosition 100 | open shutter | close shutter | opening shutter (Relay 1), slider moves to OPEN (fixed/checked)
ShutterPosition 0 | close shutter | open shutter | closing shutter (Relay 2), slider moves to CLOSE (fixed/checked)
Reported position when open | 100 | 0 | Not clear; slider OPEN -> 0; shutter open -> 100 (pls recheck)
Reported position when close | 0 | 100 | Not clear; slider CLOSE -> 100; shutter close -> 0 (pls recheck)

</body>
</html>

SteWers, 2023-08-02:
At the moment I can't test it, as I am enjoying the sun of Greece instead of the rain in Germany. I'll test it, as soon as I am back.

stefanbode, 2023-08-03:
> At the moment I can't test it, as I am enjoying the sun of Greece instead of the rain in Germany. I'll test it, as soon as I am back.

We can change. I replace you in greece and you can test in rainy, cold germany. Let me know where to go. :-)

SteWers, 2023-08-03:
> We can change. I replace you in greece and you can test in rainy, cold germany. Let me know where to go. :-)

Hmm. 🤔 NO! 😂

stefanbode, 2023-08-08:
I think we can close this. @SteWers will check after holiday, but I'm confident to pass the test.

SteWers, 2023-08-08:
I think so, too.

closed 2023-08-08 by SteWers

sukram3001, 2023-08-16:
Looks like something got lost in the change - see #19328

SteWers, 2023-08-17:
@stefanbode: I saw you found a bug. The problem from @sukram3001 seem not be present on an ESP8266. But there are a few things, which are till not solved.

In short:
- Web buttons are swapped
- Clicking on the slider is swapped
- Commands `ShutterOpen` and `ShutterClose` are swapped

NoName1311, 2023-09-10:
@SteWers can you please reopen the issue? I can confirm that this is only partially resolved.

SteWers, 2023-09-10:
@NoName1311 I think, it's better, when you open a new issue. This is very old, and I cannot see any problems with the last dev on an ESP8266.

stefanbode, 2023-09-11:
yes open a new one. or just post here on the closed one. Both fine for me to get it fixed

NoName1311, 2023-09-17:
I tested it with 13.1.0 and the bugs mentioned above were still there. I will test it again with the next release and then open a new issue if necessary. Thank you for your good work

stefanbode, 2023-09-17:
Can you please check against the actual development branch. Thx

NoName1311, 2023-09-23:
Please excuse my late reply. Today I downloaded and tested the development branch V13.1.0.3. In this version the issue is resolved. Thank you again for the good work.

stueB, 2024-02-14:
hello there again :) imho this problem is not fully resolved. 

my configuration:
tasmota 13.3 on Shelly2PM-ESP32

```
Interlock ON
SetOption80 1
ShutterMode 1
ShutterRelay1 1
ShutterInvert 1 (or 0)
```

In your table above "ShutterPosition UP" and "ShutterPosition DOWN" are not mentioned. 

### expected on ShutterInvert 0 _and_ ShutterInvert 1:
ShutterPosition UP -> Shutter moves to top position
ShutterPosition DOWN -> Shutter moves to lowest position

### in my case:
**ShutterPosition DOWN resolves in "Command: Error"** (regardless of the ShutterInvert Mode and of the current position)
ShutterInvert 0: ShutterPosition UP -> Shutter moves to top position (as expected)
**ShutterInvert 1: ShutterPosition UP -> Shutter moves to lowest position** (thats wrong imho)

stefanbode, 2024-02-15:
Regression accepted. Will fix. Up down does not work on inverted shutter

stefanbode, 2024-02-18:
@stueB : Thanks for bringing it up. A fix is on its way. Pls let me know if you still have issues. Tested on ESP32 and ESP8266

Andi-Coder, 2024-05-23:
Hello, I have another problem with tasmota 14.0.0, esp8266 and the invert function.
The error occurs when the shutter is fully closed or open (0 or 100) and I try to get the shutter position via the web api "http://device1/cm?cmnd=shutterposition".
No more correct direction MQTT open/close command is executed after the web query anymore.
Only an opposite mqtt command causes a movement.
Example: Shutter is fully open, I query shutterposition through web api and than after that I want to close the shutter through mqtt command. No reaction happed. Only when I send an MQTT open command, but then is the shutter closed.

stefanbode, 2024-05-23:
Hmm, strange that calling a position screw it all up. Can you attach some log files with loglevel 4. I am on holiday with limited testing equipment. Please do it when shutter is at 30% to get most info.

Andi-Coder, 2024-05-24:
I've tested a few scenarios. It looks like a web query will always swap the invert function for opening and closing on the first mqtt command.

`13:00:59.974 HTP: Command
13:00:59.979 SRC: WebCommand from 192.168.0.10
13:00:59.981 CMD: Grp 0, Cmd 'SHUTTERPOSITION', Idx 1, Len 0, Pld -99, Data ''
13:00:59.985 SHT: Pos. payload <> (0), payload -99, idx 1 (0), src 8
13:00:59.990 MQT: stat/dev1/SHUTTER = {"Shutter1":{"Position":38,"Direction":0,"Target":38,"Tilt":0}}
13:01:08.133 WIF: Checking connection...
13:01:08.536 SRC: MQTT
13:01:08.539 CMD: Grp 0, Cmd 'SHUTTERPOSITION', Idx 1, Len 1, Pld 0, Data '0'
13:01:08.543 SHT: Pos. payload <0> (1), payload 0, idx 1 (1), src 8
13:01:08.545 SHT: lastsource 8:, real 322579, target 0, tiltreal: 0, tilttarget: 0, payload 0
13:01:08.547 SHT: Delay Start? var1 <99>=<>, max10s?
13:01:08.549 SRC: Shutter
13:01:08.551 SHT: FUNC_SET_POWER Relaymask 0 SwitchedRelay:2 by Shutter, payload 0, powermask 2
13:01:08.552 SHT: FUNC_SET_POWER end. powermask 2
13:01:08.558 MQT: stat/dev1/POWER2 = {"POWER2":"ON"}
13:01:08.562 MQT: stat/dev1/POWER2 = ON
13:01:08.565 SHT: Shtr1 Real 322579, Start 322579, Stop 0, Dir -1, Delay 0, Rtc 0.00 [s], Freq 0, PWM 0, Tilt 0
13:01:08.570 MQT: stat/dev1/SHUTTER = {"Shutter1":{"Position":38,"Direction":-1,"Target":100,"Tilt":0}}
13:01:08.574 SHT: Shtr1 Time 0(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 322579, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:08.587 SHT: Shtr1 Time 0(-1), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 322579, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:08.636 SHT: Shtr1 Time 1(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 322579, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:08.686 SHT: Shtr1 Time 2(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 321532, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:08.736 SHT: Shtr1 Time 3(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 320485, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:08.787 SHT: Shtr1 Time 4(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 319438, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:08.836 SHT: Shtr1 Time 5(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 318391, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:08.887 SHT: Shtr1 Time 6(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 317344, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:08.937 SHT: Shtr1 Time 7(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 316297, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:08.987 SHT: Shtr1 Time 8(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 315250, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:09.038 SHT: Shtr1 Time 9(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 314203, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:09.088 SHT: Shtr1 Time 10(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 313156, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:09.138 SHT: Shtr1 Time 11(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 312109, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:09.188 SHT: Shtr1 Time 12(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 311062, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:09.237 SHT: Shtr1 Time 13(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 310015, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:09.240 SHT: Shtr1 Real 310015, Start 322579, Stop 0, Dir -1, Delay 0, Rtc 0.65 [s], Freq 0, PWM 0, Tilt 0
13:01:09.246 MQT: stat/dev1/SHUTTER = {"Shutter1":{"Position":42,"Direction":-1,"Target":100,"Tilt":0}}
13:01:09.288 SHT: Shtr1 Time 14(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 308968, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:09.338 SHT: Shtr1 Time 15(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 307921, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:09.429 SHT: Shtr1 Time 17(1), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 305827, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:09.437 SHT: Shtr1 Time 17(-1), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 305827, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:09.487 SHT: Shtr1 Time 18(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 304780, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:09.537 SHT: Shtr1 Time 19(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 303733, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:09.588 SHT: Shtr1 Time 20(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 302686, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
13:01:09.638 SHT: Shtr1 Time 21(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 301639, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
[... cut ...]
13:01:09.005 SHT: FUNC_SET_POWER Relaymask 2 SwitchedRelay:2 by Shutter, payload -99, powermask 0
13:01:09.007 SHT: FUNC_SET_POWER end. powermask 0
13:01:10.001 MQT: stat/dev1/POWER2 = {"POWER2":"OFF"}
13:01:10.004 MQT: stat/dev1/POWER2 = OFF
13:01:10.007 SHT: Shtr1 Real 294310, Start 294310, Stop 294310, Dir 0, Delay 0, Rtc 1.40 [s], Freq 0, PWM 0, Tilt 0
13:01:10.012 MQT: stat/dev1/SHUTTER1 = 47
13:01:10.017 MQT: stat/dev1/SHUTTER = {"Shutter1":{"Position":47,"Direction":0,"Target":47,"Tilt":0}}`

stefanbode, 2024-05-29:
sorry for the delay. weather was very good on holiday. I found the root cause of the behavior and will work on it. Due to limited testing it will take some time. I was able to reproduce the issue.

stefanbode, 2024-06-10:
There is a fix submitted a few minutes ago to the DEV branch. Please give it a try. should now work with MQTT

Andi-Coder, 2024-06-14:
Just tested dev branch. If I now drive to 0, it goes to 100 and vice versa.
[log.txt](https://github.com/user-attachments/files/15838061/log.txt)

stefanbode, 2024-06-17:
Isn't this as it should be?

Andi-Coder, 2024-06-18:
No, exactly the opposite happens when controlled via MQTT

plu, 2024-06-19:
Maybe we need some control over which relay does what. So I can configure that relay1 is moving the shutter up, and relay2 is moving the shutter down?

stefanbode, 2024-06-20:
Be back from holiday and now have access again to the testbench. I can agree that the fix was only one part and need an additional fix. Submitted: #21663

stueB, 2024-10-01:
> @stueB : Thanks for bringing it up. A fix is on its way. Pls let me know if you still have issues. Tested on ESP32 and ESP8266

I’ve finally managed to test it, and it’s been working with no problems with my settup so far.

stefanbode, 2024-10-27:
If you just need it revert them simple change relay 1 and 2

Unintendedsideeffects, 2024-10-27:
Not the avenue for this, maybe I'll write in discussion but physical wiring
is inverted and so if you invert Relay 1 and 2 you actually invert button
behavior, shutterinvert seemed the correct solution

stefanbode, 2024-10-28:
Maybe you write down what is your actual configuration and what do you want to archive. Relay 1 is up and 2 is down. This is mandatory. Is it is mounted in the wall and you cannot change them just flip the gpio on the configuration page. Now shutterclose and open should do what you want. If your integration now has a problem if 0 or 100 is required for the open position then use Shutterinvert. That's all you need.

PR #20752 (2024-02-18): fix issue on inverted shutters with commands comming with shutterposition
## Description:
fix #19114 with "shutterposition" commands on INVERTED shutter

**Related issue (if applicable):** fixes #<Tasmota issue number goes here>

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.6
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0. [... cut]