arendst/Tasmota #19125: No shutter support any more since update from 12.1.1 to 13.0.0

Ansgar66, 2023-07-15:
Hi community,
my Dual R2 worked fine to drive my ordinary shutter.
for some other issue I updated the device today (from 12.1.1 to 13.0.0)
Since then, the device lost all capabilities to work as a shutter and blinds driver.
setoption80 is still on,
shuttermode is 1,
but as said above, the "shutter-UI" keeps beeing lost. (I see the two togglebuttons)

Trying to downgrade didn't work either "not enough memory"

my device is a Sonoff Dual R2:

ESP Chip Id	867949 (ESP8285N08)
Flash Chip Id	0x144051 (DOUT)
Flash Size	1024 KB
Program Flash Size	1024 KB
Program Size	634 KB
Free Program Space	368 KB
Free Memory	24.9 KB

gan anybody help?

Thanks in advance,
Ansgar

SteWers, 2023-07-16:
I've no issue with the shutter in general using version 13. Only with the inverted mode. Therefore I raised another issue.
When you upload a binary directly, remember to go the step over the "minimal" version.
The easiest way to get e.g. version 12.5 is to use this as OTA-URL: http://ota.tasmota.com/tasmota/release-12.5.0/tasmota.bin.gz
Then the "minimal" step is done automatically and your shutter should work again. The problem with "lost" shutter has to investigated further. May be it works, when you update from 12.5 to 13 again.

Jason2866, 2023-07-16:
Duplicate of #19114

closed 2023-07-16 by Jason2866

SteWers, 2023-07-16:
@Jason2866 I don't think it's related to the other issue.

Jason2866, 2023-07-16:
Still it is not a valid issue. Template not filled out. You can open a Discussion

Ansgar66, 2023-07-16:
Dear Jason and SteWers
thank you for your replies and sorry for not being conform  with the regularities of this platform.
I tried to read through the rules but must admit my english is insufficient, but also deepl-translation did not really shed light on all the corners...

I share SteWers opinion the issues are not the same. I rather assume it's lack of memory...

I managed to downgrade to 12.5.0, but still the system answers  with errors if I try to command shutterrelated commands.

`20:29:02.273 CMD: setoption80 1
20:29:02.277 RSL: RESULT = {"SetOption80":"ON"}
20:29:23.361 CMD: shutteropen
20:29:23.367 RSL: RESULT = {"Command":"Error"}
`
I'll post the solution, if I found one one day, here...

Ansgar66, 2023-07-16:
got it working!: 
I used SteWer's syntax to downgrade to version 12.1.1 altering his OTA-link. on this version it works again!
Thanks again for your assistance

florian-reus, 2023-07-31:
Hi,
I also had a problem with shutter after upgrade to 13.0.0. in my setup the "interlock 1,2" setting was lost during upgrade.

stefanbode, 2023-08-02:
yes if `interlock 1,2` gets lost the shutter will automatically execute `shutterrelay1 0` to prevent damage.
To fix fist `interlock 1,2` and then redefine shutterrelay: `shutterrelay1 1`
Anyhow all other settings should be now back. Shutter gets not deleted but made invisible for security. interlock is essential