arendst/Tasmota #19130: Documentation - Missing parameters for touchscreen driver XPT2046

nobodyman1, 2023-07-16:
### PROBLEM DESCRIPTION
In the documentation of the [Universal Display Driver](https://tasmota.github.io/docs/Displays/#universal-display-driver) the description of the ```:TS``` parameter is incomplete.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
If I look into the source code I found the following function call: ```XPT2046_Touch_Init(touch_cs, irqpin, bus - 1);```.
Please add the missing parameters to the documentation!
Tank you!

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-07-16:
Mhh, this is in the documentation 
>` :TS,CS_PIN `
Defines a touch panel an SPI bus with chip select CS_PIN (or *)

closed 2023-07-16 by Jason2866

Jason2866, 2023-07-17:
Not provided as setting in Display.ini. Not used.

nobodyman1, 2023-07-17:
I could not believe, because if I don´t add this parameters to my display.ini the display and the touch aren´t initialized.
Please look at #18924 for an example.
I have two displays which I´m trying to use:
ESP32-3248S035_R needs ```:TS,33,-1,3```
AZ-Touch MOD need: ```:TS,*,27```

sfromis, 2023-07-17:
The driver code does include logic to check for 3 paramters on `:TS`
https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_xdsp_display/xdsp_17_universal.ino#L400-L417

nobodyman1, 2023-07-20:
Who can fix this issue?
Should I open an issue in https://github.com/tasmota/docs instead?
Is this the right place?

barbudor, 2023-07-20:
You can click on the pencil icon on the top right of the doc page to submit an edit