arendst/Tasmota #19164: Rotary encoder only works in one direction (and two other bugs)

Nullaccount, 2023-07-22:
### PROBLEM DESCRIPTION

The commit [541443f](https://github.com/arendst/Tasmota/blame/e3c1aa8836bab444879074f58aacb436c293f44b/tasmota/support_rotary.ino) to support_rotary.ino broke rotary support for anyone except those specifying the "Xiaomi Mi Desk" rotary configuration (i.e. not using the OPTION_A 5 setting).

By selecting ROTARY_OPTION3 as the "default" the ISR (RotaryIsArg) is only called on the falling edge of channel A. Going back to the first implementation of rotary support I'm not sure how this option was ever worked as it will only capture incremental encoders in the same state (Channel B will always be high on the rising edge of A and will always be low on the falling edge of A).

```
                              _______         _______
            GPIO_ROT1A ______|       |_______|       |______ GPIO_ROT1A
negative <--              _______         _______         __            --> positive
            GPIO_ROT1B __|       |_______|       |_______|   GPIO_ROT1B
```

It could be fixed by either making the code match the MiDesk ISR (i.e. trigger on change for both Channel A and B, then check both pins in the ISR), or, if you *really* want to have one fewer interrupts at the cost of halving the encoder resolution you could change the interrupt to trigger on CHANGE instead of RISING or FALLING. I'd probably lean towards just merging the ISRs and removing the debounce code but another solution would be to change the interrupt trigger at [support_rotary.ino#L179](https://github.com/arendst/Tasmota/blob/c7a22f175b11da35aee55146a30401f69d883c36/tasmota/tasmota_support/support_rotary.ino#L179) to CHANGE.

I found at least one reference to someone who noticed their encoder only works in one direction (https://discord.com/channels/479389167382691863/1096811519414632619/1112403559783350352)

#### OTHER BUGS:
1. The LightColorOffset function in xdrv_04_light.ino creates an unsigned int underflow instead of wrapping from 0 to 359. The effect is that if you try to offset down through the color wheel you'll get wrapped back around to 65535 and it won't work again until you spin yourself all the way back down to 359. For reference the fix I did is below. Changed first 4 lines of [LightColorOffset@xdrv_04_light.ino#L1357](https://github.com/arendst/Tasmota/blob/c7a22f175b11da35aee55146a30401f69d883c36/tasmota/tasmota_xdrv_driver/xdrv_04_light.ino#L1357)
```
void LightColorOffset(int32_t offset) {
  int16_t hue;
  uint8_t sat;
  light_state.getHSB(reinterpret_cast<uint16_t*>(&hue), &sat, nullptr); 
  offset = max(min(offset, 359),-359);
```

2. Increasing SetOption43 above 100 (or above 33 if using OPTION_A 5) will break the dimming function (uint division of 100 by >100 is 0). Increasing above 350/3 or 360/3 will then break CT and Color controls respectively. The quick fix I used is below for reference but I'll eventually rewrite my own fork to support fractional steps and I'd be inclined to say that that's the real fix 🤷‍♀️

```
  Rotary.dimmer_increment = 100 / min((uint8_t)100, max_steps);  // Dimmer 1..100 = 100
  Rotary.ct_increment =     350 / min((uint8_t)350, max_steps);  // Ct 153..500 = 347
  Rotary.color_increment =  360 / min((uint8_t)360, max_steps);  // Hue 0..359 = 360
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Custom, can be reproduced with an esp8266 and a rotary encoder.
- [x] Tasmota binary firmware version number used: 13.0.0.2(tasmota) (2_7_4_9/2.2.2-dev(38a443e))
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: PlatformIO and USB
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
NOTE: This is with OPTION_A 5 enabled and is working, without OPTION_A 5 the rotary is broken as described.
03:59:07.886 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
03:59:07.962 SRC: Backlog
03:59:07.965 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
03:59:07.971 MQT: stat/test-light/RESULT = {"NAME":"TestLight","GPIO":[32,5732,33,0,3296,3264,0,0,418,417,416,419,0,0],"FLAG":0,"BASE":18}
03:59:08.190 SRC: Backlog
03:59:08.193 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
03:59:08.198 MQT: stat/test-light/RESULT = {"Module":{"0":"TestLight"}}
03:59:08.440 SRC: Backlog
03:59:08.443 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
03:59:08.454 MQT: stat/test-light/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"5732":"Option A5"},"GPIO2":{"33":"Button2"},"GPIO3":{"0":"None"},"GPIO4":{"3296":"Rotary B1"},"GPIO5":{"3264":"Rotary A1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"418":"PWM3"},"GPIO13":{"417":"PWM2"},"GPIO14":{"416":"PWM1"},"GPIO15":{"419":"PWM4"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE

1. Setup an ESP8266 with latest tasmota
4. Connect an incremental rotary encoder
5. Configure two rotary encoder pins as "Rotary A 1" and Rotary B 1"
6. Enable something to debug (add rules, connect a light, etc)
7. Turn rotary encoder
8. Rotary encoder will only increment (or decrement if you swap the A and B pins).

#### Additional Bugs:
`SetOptions43 101`: Rotate the encoder in either direction and the dimming will not change.

`Option_A 5`: Holding Button 1 (index 0) and rotating encoder "down" will cause an underflow and set the hue to 65535. Hue will be stuck > 359 until you've spun the encoder enough to trigger the call to LightColorOffset ≈ 100 times.

### EXPECTED BEHAVIOUR
**Option_A 5 not set**: Rotary encoder should increment in one direction and decrement in another.

**Option_A 5 set**: holding button 1 (index 0) and rotating the encoder left should reduce the hue and eventually wrap around back to 359.

**With or without Option_A 5 flag**: Setting the SetOption43 "too high" shouldn't silently break the dimming/color control functions, worst case scenario they should just be capped at the maximum supported resolution. Best case scenario the encoder support would allow fractional step sizes (storing the fractions and calling external functions once a full integer has accumulated and/or using higher resolution functions where appropriate).

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Nullaccount, 2023-08-17:
This is still an issue.

arendst, 2023-08-25:
I guess the "break" is dependent on the rotary hardware. Both my MiDesk and non-MiDesk rotaries still work fine in both directions with the corresponding hardware option (either using define MI_DESK_LAMP or (no) option A5). The difference in rotary hardware is signalled by this.

I did change both the color overflow and SO43 range issues as you suggested

Nullaccount, 2023-08-26:
Could be some combination of noise and the debounce code, On a re-read the isr makes more sense now, but I could see how it could get locked into one direction or another.

closed 2023-08-26 by Nullaccount