arendst/Tasmota #19168: Script: Invalid IF statements evaluation

peclik, 2023-07-23:
### PROBLEM DESCRIPTION
The correctness of IF statement evaluation depends on spaces or empty lines before IF statement.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Generic
- [x] Tasmota binary firmware version number used: 13.0.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
00:00:00.001 HDW: ESP8266EX
00:00:00.053 CFG: Loaded from flash at F6, Count 53
00:00:00.059 QPC: Count 1
00:00:00.067 Project tasmota - door-rfid Version 13.0.0(TasmoCompiler-esp8266generic)-2_7_4_9(2023-07-21T21:52:17)
00:00:00.070 Script: nv=2, tv=0, vns=7, vmem=51, smem=8192
00:00:00.073 no meter section found!
00:00:00.075 WIE: Wiegand Rfid Reader detected
00:00:00.086 BBBBBBB
00:00:00.180 WIF: Checking connection...
00:00:00.181 WIF: Attempting connection...
...
00:00:01.003 BBBBBBB
00:00:01.402 BBBBBBB
00:00:01.800 WIF: Checking connection...
...
03:34:42.002 BBBBBBB
03:34:42.046 MQT: Attempting connection...
03:34:42.057 MQT: Connected
...
03:34:42.414 AAAAAAA
03:34:42.416 BBBBBBB
03:34:43.419 BBBBBBB
03:34:44.420 BBBBBBB
03:34:45.459 QPC: Reset
03:34:45.462 BBBBBBB
03:34:46.535 MQT: house/door/rfid/tele/STATE = {"Time":"2023-07-23T03:34:46","Uptime":"0T00:00:08","UptimeSec":8,"Vcc":2.998,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"ufo.skolni.home","BSSId":"CC:2D:E0:30:5A:C3","Channel":3,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}
03:34:46.560 BBBBBBB
03:34:46.619 HTP: Main Menu
03:34:47.420 APP: Boot Count 22
03:34:47.422 BBBBBBB
03:34:47.578 CFG: Saved to flash at F5, Count 54, Bytes 4096
03:34:48.436 BBBBBBB
03:34:49.437 BBBBBBB
03:34:50.420 BBBBBBB
03:34:51.477 BBBBBBB
...
03:34:52.442 BBBBBBB
03:34:53.444 BBBBBBB
03:34:54.427 BBBBBBB
03:34:55.426 BBBBBBB

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
03:30:18.821 MQT: house/door/rfid/stat/STATUS = {"Status":{"Module":0,"DeviceName":"door-rfid","FriendlyName":["door-rfid"],"Topic":"door/rfid","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
03:30:18.833 MQT: house/door/rfid/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:54:25","StartupUTC":"2023-07-23T00:35:53","Sleep":50,"CfgHolder":4617,"BootCount":21,"BCResetTime":"2023-07-21T16:50:07","SaveCount":52,"SaveAddress":"F7000"}}
03:30:18.840 MQT: house/door/rfid/stat/STATUS2 = {"StatusFWR":{"Version":"13.0.0(TasmoCompiler-esp8266generic)","BuildDateTime":"2023-07-21T21:52:17","Boot":22,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"354/699"}}
03:30:18.849 MQT: house/door/rfid/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ufo.skolni.home",""],"TelePeriod":20,"Resolution":"558180C0","SetOption":["00008008","2805C80001000600003C5A00192800000000","00000080","00006000","00004000","00000000"]}}
03:30:18.861 MQT: house/door/rfid/stat/STATUS4 = {"StatusMEM":{"ProgramSize":540,"Free":460,"Heap":16,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8700058D","04080001","02040047","00000080","00000008","00000000","04000000","04000000","00000000"],"Drivers":"1,2,3,9,10,12,62","Sensors":"1,3,5,15,22,53,82","I2CDriver":""}}
03:30:18.873 MQT: house/door/rfid/stat/STATUS5 = {"StatusNET":{"Hostname":"door-rfid","IPAddress":"192.168.50.150","Gateway":"192.168.50.3","Subnetmask":"255.255.255.0","DNSServer1":"192.168.50.3","DNSServer2":"93.99.200.155","Mac":"50:02:91:47:FF:B4","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
03:30:18.881 MQT: house/door/rfid/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.50.20","MqttPort":1883,"MqttClientMask":"door-rfid","MqttClient":"door-rfid","MqttUser":"door-rfid","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
03:30:18.888 MQT: house/door/rfid/stat/STATUS7 = {"StatusTIM":{"UTC":"2023-07-23T01:30:18","Local":"2023-07-23T03:30:18","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"06:12","Sunset":"21:41"}}
03:30:18.896 MQT: house/door/rfid/stat/STATUS10 = {"StatusSNS":{"Time":"2023-07-23T03:30:18"}}
03:30:18.905 MQT: house/door/rfid/stat/STATUS11 = {"StatusSTS":{"Time":"2023-07-23T03:30:18","Uptime":"0T00:54:25","UptimeSec":3265,"Vcc":3.002,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxxxx","BSSId":"CC:2D:E0:11:11:C3","Channel":3,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```

### TO REPRODUCE
Using the following Script code:

```
>D
code=0
t:x=3

>S
if chg[code]>0 {
  print AAAAAAA

  if 0==0 {
  }
  if 1==1 {
    print BBBBBBB
  }
}

if x==0 {
  x=-1
  code=77
}
```

In the log, one can see:
AAAAAAA
BBBBBBBB
BBBBBBBB
BBBBBBBB

### EXPECTED BEHAVIOUR
The program would log:
AAAAAAA
BBBBBBBB
just once.

It will work correctly if either spaces are deleted before ` if 0==0 ` or empty line is deleted before the statement.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2023-07-23:
ok, found the bug, will do a pr soon.  until that delete empty lines.

closed 2023-07-23 by arendst

peclik, 2023-07-23:
It works. Thanks.

PR #19171 (2023-07-23): fix empty line problem
## Description:

**Related issue (if applicable):** fixes #19168

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.10
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOT [... cut]