arendst/Tasmota #19185: DHT12 need workaround to start working

vinibali, 2023-07-24:
### PROBLEM DESCRIPTION
DHT12 i2c temperature sensor needs additional steps to start working.
I need to execute i2cscan command from console and trigger a reboot.
After that until the next powercycle it works just fine.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): NodeMCU
- [ ] Tasmota binary firmware version number used: 13.0.0.3
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: OTA
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:55:37.617 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
21:55:37.848 RSL: RESULT = {"Module":{"0":"Generic"}}
21:55:38.056 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
N/A

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:57:40.064 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_287E02","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
21:57:40.095 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-sensors.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:06:15","StartupUTC":"2023-07-24T20:51:25","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"2023-07-08T09:28:06","SaveCount":26,"SaveAddress":"F8000"}}
21:57:40.128 RSL: STATUS2 = {"StatusFWR":{"Version":"13.0.0.3(sensors)","BuildDateTime":"2023-07-24T16:08:25","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"368/699"}}
21:57:40.148 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["OpenWrt",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
21:57:40.185 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":700,"Free":300,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9A8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","54000020","0008C081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34,62,65,66,68","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,98,103,105","I2CDriver":"7,8,9,10,!11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,69,76"}}
21:57:40.235 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-287E02-7682","IPAddress":"192.168.1.102","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"EC:FA:BC:28:7E:02","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
21:57:40.257 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_287E02","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:57:40.277 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-07-24T20:57:40","Local":"2023-07-24T21:57:40","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"05:13","Sunset":"20:39"}}
21:57:40.298 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-07-24T21:57:40","DHT12":{"Temperature":27.7,"Humidity":24.4,"DewPoint":5.5},"TempUnit":"C"}}
21:57:40.309 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-07-24T21:57:40","Uptime":"0T00:06:15","UptimeSec":375,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"OpenWrt","BSSId":"B0:4E:26:F0:8D:EA","Channel":11,"Mode":"11n","RSSI":100,"Signal":-21,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
N/A
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
* Add configuration
* Apply configuration
* No sensor data found on the main screen

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
* Add configuration
* Apply configuration
* Sensor data can be found on the main screen

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
NA

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
NA

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-07-25:
Duplicate, search in issues for solution.

closed 2023-07-25 by Jason2866

vinibali, 2023-07-30:
I was reading trough many issues related to the DHT family, but I haven't found any which is relevant.
Can you please help me any find this?

Jason2866, 2023-07-30:
https://github.com/arendst/Tasmota/issues/17944