arendst/Tasmota #19194: iTron Cyble Pulse counter is higher than real pulses

hallard, 2023-07-26:
### PROBLEM DESCRIPTION
Pulse counter does not match real pulse 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: 13.0.0(tasmota32)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmota WebFlasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
09:40:10.322 RSL: RESULT = {"NAME":"Tiny Pico","GPIO":[1,1,1,1,1376,66,1,1,1,1,355,129,1,1,736,672,0,1,1,704,0,352,353,354,0,0,0,0,1,128,1,1,1,0,0,1],"FLAG":0,"BASE":1}
09:40:10.541 RSL: RESULT = {"Module":{"0":"Tiny Pico"}}
09:40:10.749 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"1376":"WS28121"},"GPIO5":{"66":"Button_n3"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"355":"Counter4"},"GPIO15":{"129":"Button_in2"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"736":"SPI CLK1"},"GPIO19":{"672":"SPI MISO1"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"704":"SPI MOSI1"},"GPIO24":{"0":"None"},"GPIO25":{"352":"Counter1"},"GPIO26":{"353":"Counter2"},"GPIO27":{"354":"Counter3"},"GPIO32":{"0":"None"},"GPIO33":{"128":"Button_in1"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
09:43:02.814 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_D348B0","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
09:43:02.822 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"RTC Watch dog reset digital core and rtc module","Uptime":"1T13:19:52","StartupUTC":"2023-07-24T19:23:10","Sleep":50,"CfgHolder":4617,"BootCount":25,"BCResetTime":"2023-07-21T15:26:24","SaveCount":78}}
09:43:02.829 RSL: STATUS2 = {"StatusFWR":{"Version":"13.0.0(tasmota32)","BuildDateTime":"2023-06-26T10:54:38","Core":"2_0_10","SDK":"4.4.5.230614","CpuFrequency":160,"Hardware":"ESP32-PICO-D4 rev.1","CR":"388/699"}}
09:43:02.832 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["CH2I-HOTSPOT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000800B","2805C80001000600003C5A0A192800000000","00800080","00006000","00004000","00000000"]}}
09:43:02.849 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1870,"Free":1009,"Heap":104,"StackLowMark":3,"PsrMax":4093,"PsrFree":4093,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","480840D2","20200000","D4BC482D","000A80B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,68,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77"}}
09:43:02.858 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-D348B0-2224","IPAddress":"192.168.1.183","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"212.27.40.240","DNSServer2":"212.27.40.241","Mac":"98:CD:AC:D3:48:B0","IP6Global":"fd00::1:9acd:acff:fed3:48b0","IP6Local":"fe80::9acd:acff:fed3:48b0","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"212.27.40.240","DNSServer2":"212.27.40.241","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
09:43:02.863 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D348B0","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
09:43:02.869 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-07-26T08:43:02","Local":"2023-07-26T09:43:02","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"05:15","Sunset":"20:37"}}
09:43:02.877 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-07-26T09:43:02","COUNTER":{"C1":5883,"C2":4041,"C3":5892,"C4":172}}}
09:43:02.887 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-07-26T09:43:02","Uptime":"1T13:19:52","UptimeSec":134392,"Heap":102,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"Berry":{"HeapUsed":8,"Objects":123},"POWER":"OFF","Dimmer":52,"Color":"770085","HSBColor":"294,100,52","Channel":[47,0,52],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"CH2I-HOTSPOT","BSSId":"94:83:C4:12:12:76","Channel":1,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Configure a GPIO as counter (here GPIO14)

[image]
Then connect Itron Water Meter Cyble Sensor onto water meter and then start to pass 10 liters onto the water meter (measuring with Official Water meter number) and then check count. The pulse output has been connected to Logic Analyser in the same time to confirm issue.

Reset all counters with 

`backlog 0 counter1 0 ; counter2 0 ; counter3 0 ;counter4 0`

With 10 liters passed thru the Cyble sensor has output 100 pulses, exactly number expected
[image]

Each pulse is by concept 65ms long
[image]

### EXPECTED BEHAVIOUR

correct pulse number detected, at least near 100

### SCREENSHOTS

[image]

Tasmota counter show 193 (almost twice the real number)
[image]

Made different tests, issue always the same

### ADDITIONAL CONTEXT

May ba a specific configuration needs to be done in my case?

Jason2866, 2023-07-26:
Have you checked the signal with an oscilloscope? Probably signal waveform is not as it looks like in the logic analyzer. So a double count can happen.

hallard, 2023-07-26:
Not yet, it's an electronic pulse generated by the official Itron head for water counter (calibrated and no bounce that, is why this head is excellent, it has a battery inside to do the stuff and output the 65ms pulse signal)

It's used to invoice customers all around the world. I guess if it has an issue we would know it :-) 

I already used it with my LoRaWAN custom board and MBED-OS and it was working fine this is why I'm really surprised.

But I will check in last solution but

hallard, 2023-07-26:
I removed all berry script and other counters just in case, same issue.

[image]

sfromis, 2023-07-26:
I'd suggest playing around with the counter debounce settings.

hallard, 2023-07-26:
@sfromis 

`CounterDebounceLow 20`

Did the trick, not sure to understand why it does not woks without that one since the pulse signal is clean

anyway issue solved, thanks

Jason2866, 2023-07-26:
Mhh, don't think the signal is clean. Maybe it needs a pull up / pull down resistor?

hallard, 2023-07-26:
Correct the only reason why, I think same, really need to check on the scope, since the head is battery powered (3.6V) it output 0V or 3.6V (I guess) so unlike reed it doest not need pullup/down

Jason2866, 2023-07-26:
Commercial cases will have a schmitt trigger input to clean signals. So resistors are not needed at all.

closed 2023-07-26 by Jason2866

sfromis, 2023-07-26:
While the ESP32 does have internal pullup resistors by default enabled by Tasmota, they may not be strong enough.

hallard, 2023-07-26:
Correct for the output, need to check with 1K resistor pullup (to decrease the default ESP internal that may be around 50K) I will also check how low can I reduce `debounceLow` parameter