arendst/Tasmota #19207: Four files using wrong compiler directive #ifdef EPS8266 instead of #ifdef ESP8266

msedv, 2023-07-28:
### PROBLEM DESCRIPTION

In these four files #ifdef EPS8266 is used instead of #ifdef ESP8266 - which e.g. broke the MFRC522 functionality between 12.5 and 13.x:

- tasmota/tasmota_xdrv_driver/xdrv_10_scripter.ino
- tasmota/tasmota_xdrv_driver/xdrv_50_filesystem.ino
- tasmota/tasmota_xsns_sensor/xsns_80_mfrc522.ino
- CHANGELOG.md

Symptoms in xsns_80_mfrc522.ino: no card detected, no error  since SPI isn't opened.

I corrected the files locally and tried to to a git push but get an error 403.

And: whoever

arendst, 2023-07-29:
My fault. I’m out until next friday so if you manage to do a PR would be great

msedv, 2023-07-29:
Done: https://github.com/arendst/Tasmota/pull/19209

closed 2023-07-29 by Jason2866