arendst/Tasmota #19217: Tasmota-IR 13.0.0 HVAC vendor ELECTRA_AC - can't turn off AC

Malakai-DB, 2023-07-30:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I am currently using Tasmota-IR on 2 devices to control (among others) 2 ACs which use the ELECTRA_AC vendor.
On version 12.5.0 of Tasmota, I had no issue sending commands to power on/off the ACs from Home-Assistant.
I am using the following configuration in Home-Assistant for that :

```
switch:
    - name: first_ac
      unique_id: "first_ac"
      command_topic: "cmnd/ir_ac1/IRHVAC"
      payload_on: '{"Vendor":"ELECTRA_AC","Model":-1,"Mode":"Cool","Power":"On","Celsius":"On","Temp":20,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}'
      payload_off: '{"Vendor":"ELECTRA_AC","Model":-1,"Mode":"Cool","Power":"Off","Celsius":"On","Temp":20,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}'
      state_topic: "stat/ir_ac1/RESULT"
      state_on: '{"IRHVAC":{"Vendor":"ELECTRA_AC","Model":-1,"Mode":"Cool","Power":"On","Celsius":"On","Temp":20,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}'
      state_off: '{"IRHVAC":{"Vendor":"ELECTRA_AC","Model":-1,"Mode":"Cool","Power":"Off","Celsius":"On","Temp":20,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}'

```
I upgraded to 13.0.0 and I can't power OFF the ACs anymore. It is working when I want them to power ON, but not OFF (it is true for both of them, with 2 IR devices as the 2 ACs are in different rooms). I can however control any other IR device that I was able to control on 12.5.0 (TV / STB / etc).
Once I downgraded to 12.5.0, everything came back to normal, I can power ON and OFF both ACs.

Between the upgrades (from 12.5.0 to 13.0.0) I never changed any of the configurations (in Tasmoto or Home-Assistant).

Is this a known issue?
Does anybody else experience it?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): [Orvibo Magic Cube](https://templates.blakadder.com/orvibo_CT10W.html)
- [ X] Tasmota binary firmware version number used: 12.5.0 (working) and 13.0.0 (not working)
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [ X] Flashing tools used: don't remember
- [ X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
MQT: stat/ir_ac1/RESULT = {"NAME":"Orvibo CT10W","GPIO":[0,0,0,0,32,1088,0,0,288,0,1056,289,0,0],"FLAG":0,"BASE":18}
MQT: stat/ir_ac1/RESULT = {"Module":{"0":"Orvibo CT10W"}}
MQT: stat/ir_ac1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"32":"Button1"},"GPIO5":{"1088":"IRrecv"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"288":"Led1"},"GPIO13":{"0":"None"},"GPIO14":{"1056":"IRsend1"},"GPIO15":{"289":"Led2"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
stat/ir_ac1/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
stat/ir_ac1/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
stat/ir_ac1/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
MQT: stat/ir_ac1/STATUS = {"Status":{"Module":0,"DeviceName":"ir-ac1","FriendlyName":["ir-ac1"],"Topic":"ir_ac1","ButtonTopic":"0","Power":-1,"PowerOnState":2,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
MQT: stat/ir_ac1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-ir.bin.gz","RestartReason":"Software/System restart","Uptime":"7T13:46:35","StartupUTC":"2023-07-22T17:09:16","Sleep":50,"CfgHolder":4617,"BootCount":32,"BCResetTime":"2022-03-09T11:27:40","SaveCount":113,"SaveAddress":"F9000"}}
MQT: stat/ir_ac1/STATUS2 = {"StatusFWR":{"Version":"12.5.0(ir)","BuildDateTime":"2023-04-17T08:03:55","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"392/699"}}
MQT: stat/ir_ac1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Private",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A00192800000000","00000080","00006000","00004000","00000000"]}}
MQT: stat/ir_ac1/STATUS4 = {"StatusMEM":{"ProgramSize":638,"Free":364,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440A1","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","0F1A0582","04000001","00000003","00000000","00000000","00000000","00000000","04000000","00000080"],"Drivers":"1,2,4,5,7,9,10,12,62","Sensors":"1"}}
MQT: stat/ir_ac1/STATUS5 = {"StatusNET":{"Hostname":"ir-ac1","IPAddress":"10.0.0.17","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer1":"10.0.0.1","DNSServer2":"0.0.0.0","Mac":"84:XX:XX:XX:XX:XX","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
MQT: stat/ir_ac1/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.5","MqttPort":1883,"MqttClientMask":"ir-ac1","MqttClient":"ir-ac1","MqttUser":"ir-ac1","MqttCount":10,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
MQT: stat/ir_ac1/STATUS7 = {"StatusTIM":{"UTC":"2023-07-30T06:55:51","Local":"2023-07-30T09:55:51","StartDST":"2023-03-26T03:00:00","EndDST":"2023-10-29T04:00:00","Timezone":99,"Sunrise":"07:20","Sunset":"22:32"}}
MQT: stat/ir_ac1/STATUS10 = {"StatusSNS":{"Time":"2023-07-30T09:55:51"}}
MQT: stat/ir_ac1/STATUS11 = {"StatusSTS":{"Time":"2023-07-30T09:55:51","Uptime":"7T13:46:35","UptimeSec":654395,"Vcc":3.514,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":10,"Wifi":{"AP":1,"SSId":"Private","BSSId":"02:XX:XX:XX:XX:XX","Channel":6,"Mode":"11n","RSSI":94,"Signal":-53,"LinkCount":5,"Downtime":"0T00:00:15"}}}

```
- [ X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
- [X ] With version 13.0.0 when I power ON and then OFF the AC, all I see is the power ON output:
```lua
  Console output here:
SRC: MQTT
CMD: Grp 0, Cmd 'IRHVAC', Idx 1, Len 242, Pld -99, Data '{"Vendor":"ELECTRA_AC","Model":-1,"Mode":"Cool","Power":"On","Celsius":"On","Temp":20,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}'
MQT: stat/ir_ac1/RESULT = {"IRHVAC":{"Vendor":"ELECTRA_AC","Model":-1,"Command":"Control","Mode":"Cool","Power":"On","Celsius":"On","Temp":20,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1,"iFeel":"Off","SensorTemp":"n/a"}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Configure Tasmota and Home-Assistant with MQTT and try to power OFF the AC (Electra Vendor) with version 13.0.0 of Tasmota.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The Electra AC should power OFF with version 13.0.0 as it is happening with version 12.5.0

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-07-30:
Did you really searched?

https://github.com/arendst/Tasmota/issues/18923

Malakai-DB, 2023-07-30:
I must admit I searched for ELECTRA and IR keywords, but not HVAC and MQTT, sorry.

This being said, the issue doesn't seem to be quite the same (I don't think he is using an IR blaster in his setup). What I can't figure out, is if the bug corrected in the dev branch will also fix my issue. If it is the case, I will just wait for the next stable release (I don't have another IR blaster to fiddle with and wouldn't want to "disturb" climate control in the house for the moment by testing).

barbudor, 2023-07-30:
Why just not using the dev version?
'release' is just a word. It's a freeze of the development version on one day but it doesn't I sure any better reliability

In any case, the dev team will not investigate any issue that has not been reproduced on the current dev version.

Malakai-DB, 2023-07-30:
With the latest dev version (13.0.0.3) I get a reset (and the unit keeps restarting after a few minutes) and I see the following messages in the console:
```
MQT: tele/ir_ac1/INFO1 = {"Info1":{"Module":"Orvibo CT10W","Version":"13.0.0.3(ir)","FallbackTopic":"cmnd/ir-ac1_fb/","GroupTopic":"cmnd/tasmotas/"}}
MQT: tele/ir_ac1/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"ir-ac1","IPAddress":"10.0.0.17"}}
MQT: tele/ir_ac1/INFO3 = {"Info3":{"RestartReason":{"Exception":3,"Reason":"Exception","EPC":["4000bf64","00000000","00000000"],"EXCVADDR":"40295d2f","DEPC":"00000000","CallChain":["40295d2f","402541d4","4029006c","402012fe","40250230","4022fded","4020e7d3","40250b64","4020ef06","4028d940","4020f794","4021c6d4","4028d938","4028d930","40290878","40100a5a","4021c950","40212788","4021fb69","4020de68","40224ff6","40295a88","4020f999","40102cc4","40104d32","40216d5e","40104681","4021c8a7","4021fca0","401013f7","40267d1c"]},"BootCount":40}}
```
I think I will wait for the next stable release and check again if the issue is still present.
@barbudor : thanks for the time spent on this.

closed 2023-07-30 by Malakai-DB