arendst/Tasmota #19234: shutter in button mode controls all other group topic shutters, but stop himself immediate with 13.0.0

florian-reus, 2023-07-31:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
when used in shutter button mode, all other group topic shutters are started. local shutter is stopped right away (~half a second)
looks similar as https://github.com/arendst/Tasmota/issues/19085 
--> starting with local button, receiving it's own mqtt message and stopping right away

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): different 2 gang switches, Athom and other brand
- [ ] Tasmota binary firmware version number used: 13.0.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: none, OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:50:06.351 CMD: Backlog Template; Module; GPIO 255
19:50:06.426 MQT: stat/rollo/eg/esszimmer/RESULT = {"NAME":"Rollo","GPIO":[1,0,321,0,0,33,0,0,32,224,0,225,320,0],"FLAG":0,"BASE":18}
19:50:06.651 MQT: stat/rollo/eg/esszimmer/RESULT = {"Module":{"0":"Rollo"}}
19:50:06.955 MQT: stat/rollo/eg/esszimmer/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"321":"Led_i2"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"33":"Button2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"32":"Button1"},"GPIO13":{"224":"Relay1"},"GPIO14":{"0":"None"},"GPIO15":{"225":"Relay2"},"GPIO16":{"320":"Led_i1"},"GPIO17":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here: no rules used 

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.001 HDW: ESP8266EX
00:00:00.046 CFG: Loaded from flash at FB, Count 4216
00:00:00.066 SHT: ShutterMode: 1
00:00:00.070 Project tasmota - rollo_esszimmer Version 13.0.0(tasmota)-2_7_4_9(2023-06-26T10:42:07)
00:00:00.711 WIF: Connecting to AP1 meta_legacy Channel 9 BSSId D0:21:F9:80:21:95 in mode 11n as rollo-esszimmer...
00:00:03.807 WIF: Connected
00:00:04.011 HTP: Web server active on rollo-esszimmer with IP address 192.168.9.213
19:21:59.012 MQT: Attempting connection...
19:21:59.728 MQT: Connected
19:21:59.731 MQT: tele/rollo/eg/esszimmer/LWT = Online (retained)
19:21:59.733 MQT: cmnd/rollo/eg/esszimmer/POWER = 
19:21:59.744 MQT: tele/rollo/eg/esszimmer/INFO1 = {"Info1":{"Module":"Rollo","Version":"13.0.0(tasmota)","FallbackTopic":"cmnd/esszimmer_fb/","GroupTopic":"cmnd/rollo/eg/tasmotas/"}}
19:21:59.752 MQT: tele/rollo/eg/esszimmer/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"rollo-esszimmer","IPAddress":"192.168.9.213"}}
19:21:59.766 MQT: tele/rollo/eg/esszimmer/INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":1542}}
19:21:59.776 MQT: stat/rollo/eg/esszimmer/RESULT = {"POWER1":"OFF"}
19:21:59.781 MQT: stat/rollo/eg/esszimmer/POWER1 = OFF (retained)
19:21:59.787 MQT: stat/rollo/eg/esszimmer/RESULT = {"POWER2":"OFF"}
19:21:59.794 MQT: stat/rollo/eg/esszimmer/POWER2 = OFF (retained)
19:22:00.912 MQT: stat/rollo/eg/esszimmer/STATUS10 = {"StatusSNS":{"Time":"2023-07-31T19:22:00","Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}}
19:22:03.415 MQT: tele/rollo/eg/esszimmer/STATE = {"Time":"2023-07-31T19:22:03","Uptime":"0T00:00:10","UptimeSec":10,"Heap":26,"SleepMode":"Dynamic","Sleep":100,"LoadAvg":14,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"meta_legacy","BSSId":"D0:21:F9:80:21:95","Channel":9,"Mode":"11n","RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:05"}}
19:22:03.445 MQT: tele/rollo/eg/esszimmer/SENSOR = {"Time":"2023-07-31T19:22:03","Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
19:24:45.596 MQT: stat/rollo/eg/esszimmer/RESULT = {"POWER2":"ON"}
19:24:45.600 MQT: stat/rollo/eg/esszimmer/POWER2 = ON (retained)
19:24:45.604 MQT: stat/rollo/eg/esszimmer/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":0,"Tilt":0}}
19:24:45.610 MQT: cmnd/rollo/eg/tasmotas/ShutterPosition1 = 0
19:24:45.615 MQT: cmnd/rollo/eg/tasmotas/ShutterPosition2 = 0
19:24:45.620 MQT: cmnd/rollo/eg/tasmotas/ShutterPosition3 = 0
19:24:45.626 MQT: cmnd/rollo/eg/tasmotas/ShutterPosition4 = 0
19:24:45.631 MQT: stat/rollo/eg/esszimmer/RESULT = {"Shutter1":{"Button2":0}}
[... 32 lines cut ...]
19:25:23.438 MQT: stat/rollo/eg/esszimmer/RESULT = {"Shutter1":{"Position":6,"Direction":-1,"Target":0,"Tilt":0}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
-shutter mode
- several shutters with same group topic
- shutter button mode
- 19:46:54.168 MQT: stat/rollo/eg/esszimmer/RESULT = {"ShutterButton1":"1 100 50 74 100 - - - 1 1|2 0 50 24 0 - - - 1 1"}
- long press, so all other group topic shutters start
- short start, immediate stop of local shutter
- worked before 13.0.0 as described in manual

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
- as before - also local shutter closes or opens

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2023-08-02:
#19085 is not reproduceable. This is just because of missing index to the operation. I do not see a connection to this here. It may that we have a breaking change with long press. In the old versions you could asssign a long press event. e.g. move all shutter. Anyhow quite more common on the long press event is, that the shutter stop after releasing the button. Am I right that you configured the "close all" to the long press event? We have not made any changes to the esp8266. but the esp32 has way more options now. Therfore we have flexibility here to make some additional tweaks

stefanbode, 2023-08-02:
e.g. we can supress the stop of the local shutter if the long press event is configured. May this help?`Easy fix

florian-reus, 2023-08-02:
sorry, I was not precise. I only use button control -- ShutterButton<x>
command. No long press defined.

florian-reus, 2023-08-03:
tested now - no difference with longpress on or off (setoption73).
shutter stops when I release the longpress

stefanbode, 2023-08-03:
hi @florian-reus . I still did not get it. Specially because there some goupings defined. Pls can you specify what you do expect on which button event. As said the long-press event for shutters changed in the way that the shutter moves as long as you press and as soon as you release it stops moving. this works fine with ONE shutter. looking forward to hear from you. there is no longpress option required. the shutterbutton has an option for 1,2,3, press and hold=long press

stefanbode, 2023-08-03:
i'm a little scary about this one:
```
19:24:45.610 MQT: cmnd/rollo/eg/tasmotas/ShutterPosition1 = 0
19:24:45.615 MQT: cmnd/rollo/eg/tasmotas/ShutterPosition2 = 0
19:24:45.620 MQT: cmnd/rollo/eg/tasmotas/ShutterPosition3 = 0
19:24:45.626 MQT: cmnd/rollo/eg/tasmotas/ShutterPosition4 = 0
```
if there is a group command this should only be send to the shutter you defined. In this case you operate shutter1 and therefore I expect only: 
```
19:24:45.610 MQT: cmnd/rollo/eg/tasmotas/ShutterPosition1 = 0
```
nothing more. Anyhow lokking forward to a table of expected bahavior

stefanbode, 2023-08-05:
@florian-reus : Pls take a look on the submitted change. The local shutter will NOT stop after releasing the button if you have enabled the group topic for the HOLD event. In the shutterbutton definition the BOLD "1" is the one that changes the behavior. If there is no MQTT broadcast the local shutter will stop after releasing the button
ShutterButton1 1 100 50 74 100 - - - **1** 1
ShutterButton1 2 0 50 24 0 - - - **1** 1

closed 2023-08-05 by arendst

florian-reus, 2023-08-31:
tested on 13.1.0 - works again as desired, no config change done.