arendst/Tasmota #19235: Berry: gpio.pin() does not work for OPTION_A

hami89, 2023-07-31:
### PROBLEM DESCRIPTION
In berry script, the `gpio.pin_used(gpio.OPTION_A,0)` command always returns -1, even if such a gpio pin is configured. Meanwhile, `gpio.pin_used(gpio.OPTION_A, 0)` correctly returns true.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): esp32-devkit1 (esp-wroom-32)
- [x] Tasmota binary firmware version number used: 13.0.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: web tool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
23:54:38.340 CMD: Backlog Template; Module; GPIO 255
23:54:38.373 RSL: RESULT = {"NAME":"MyTemplate","GPIO":[1,1,1,1,1,263,1,1,1,1,261,544,1536,1472,262,6211,0,6210,6208,6209,0,258,259,260,0,0,0,0,256,257,4706,1,4704,0,0,4705],"FLAG":0,"BASE":1}
23:54:38.595 RSL: RESULT = {"Module":{"0":"MyTemplate"}}
23:54:38.808 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"263":"Relay_i8"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"261":"Relay_i6"},"GPIO15":{"544":"LedLink"},"GPIO16":{"1536":"PZEM016 Rx"},"GPIO17":{"1472":"PZEM0XX Tx"},"GPIO18":{"262":"Relay_i7"},"GPIO19":{"6211":"Option A4"},"GPIO20":{"0":"None"},"GPIO21":{"6210":"Option A3"},"GPIO22":{"6208":"Option A1"},"GPIO23":{"6209":"Option A2"},"GPIO24":{"0":"None"},"GPIO25":{"258":"Relay_i3"},"GPIO26":{"259":"Relay_i4"},"GPIO27":{"260":"Relay_i5"},"GPIO32":{"256":"Relay_i1"},"GPIO33":{"257":"Relay_i2"},"GPIO34":{"4706":"ADC Input3"},"GPIO35":{"0":"None"},"GPIO36":{"4704":"ADC Input1"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"4705":"ADC Input2"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
N/A
```
- [ ] Provide the output of this command: `Status 0`:
```lua
N/A
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
N/A
```

### TO REPRODUCE
Take an esp32 based device and configure a gpio pin as "Option A".
From the berry console, issue command `gpio.pin(gpio.OPTION_A,0)`.
It will return -1.

### EXPECTED BEHAVIOUR
The `gpio.pin(gpio.OPTION_A,0)` should return the configured gpio number (22 in this case) for the Option A pin.

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
I am developing a custom device.
I configured gpio 19, 21, 22, 23 respectively as Option-A 1, Option-A 2, Option-A 3 and Option-A 4.

I am using a berry-script based driver to handle these pins.
In berry, ```gpio.pin_used(gpio.OPTION_A, 0)``` works as expected, returning true for all 4 pins (0-3)"

_**Berry console output:**_
```lua
| gpio.pin_used(gpio.OPTION_A,0)
true
| gpio.pin_used(gpio.OPTION_A,1)
true
| gpio.pin_used(gpio.OPTION_A,2)
true
| gpio.pin_used(gpio.OPTION_A,3)
true
```

But ```gpio.pin(gpio.OPTION_A, 0)``` returns 0, no matter the input. (But it works correctly for all other devices in my config).

_**Berry console output:**_
```lua
| gpio.pin(gpio.OPTION_A,0)
-1
| gpio.pin(gpio.OPTION_A,1)
-1
| gpio.pin(gpio.OPTION_A,2)
-1
| gpio.pin(gpio.OPTION_A,3)
-1
```

I would expect ```gpio.pin(gpio.OPTION_A, 0)``` to return 22 (GPIO 22), as `gpio 255` command output above shows, that it is configured to gpio 22.

I would use `gpio.pin()` output to find the gpio assignment from the berry driver. I worked it around by issuing a `tasmota.cmd("gpio 255")` command from berry and processing the output, but it would be much easier to get the result in one code line.

sfromis, 2023-08-01:
Well, OPTION_A are not "physical pins", but a workaround for providing a runtime configurable option for drivers to modify their behavior, without the driver expecting that they can get a physical pin number for an option. Only whether the option is selected. Thus it makes sense that `gpio.pin_used` gives you what OPTION_A is meant for, while you cannot misuse gpio.pin to treat it as a physical pin assignment. The pins where OPTION_A is selected are not assigned to anything, they are still unused, with or without Berry. Thus it is working as designed that gpio.pin returns -1 for not assigned.

To reserve pins for Berry usage. you can configure as INPUT instead of expecting OPTION_A to be suitable for such.

s-hadinger, 2023-08-01:
This is due to the internal implementation of Tasmota. `OPTION_A` is a virtual type and is not actually attached to a GPIO. So returning the GPIO number for `OPTION_A` actually doesn't make much sense.

There is special code in `gpio.pin_used()` to detect `OPTION_A` although there is no GPIO attached.

hami89, 2023-08-01:
Well, I see your logic and I can accept that.
Still, if a pin **can be set** as Option A (or anything else) it would be logical to be able to query the same thing.

Please, at least update the documentation about this.

closed 2023-08-01 by hami89

sfromis, 2023-08-01:
If what you want is to "reserve" a pin for Berry, and have the Berry code find it without having to know the physical pin gpio number, you can as already suggested configure the pin as "Input" instead of trying to misuse Option A.

hami89, 2023-08-01:
As I mentioned, it can be worked around by either decoding the template or by `var pins = tasmota.cmd("gpio 255")` and using the resulting berry object.
I closed the issue anyway.

sfromis, 2023-08-01:
What I meant was just do not *misuse* the Option A feature for something it is not meant for, especially when you have a more suitable option.